openwrt/packages #726: privoxy yields config.new (instead of config)

enigmagroup, 2015-01-04:
The build script wants to rename ```/etc/privoxy/config```, which is inexistent.

The following patch solves the issue:

```diff
diff --git a/net/privoxy/Makefile b/net/privoxy/Makefile
index 4209cae..93d7967 100644
--- a/net/privoxy/Makefile
+++ b/net/privoxy/Makefile
@@ -55,7 +55,7 @@ MAKE_FLAGS:=
 define Build/Install
        $(call Build/Install/Default,)
        # rename original sample config from pkg_source to save existing one during install
-       mv $(PKG_INSTALL_DIR)/etc/privoxy/config $(PKG_INSTALL_DIR)/etc/privoxy/config.privoxy
+       mv $(PKG_INSTALL_DIR)/etc/privoxy/config.new $(PKG_INSTALL_DIR)/etc/privoxy/config.privoxy
 endef
 
 define Package/privoxy/conffiles
```

chris5560, 2015-01-04:
@enigmagroup 
This is done during **build install** directly after the default **build install** of privoxy.
It really exists in the **build** directory.
Have a look at **build_dir/target-.../privoxy-3.0.22-stable/ipkg-install/etc/privoxy/**
We need to rename here because otherwise it's copied during package/install where it is no longer needed because the package is using UCI configuration.
I crosschecked with **make package/privoxy/install V=s** and there is no error reported about non-existing file.
Christian

enigmagroup, 2015-01-08:
This is what I get:

```
Rewriting config for this installation
sed 's+^confdir \.+confdir /etc/privoxy+' config | \
        sed 's+^logdir \.+logdir /var/log/privoxy+' >config.tmp
mv config.updated config
Installing templates to /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/templates
Installing configuration files to /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy
Installing config as config.new
Installing trust as trust.new
Installing fresh default.action
Installing match-all.action as match-all.action.new
Installing user.action as user.action.new
Installing fresh default.filter
Installing user.filter as user.filter.new
Warning: Older config files are preserved. Check new versions for changes!
Checking logfiles in /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/var/log/privoxy
rm -f config.base config.tmp
Privoxy 3.0.22 stable installation succeeded!
The Privoxy configuration files have been installed in /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy
make[4]: Leaving directory `/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable'
# rename original sample config from pkg_source to save existing one during install
mv /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/config /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/config.privoxy
mv: cannot stat `/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/config': No such file or directory
make[3]: *** [/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/privoxy-3.0.22-stable/.built] Error 1
make[3]: Leaving directory `/home/x/openwrt-64.mostrecent/feeds/packages/net/privoxy'
make[2]: *** [package/feeds/packages/privoxy/compile] Error 2
make[2]: Leaving directory `/home/x/openwrt-64.mostrecent'
make[1]: *** [/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/x/openwrt-64.mostrecent'
make: *** [world] Error 2
```

enigmagroup, 2015-01-08:
Try it with **make package/privoxy/{clean,install} V=99**

chris5560, 2015-01-08:
@enigmagroup 
Don't know where
**"Warning: Older config files are preserved. Check new versions for changes!"**
your old config files coming from that already exists in the .../ipkg-install/etc/privoxy directory.
A make package/privoxy/clean should normally completely delete "privoxy-3.0.22....." from "target-i386...." directory.

To be shure everything works fine I did "make dirclean" and with make V=s I get:
```
/usr/bin/install -c -m 0644 privoxy.1  /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/usr/man/man1/privoxy.1
Rewriting config for this installation
sed 's+^confdir \.+confdir /etc/privoxy+' config | \
sed 's+^logdir \.+logdir /var/log/privoxy+' >config.tmp
mv config.updated config
Installing templates to /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/templates
Installing configuration files to /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy
Installing fresh default.action
Installing fresh default.filter
Creating logfiles in /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/var/log/privoxy
rm -f config.base config.tmp
Privoxy 3.0.22 stable installation succeeded!
The Privoxy configuration files have been installed in /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy
make[3]: Leaving directory '/home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable'
# rename original sample config from pkg_source to save existing one during install
mv /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/config /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/config.privoxy
touch /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/.built
mkdir -p /home/x/owbuild/trunk/bin/x86/packages /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/CONTROL /home/x/owbuild/trunk/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo
install -d -m0755 /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/usr/sbin
cp -fpR /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/usr/sbin/privoxy /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/usr/sbin/
install -d -m0755 /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/init.d
install -m0755 ./files/privoxy.init /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/init.d/privoxy
install -d -m0755 /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/hotplug.d/iface
install -m0755 ./files/privoxy.hotplug /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/hotplug.d/iface/80-privoxy
install -d -m0755 /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/privoxy
cp -fpR /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-install/etc/privoxy/* /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/privoxy/
# temporary needed if updating from old version
# otherwise old config file will be delete by opkg
install -m0600 ./files/privoxy.oldconfig /home/x/owbuild/trunk/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.22-stable/ipkg-x86/privoxy/etc/privoxy/config
```
Regardless this problem I found out, that from time to time it's a good idea to rebuild the "trunk" directory completly. Minimum is "make dirclean". A lot of compile problems will be fixed this way.
Christian

enigmagroup, 2015-02-11:
Hi Christian,

I have checked out three fresh openwrt-directories, building for x86, x86_64 and sunxi (banana pi). I have created a script that compiles openwrt in all those dirs. Here is the result: http://paste.scratchbook.ch/view/544f936e#PDbE1Q0y13FM4v7MpIJO19jjkkQz1I0f

It fails at privoxy and the problem is still the same.

What is wrong about my patch, or how do I need to modify it to get accepted? Or, can we maybe hide errors in "mv", like ```2> /dev/null```?

Thanks a lot for your help!

enigmagroup, 2015-02-11:
This is my build dir just after the compilation (and fail):

```
~/x/openwrt-32/build_dir/target-i386_i486_uClibc-0.9.33.2/privoxy-3.0.23-stable/ipkg-install/etc/privoxy $ l
total 228K
drwxr-xr-x 3 x x 4.0K Feb 11 04:23 .
drwxr-xr-x 3 x x 4.0K Feb 11 04:23 ..
-rw-rw---- 1 x x  61K Feb 11 04:23 config.new
-rw-rw---- 1 x x  88K Feb 11 04:23 default.action
-rw-rw---- 1 x x  40K Feb 11 04:23 default.filter
-rw-rw---- 1 x x  627 Feb 11 04:23 match-all.action.new
drwxr-xr-x 2 x x 4.0K Feb 11 04:23 templates
-rw-rw---- 1 x x 3.7K Feb 11 04:23 trust.new
-rw-rw---- 1 x x 8.6K Feb 11 04:23 user.action.new
-rw-rw---- 1 x x 3.3K Feb 11 04:23 user.filter.new
```

chris5560, 2015-02-11:
Hi,

looking into your "build_dir" listing shows that there are to much *.new files.
It looks like the directory is not empty (cleared) when starting make.
From your first post I remember that privoxy's own install not overwriting existing files.
It moves the new files to *.new . Same problem here but not only the "config" file.
Before I do a final check of my packages I do a "make dirclean".
Sometimes if running into "undefined" errors I reload the whole build structure.
Most problems are fixed afterwards.

Please contact me via direct mail.
I will prepare a minimized Makefile for you, building privoxy with default settings.
Without Build/Install and other add-ons to convert privoxy-config into uci-config.

Christian

enigmagroup, 2015-02-11:
> looking into your "build_dir" listing shows that there are to much *.new files.
> It looks like the directory is not empty (cleared) when starting make.
It is a freshly checked out openwrt.git

Here are my exact steps I took:

```
git clone git://git.openwrt.org/openwrt.git
cd openwrt
vi feeds.conf
cat feeds.conf
src-git packages https://github.com/openwrt/packages.git
src-git oldpackages http://git.openwrt.org/packages.git
src-git management https://github.com/openwrt-management/packages.git
src-git routing https://github.com/openwrt-routing/packages.git
src-git telephony https://github.com/openwrt/telephony.git

./scripts/feeds update -a
./scripts/feeds install -a

make menuconfig
[select privoxy]

make
```

The one billion dollar question would be: why is this happening only to me? (That privoxy creates a lot of .new files).

I found out that even a ```make dirclean``` doesn't clean up everything (toolchaing and tools stuff, sometimes matters...)

chris5560, 2015-02-11:
Hi,

if I run "make package/privoxy/clean" the directory "privoxy-3.0.23-stable" is completely removed from build_dir/target...
Does there something stay at your side for whatever reason ?
Currently no other idea.  :-(

Christian

enigmagroup, 2015-02-11:
Hi,

> if I run "make package/privoxy/clean" the directory "privoxy-3.0.23-stable" is completely removed from build_dir/target...
> Does there something stay at your side for whatever reason ?

Now I ran "make package/privoxy/clean" too and recompiled it and not the whole openwrt dir. Tadaa, the problem persists. Here's my log: http://paste.scratchbook.ch/view/raw/9f1df3ee

Maybe this line just needs to be removed: https://github.com/openwrt/packages/blob/master/net/privoxy/Makefile#L69 ?

chris5560, 2015-02-12:
Please remove the whole block "define Build/Install ... endef"

Other option to try to rename your directory from "openwrt-64.mostrecent" to "openwrt-64_mostrecent".
I tried a package name with a dot and everything works fine but it did not install in the final build.
All without error in "V=99".

enigmagroup, 2015-02-12:
> Please remove the whole block "define Build/Install ... endef"

Ok. Would you incorporate that change into the Makefile here on GitHub? The reason I'm asking is, I have a buildbot and I'm not gonna manually patch Privoxy every time. The buildbot does a ```make dirclean``` and will always fail when this change is not done here in the official Privoxy Makefile.

Another option to me would be forking privoxy into my own repo under a different package name but argh. Sigh.

chris5560, 2015-02-12:
I will remove the whole special handling for updating from old privoxy config to uci config.
But I would like to do some more build testing including installing the final *.ipkg to my test boxes.
Please give me one ore two more days.

enigmagroup, 2015-02-12:
Of course, no hurry. Thank you so much!

closed 2015-02-16 by sbyx

enigmagroup, 2015-02-17:
Thank you very much! It all works, good job!

PR #910 (2015-02-16): privoxy: remove upgrade handling, conffile and initscript fixes
* remove upgrade handling for version 3.0.21 or older packages. Also fix #726
* added user.action, user.filter and user.trust files to conffiles (might be modified by user on existing system)
* move permission setting from Makefile (install) to initscript because permissions are overwritten during/inside buildroot
commit 87b9255c (2015-02-16): privoxy: remove upgrade handling, conffile and initscript fixes

* remove upgrade handling for version 3.0.21 or older packages. Also fix #726
* added user.action, user.filter and user.trust files to conffiles (might be modified by user on running system)
* move permission setting from Makefile (install) to initscript because permissions are overwritten during/inside buildroot