openwrt/packages #756: Python: failed to build _ssl + _socket module on x86_64

enigmagroup, 2015-01-08:
Somewere during the build of Python, I encounter the following errors:

```
building '_ssl' extension
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -I. -IInclude 
-I./Include -I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/include -I/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/x/op
enwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/usr/include/x86_64-linux-gnu -I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/x/ope
nwrt-64.mostrecent/staging_dir/host/bin/Include -I/home/x/openwrt-64.mostrecent/staging_dir/host/bin -c /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/_ssl.c -o build/temp.lin
ux2-x86_64-2.7/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/_ssl.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
In file included from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/kernel.h:4:0,
				 from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/netlink.h:4,
				 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.h:45,
				 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/_ssl.c:47:
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:8:2: error: unknown type name '__kernel_long_t'
  __kernel_long_t uptime;  /* Seconds since boot */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:9:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t loads[3]; /* 1, 5, and 15 minute load averages */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:10:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t totalram; /* Total usable main memory size */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:11:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t freeram; /* Available memory size */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:12:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t sharedram; /* Amount of shared memory */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:13:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t bufferram; /* Memory used by buffers */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:14:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t totalswap; /* Total swap space size */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:15:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t freeswap; /* swap space still available */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:18:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t totalhigh; /* Total high memory size */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:19:2: error: unknown type name '__kernel_ulong_t'
  __kernel_ulong_t freehigh; /* Available high memory size */
  ^
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:21:22: error: '__kernel_ulong_t' undeclared here (not in a function)
  char _f[20-2*sizeof(kernel_ulong_t)-sizeof(u32)]; /* Padding: libc5 uses this.. */
					  ^
```

[...]

```
Python build finished, but the necessary bits to build these modules were not found:
_tkinter           bsddb185           dl
imageop            linuxaudiodev      nis
ossaudiodev        sunaudiodev
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
_bsddb             _socket            _ssl
dbm
```

The packaging of Python itself continues.
I discovered this error on the running target system, when ```import urllib``` threw an error about the missing socket module.

commodo, 2015-01-08:
Some weird issues appeared after the ugprade to 2.7.9.
After I rolled back to 2.7.8 they were there too, which is weird.

I did not see the _socket & _ssl failures yet.
Will look into it and come back with some findings, or questions if I do not manage to reproduce.

enigmagroup, 2015-01-08:
Ok great, thank you!

commodo, 2015-01-11:
What's the base OS you're using, or the config of the host system you're running this on ?
Did you try with a clean build ?

I am not able to reproduce this yet.

enigmagroup, 2015-01-12:
I remember checking out freshly from sources, so it must be even more clean than a 'make dirclean'.

```# uname -a```
```Linux classified 3.2.0-4-amd64 #1 SMP Debian 3.2.60-1+deb7u3 x86_64 GNU/Linux```

I will do a build on another machine from source and see what happens.

commodo, 2015-01-12:
I managed to build Python on fresh Debian 7 VMs.
It worked on 32 bit Debian & 64 bit.

I was hunting some other weird issue with Python not building correctly.
On Debian Python was building fine.

On my OpenSUSE machines, there were some errors.

enigmagroup, 2015-01-12:
Compilation started. We will know more in ~2h.

enigmagroup, 2015-01-12:
Ok, no errors on my second workstation (debian jessie). Still the same error after fresh checkout on debian wheezy.

I have to investigate. Very strange error...
I think I close this issue and reopen it if there would be something to do on your side.

Thanks for your support so far!

enigmagroup, 2015-01-12:
I think I found the issue that is triggering the errors. Python indeed includes a directory on the \*host* machine.

```
building '_socket' extension
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -I. -IInclude -I./Include -I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/include -I/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/usr/include/x86_64-linux-gnu -I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-64.mostrecent/staging_dir/host/bin/Include -I/home/x/openwrt-64.mostrecent/staging_dir/host/bin -c /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c -o build/temp.linux2-x86_64-2.7/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.o
In file included from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:183:0:
/usr/include/x86_64-linux-gnu/sys/param.h:76:0: warning: "MAX" redefined [enabled by default]
 #define MAX(a,b) (((a)>(b))?(a):(b))
 ^
/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:102:0: note: this is the location of the previous definition
 #define MAX(x, y) ((x) < (y) ? (y) : (x))
 ^
cc1: note: someone does not honour COPTS correctly, passed 2 times
In file included from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/kernel.h:4:0,
                 from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/netlink.h:4,
                 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.h:45,
                 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:262:
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:8:2: error: unknown type name '__kernel_long_t'
  __kernel_long_t uptime;  /* Seconds since boot */
  ^
```

Reformatted for better visualisation:

```
building '_socket' extension
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable 
-I. 
-IInclude 
-I./Include 
-I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/include 
-I/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include 
-I/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include 

-I/usr/include/x86_64-linux-gnu    <-- this is the black sheep!

-I/home/x/openwrt-64.mostrecent/staging_dir/target-x86_64_uClibc-0.9.33.2/usr/include 
-I/home/x/openwrt-64.mostrecent/staging_dir/host/bin/Include 
-I/home/x/openwrt-64.mostrecent/staging_dir/host/bin
-c /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c -o build/temp.linux2-x86_64-2.7/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.o
In file included from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:183:0:
/usr/include/x86_64-linux-gnu/sys/param.h:76:0: warning: "MAX" redefined [enabled by default]
 #define MAX(a,b) (((a)>(b))?(a):(b))
 ^
/home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:102:0: note: this is the location of the previous definition
 #define MAX(x, y) ((x) < (y) ? (y) : (x))
 ^
cc1: note: someone does not honour COPTS correctly, passed 2 times
In file included from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/kernel.h:4:0,
                 from /home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/netlink.h:4,
                 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.h:45,
                 from /home/x/openwrt-64.mostrecent/build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/Modules/socketmodule.c:262:
/home/x/openwrt-64.mostrecent/staging_dir/toolchain-x86_64_gcc-4.8-linaro_uClibc-0.9.33.2/include/linux/sysinfo.h:8:2: error: unknown type name '__kernel_long_t'
  __kernel_long_t uptime;  /* Seconds since boot */
  ^
```

You can clearly see that it accesses files of there:

```/usr/include/x86_64-linux-gnu/sys/param.h:76:0: warning: "MAX" redefined```

I guess my "ancient" debian stable kernel doesn't have specific kernel params needed for building those python extensions; whereas on my debian testing workstation, the host system has a fairly recent kernel.

So - how to tell the Python build process \*not* to include host system directories?

commodo, 2015-01-13:
It's weird that this happens.
I'll look for a way to patch or configure the Python build to remove the host system's header includes.

I think already did something similar a while back for compilation of Python C extensions, but it seems I still need to do something a bit more.

Thanks for reporting

commodo, 2015-01-15:
Found the issue and created a patch for that. 
See PR https://github.com/openwrt/packages/pull/784
Seems the issue is only reproduce-able on Debian/Ubuntu, as the Python setup.py script tries to add multiarch paths and stuff, and when the OpenWRT x86_64 toolchain comes along it clashes with the host's.
No idea why this happens in the latest Python version; but I guess removing that for future versions is a good idea.

Thanks for reporting this.
I only managed to reproduce this just now, since I had to install a Debian Wheezy amd64 VM and dump all build logs to file and search through the logs.

The problem with failing build extensions is not new; not sure how to fix it yet, since Python does not stop when an extension (or more) fails to build.

Thanks

enigmagroup, 2015-01-15:
Ok thanks!
Gosh, I didn't know that I am using such an exotic operating system ;)

Thanks a lot for the investigation & your efforts.

commodo, 2015-01-16:
Well, I'm a Debian fan too :)
It's just that the HW it supports makes it a pain to use as a desktop OS with all the new HW coming out, so I have to use something else.
For my own server I use Debian tho.

And I guess the fact that Debian/Ubuntu is popular is why that multiarch support was added to the Python build script.
They just didn't think that some other people are cross-compiling Python a bit differently.

commodo, 2015-01-19:
@thess @hnyman @sbyx @enigmagroup   I think this issue can be closed. 
@enigmagroup : please confirm.
( I listed a few people with rights + the reporter of the issue )

enigmagroup, 2015-01-19:
Wonderful. Everything works!
Thanks a lot!

closed 2015-01-19 by enigmagroup