openwrt/packages #758: avahi-daemon: avahi-daemon exits immediately

ralemb, 2015-01-09:
As avahi-daemon moved to github: 
This problem described here: https://dev.openwrt.org/ticket/12971 still exists
avahi-daemon exits immediately, if dbus isn't installed and enable-dbus=yes is set.
But of course enable-dbus=yes is needed, if dbus is installed.

Imo the setting "enable-dbus=warn" would be the best solution.

thess, 2015-01-10:
The default package build (CONFIG_ALL) builds only the avahi-daemon with dbus support. However, the package does not force the install of dbus and hence the problem you see. The work-around you suggest (enable-dbus=warn) would solve the problem of avahi-daemon exiting when dbus is not present, but it is not really a dbus-free (libdbus dependency) version.

I think the best solution would be to have 2 variations of avahi-daemon and let one decide which version to install. Stay tuned...

ralemb, 2015-01-11:
Being a total noob I don't know, whether it's worth the hassle. [Avahi FAQ #6] (http://www.avahi.org/wiki/Avah4users):
`Blah, you fools, D-Bus is bloat and sucks!
Then Avahi is not for you, sorry. As it seems we think a little different about D-Bus than you do. `

In my use case I first installed mpd/upmpdcli. They don't need D-Bus with avahi.
Then I installed shairport. That talks to avahi per D-Bus,
Could opkg handle this situation with alternative avahi-variations?

I think due to the progress of Poettering's ideas in the Linux world there will be more and more applications relying on D-Bus.
OpenWRT developers are working on their own d-bus-free implementations of the protocols, if I get it right. But I don't know about the achieved progress.

(enable-dbus=warn) wouldn't raise any philosophical issues, just make the thing work out of the box.

thess, 2015-01-11:
>Could opkg handle this situation with alternative avahi-variations?

Yes, it can - I am close to having a version of avahi daemon with/without D-Bus.building properly along with modified build dependencies in a couple of other packages (e.g. shairport). If it doesn't work out I'll modify the default in the config file as a fall-back strategy and live with the added libdbus overhead.

For those of use who do custom builds, this will change things slightly due to the fact one will have to select the desired avahi-daemon variant if no other package selects it (mpd & openssh avahi service options for example)

thess, 2015-01-12:
Another random observation...

> Then I installed shairport. That talks to avahi per D-Bus,

Yes, it does -- And, if you install shairport, all of the correct dependencies (including dbus) are installed. So, the only real problem here is if you install the default avahi-daemon package alone and try to start it you will see this error.

ralemb, 2015-01-13:
Yes. But what is your point?
I installed shairport a week after mpd for testing purposes. And i'm still not sure whether I want to keep it when CC becomes stable. 

`opkg install avahi-daemon avahi-daemon-service-ssh openssh-sftp-server openssh-sftp-avahi-service` simply doesn't work. 
You have to examine logs, google the exact error, get one (1) hit (a 2 year old open ticket), have to edit avahi-daemon.conf. By chance after installing another package you have to remember and revert your edits. 
This could be improved.

thess, 2015-01-13:
None really.
I still see this as a build/package problem, but I have having difficulties getting two packages built (with/without dbus) in a single build (CONFIG_ALL).

To that end, I will probably take your suggested work-around and push a change to the default package later today. Thanks for your patience.

thess, 2015-01-20:
You might want to try out some new packages after today. There are now 2 complete versions of libavahi and avahi-daemon with and without dbus support. See #791.

thess, 2015-01-24:
Build looks good - closing this

closed 2015-01-24 by thess

ralemb, 2015-01-25:
Sorry, for me avahi-nodbus-daemon is not working oob.
avahi-daemon startup did not complete without ipv6 enabled.

What I did:
I removed avahi-daemon and dbus with dependencies.
I had to force-remove, hope I dit it right, but opkg list-installed and ps didn't show anything related after rebooting the device.
`opkg install avahi-nodbus-daemon avahi-daemon-service-ssh openssh-sftp-avahi-service mpd-avahi-service`

After reboot:
avahi-browse -ak from my workstation doesn't show anything.

After setiing `publish-workstation=yes` and restarting avahi-daemon 
`eth0 IPv4 wdmynet [xx:xx:xx:xx:xx:xx]                   _workstation._tcp` shows up.
After setiing `use-ipv6=yes` and restarting avahi-daemon 
`+   eth0 IPv4 wdmynet                                       _ssh._tcp` as well as the other services show up.
logread with `use-ipv6=no` (oob)
```
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Successfully dropped root privileges.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: avahi-daemon 0.6.31 starting up.
Sun Jan 25 16:20:42 2015 daemon.warn avahi-daemon[1587]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Loading service file /etc/avahi/services/mpd.service.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Loading service file /etc/avahi/services/sftp-ssh.service.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Loading service file /etc/avahi/services/ssh.service.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.0.33.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: New relevant interface br-lan.IPv4 for mDNS.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Network interface enumeration completed.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Registering new address record for fe80::290:a9ff:fece:88ad on wlan1.*.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Registering new address record for fe80::290:a9ff:fece:88ab on wlan0.*.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Registering new address record for 2002:5f5a:8bd9:1:290:a9ff:fece:88aa on br-lan.*.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Registering new address record for 192.168.0.33 on br-lan.IPv4.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Server startup complete. Host name is wdmynet.local. Local service cookie is 1923730000.
Sun Jan 25 16:20:42 2015 daemon.info avahi-daemon[1587]: Registering HINFO record with values 'MIPS'/'LINUX'.
```
logread with `use-ipv6=yes`
```
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Successfully dropped root privileges.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: avahi-daemon 0.6.31 starting up.
Sun Jan 25 16:50:09 2015 daemon.warn avahi-daemon[1373]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Loading service file /etc/avahi/services/mpd.service.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Loading service file /etc/avahi/services/sftp-ssh.service.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Loading service file /etc/avahi/services/ssh.service.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Joining mDNS multicast group on interface wlan1.IPv6 with address fe80::290:a9ff:fece:88ad.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: New relevant interface wlan1.IPv6 for mDNS.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Joining mDNS multicast group on interface wlan0.IPv6 with address fe80::290:a9ff:fece:88ab.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: New relevant interface wlan0.IPv6 for mDNS.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Joining mDNS multicast group on interface br-lan.IPv6 with address 2002:5f5a:8bd9:1:290:a9ff:fece:88aa.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: New relevant interface br-lan.IPv6 for mDNS.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.0.33.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: New relevant interface br-lan.IPv4 for mDNS.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Network interface enumeration completed.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Registering new address record for fe80::290:a9ff:fece:88ad on wlan1.*.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Registering new address record for fe80::290:a9ff:fece:88ab on wlan0.*.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Registering new address record for 2002:5f5a:8bd9:1:290:a9ff:fece:88aa on br-lan.*.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Registering new address record for 192.168.0.33 on br-lan.IPv4.
Sun Jan 25 16:50:09 2015 daemon.info avahi-daemon[1373]: Registering HINFO record with values 'MIPS'/'LINUX'.
Sun Jan 25 16:50:10 2015 daemon.info avahi-daemon[1373]: Server startup complete. Host name is wdmynet.local. Local service cookie is 1173573552.
Sun Jan 25 16:50:11 2015 daemon.info avahi-daemon[1373]: Service "wdmynet" (/etc/avahi/services/ssh.service) successfully established.
Sun Jan 25 16:50:11 2015 daemon.info avahi-daemon[1373]: Service "wdmynet" (/etc/avahi/services/sftp-ssh.service) successfully established.
Sun Jan 25 16:50:11 2015 daemon.info avahi-daemon[1373]: Service "wdmynet" (/etc/avahi/services/mpd.service) successfully established.
```

ralemb, 2015-01-25:
Sorry, imo opkg cannot handle the package switching for a noob (aka me).
You cannot install shairport if avahi-nodbus-daemon is installed. Even worse opkg claims shairport installed after a second attempt to install.
What I did:
Removed avahi-nodbus-daemon etc and rebooted.
Installed avahi-daemon-service-ssh then shairport. opkg notices the failure of shairport but oversees the failure of libavahi-client. 
Repeated opkg install shairport and got no error.
```
root@wdmynet:~# opkg install avahi-daemon-service-ssh
Installing avahi-daemon-service-ssh (0.6.31-10) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/avahi-daemon-service-ssh_0.6.31-10_ar71xx.ipk.
Installing avahi-nodbus-daemon (0.6.31-10) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/avahi-nodbus-daemon_0.6.31-10_ar71xx.ipk.
Installing libavahi-nodbus-support (0.6.31-10) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/libavahi-nodbus-support_0.6.31-10_ar71xx.ipk.
Installing libdaemon (0.14-5) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/libdaemon_0.14-5_ar71xx.ipk.
Configuring libavahi-nodbus-support.
Configuring libdaemon.
Configuring avahi-nodbus-daemon.
Configuring avahi-daemon-service-ssh.
```
```
root@wdmynet:~# opkg install shairport
Installing shairport (2014-10-28) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/shairport_2014-10-28_ar71xx.ipk.
Installing libavahi-client (0.6.31-10) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/libavahi-client_0.6.31-10_ar71xx.ipk.
Installing avahi-dbus-daemon (0.6.31-10) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/avahi-dbus-daemon_0.6.31-10_ar71xx.ipk.
Configuring libavahi-client.
Collected errors:
 * check_data_file_clashes: Package avahi-dbus-daemon wants to install file /etc/init.d/avahi-daemon
	But that file is already provided by package  * avahi-nodbus-daemon
 * check_data_file_clashes: Package avahi-dbus-daemon wants to install file /usr/sbin/avahi-daemon
	But that file is already provided by package  * avahi-nodbus-daemon
 * opkg_install_cmd: Cannot install package shairport.
root@wdmynet:~# opkg install shairport
Installing shairport (2014-10-28) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/packages/shairport_2014-10-28_ar71xx.ipk.
Configuring shairport.
```
```
root@wdmynet:~# opkg list-installed | grep shairport
shairport - 2014-10-28
root@wdmynet:~# opkg list-installed | grep avahi
avahi-daemon-service-ssh - 0.6.31-10
avahi-nodbus-daemon - 0.6.31-10
libavahi-client - 0.6.31-10
libavahi-nodbus-support - 0.6.31-10
```

probonopd, 2015-01-25:
Slightly OT, if you are just interested in shairport without the need for avahi, then you might be interested in https://github.com/probonopd/shairport-sync-for-openwrt - I have been considering to suggest it for the OpenWrt packages feed. I think avahi is a lot of overhead in many cases.

ralemb, 2015-01-25:
@probonopd: I like avahi. I like thunar or caja seeing availlable shares, cups seeing printers, pulseaudio seeing sinks and ping finding seldom_used_machine.local.
OT: +1 for inclusion in OpenWrt packages feed.
I saw your work before but just learning OpenWrt I was not sure how to include. 
Adding a git-src line in opkg.conf? 
Just more OT: Inserting samples would break "bitperfect" though this probably doesn't matter here.

thess, 2015-01-25:
@probonopd - please do not hijack this thread.

@ralemp -- try removing ALL avahi related packages and then install shairport. You should be good to go from there. Installing libavahi-client or avahi-utils should also install the dbus version.

There seems to be no way to handle generic dependencies within opkg and it appears that when you install any of the avahi service configs, it installs the nodbus version by default (which could be all you wanted). I'll see if there is a way to get a better warning when you need to install the other variant.

thess, 2015-01-25:
I cannot seem to reproduce your issue with the nodbus daemon or use-ipv6=no. If you are running on an ipv6 network then yes, you will have to enable it.

ralemb, 2015-01-26:
Please excuse my funny English  I'm german.
I apologize for not having had time to test commit #791 in time.

I have no problem starting avahi-daemon on my other ipv6-enabled machines with use-ipv6=no. 
use-ipv6=yes/no should determine whether ipv6 is used for advertising, not hang avahi-daemon.
But this may be related to the fact that my device is a dumb access point with eth0/eth1/wlan0/wlan1 bridged. Strange enough wlan0/wlan1 get a link local ipv6 address. I can give more details if you want but I don't want to overuse your patience.

opkg seems quite limited. I feel it pointless to create package dependencies the package manager can't resolve. 
As far as I'm concerned: If i have to or want to install sth avahi related I first of all do `opkg install avahi-daemon`. This pulls avahi-dbus-daemon and opkg stays happy.