openwrt/packages #767: python 2.7.3 fails to build on MAC OS X / BSD

plntyk, 2015-01-11:
there was a patch but it was not applied for unknown reasons

forum report: https://forum.openwrt.org/viewtopic.php?id=50205
prev patchwork entry 3424

Since this is the second time this issue popped up maybe applying this makes sense?
I don't have MacOS X so I cannot test this

```
Hi,

this small fix allows cross-compile of python on BSD hosts.

Patch

Index: lang/python/Makefile
===================================================================
--- lang/python/Makefile    (revision 35899)
+++ lang/python/Makefile    (working copy)
@@ -9,7 +9,7 @@ 
 
 PKG_NAME:=python
 PKG_VERSION:=2.7.3
-PKG_RELEASE:=2
+PKG_RELEASE:=3
 
 PKG_SOURCE:=Python-$(PKG_VERSION).tar.xz
 PKG_SOURCE_URL:=http://www.python.org/ftp/python/$(PKG_VERSION)
@@ -162,6 +162,7 @@ 
         --with-threads \
         --with-system-ffi="$(STAGING_DIR)/usr" \
         $(ENABLE_IPV6) \
+        MACHDEP=linux2 \
         ac_cv_have_chflags=no \
         ac_cv_have_lchflags=no \
         ac_cv_py_format_size_t=no \
```

commodo, 2015-01-11:
Did you also reproduce this with Python 2.7.9 ? 
I can apply this, but the version of Python has changed since the patch, and a lot of things in Python have made it more cross-compiler friendly.

commodo, 2015-01-15:
@plntyk  ping 

I applied the change (to add MACHDEP=linux2) .
I'll have to re-visit this sometime in the next couple of days.

For some reason, adding that argument, disabled the creation of the `libpython.so` file which is weird.
Maybe I did something wrong and was too tired to notice what I did, so I'll have to dig deeper into it.

Thanks

plntyk, 2015-01-15:
I just reported the issue from the forum here - but
I am atm struggling with setting up BSD and testing the build there (several traps there)

Short summary:
- I do not have a MacOS X machine nor access to one where this issue was last seen (forum, 2014)
__So I cannot say anything about MacOSX // looking up the m4/autoconf internals will take too much time (and I am not sure I can even understand these)__

- PC-BSD 10.1 throws several errors in tools (similar to FreeBSD) atm I stopped testing there (frustrating ^^)
- __FreeBSD 9.3 compiles python 2.7.9 without errors__ (has same tools issue)

however atm there are 4 "patches" to get compile on FreeBSD 9.3 to packages stage:
- revert r37183 (I dont have any idea about side effects on Linux compilation or why there are these headers in the first place) to make tools/xz compile
- add "--enable-iconv=no --with-libiconv=gnu" and "usr/local/{include,lib} -liconv" LD/CFLAGS to tools/pkg-config makefile to make 
- tools/mpfr needs for gcc49 : https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=193340
- prevent tools/mkimage (unresolved) - I cannot even get tools from vanilla u-boot git tree built in FreeBSD either (-nostdinc is used and this prevents including a headerfile thats "not found" sys/_types.h)

So from preliminary test this does not seem to be an issue with BSD anymore. (only speaking for FreeBSD) 
If there are side effects of using the patch from Waldemar with Linux which is arguably the main build platform then just stash this away - until some developer with a MacOS X is coming around to test/debug/review

commodo, 2015-01-19:
So, I decided to test this again and pay a bit more attention (in case I didn't before).
Adding `MACHDEP=linux2` seems to disable the creation of `libpython.so` and builds a monolithic python executable.
Still no idea why.
If this would not happen, I'd have no objection to adding this patch.

If you say that it works on FreeBSD, then I'd like to take the lazy way and not try to fix/investigate this further.
Just to clarify: the build tools issue you're having is not related to Python, right ?

I'll keep the patch for later consideration :)
Thanks for pointing it out.

Python differs a lot from 2.7.3 to 2.7.9 build-wise, in the sense that the cross-compilation is a bit better.

commodo, 2015-02-13:
@plntyk 
Any reason to keep this issue open ?

I'd vote for closing it now, and see if it comes up again.
Thoughts ?

closed 2015-03-11 by champtar