openwrt/packages #769: davfs2 breaks in trunk

danielpetre, 2015-01-11:
Hello, seems davfs breaks in trunk 43930 with:

make[6]: Entering directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/src'
mips-openwrt-linux-uclibc-gcc -Wall -Werror=format-security -fstack-protector --param=ssp-buffer-size=4 -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -I/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include/neon -L/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib -o mount.davfs cache.o dav_coda.o dav_fuse.o kernel_interface.o mount_davfs.o webdav.o -L/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib -lcrypto -lssl -lneon -lz -L/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -lssl -lcrypto -lexpat
/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/mips-openwrt-linux-uclibc/4.8.3/../../../../mips-openwrt-linux-uclibc/bin/ld: cannot find -lssp_nonshared
/home/dani/chaos_calmer/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/mips-openwrt-linux-uclibc/4.8.3/../../../../mips-openwrt-linux-uclibc/bin/ld: cannot find -lssp
cache.o: In function `set_cache_file_times':
cache.c:(.text+0x7ec): warning: the use of OBSOLESCENT utime' is discouraged, use utimes'
/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libcrypto.so: warning: gethostbyname is obsolescent, use getnameinfo() instead.
collect2: error: ld returned 1 exit status
Makefile:329: recipe for target 'mount.davfs' failed
make[6]: * [mount.davfs] Error 1
make[6]: Leaving directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/src'
make[6]: Entering directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
make[6]: Leaving directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Makefile:387: recipe for target 'all-recursive' failed
make[5]: * [all-recursive] Error 1
make[5]: Leaving directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Makefile:306: recipe for target 'all' failed
make[4]: * [all] Error 2
make[4]: Leaving directory '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2'
Makefile:67: recipe for target '/home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/.built' failed
make[3]: * home/dani/chaos_calmer/build_dir/target-mips_34kc_uClibc-0.9.33.2/davfs2-1.5.2/.built Error 2
make[3]: Leaving directory '/home/dani/chaos_calmer/feeds/packages/net/davfs2'
package/Makefile:174: recipe for target 'package/feeds/packages/davfs2/compile' failed
make[2]: * [package/feeds/packages/davfs2/compile] Error 2
make[2]: Leaving directory '/home/dani/chaos_calmer'
package/Makefile:171: recipe for target '/home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: * home/dani/chaos_calmer/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile Error 2
make[1]: Leaving directory '/home/dani/chaos_calmer'
/home/dani/chaos_calmer/include/toplevel.mk:173: recipe for target 'world' failed
make: * [world] Error 2

plntyk, 2015-01-15:
- the buildbot has errors with this package too (example: ar71xx)
- the pull request fixed the error at my end

champtar, 2015-01-20:
buildbot seems broken since before the fix
closing this for now

champtar, 2015-01-20:
http://buildbot.openwrt.org:8010/builders/ar71xx

closed 2015-01-20 by champtar