openwrt/packages #777: libmodbus _sometimes_ fails to compile (aclocal-1.14 missing)

karlp, 2015-01-14:
I get this occasionally after doing "make clean && make -j8 V=s"

Note that this can always be fixed by "make package/libmodbus/{clean,compile} V=s"
and then rerunning "make -j8 V=s"

```
... more config snipped, entire log can be provided....
config.status: tests/unit-test.h is unchanged
config.status: executing depfiles commands
config.status: executing libtool commands
configure: WARNING: unrecognized options: --disable-nls
rm -f /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/.configured_
touch /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/.configured_
CFLAGS="-Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -fpic  -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -fpic  -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/include " LDFLAGS="-L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/usr/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/target-mips_mips32_uClibc-0.9.33.2/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/karlp/src/smartgate_firmware/openwrt-bb/staging_dir/toolchain-mips_mips32_gcc-4.8-linaro_uClibc-0.9.33.2/lib " make -j1 -C /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/. AR=mips-openwrt-linux-uclibc-ar AS="ccache_cc -c -Os -pipe -mno-branch-likely -mips32 -mtune=mips32 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -fpic" LD=mips-openwrt-linux-uclibc-ld NM=mips-openwrt-linux-uclibc-nm CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB=mips-openwrt-linux-uclibc-ranlib STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CROSS="mips-openwrt-linux-uclibc-" ARCH="mips" ;
make[4]: Entering directory `/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6'
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/sh /home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/missing aclocal-1.14 -I m4
/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/missing: line 81: aclocal-1.14: command not found
WARNING: 'aclocal-1.14' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make[4]: *** [aclocal.m4] Error 127
make[4]: Leaving directory `/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6'
make[3]: *** [/home/karlp/src/smartgate_firmware/openwrt-bb/build_dir/target-mips_mips32_uClibc-0.9.33.2/libmodbus-3.0.6/.built] Error 2
make[3]: Leaving directory `/home/karlp/src/smartgate_firmware/openwrt-bb/feeds/packages/libs/libmodbus'
```

champtar, 2015-03-11:
@karlp, is it still an issue?

mhei, 2015-03-12:
According to the directory naming, @karlp is using BB, right? I did not notice any commit which should improve/fix this issue, so I guess its still there?! Should be care about this, or should we focus on CC?

karlp, 2015-03-12:
I would presume still an issue, but I wouldn't focus on it heavily, it was more filed because I didn't know what was causing it, and I was hoping it might ring a bell for someone else :smile:  I haven't been building much the last few weeks, but will be again shortly.  I've seen the same failure with other packages occasionally, but most often with libmodbus.  (yes, this was with BB)

champtar, 2015-03-16:
as this is with BB, and it's not blocking, i'm closing this issue.
If it persist with CC, i think it's better to take this to dev mailing-list

closed 2015-03-16 by champtar