openwrt/packages #780: openconnect NO_DEVICE error after "ifup wan"

xuefer, 2015-01-15:
```
ifup wan
ifup moooc1
```

after everything works

```
ifup wan
# ifstatus moooc1
{
        "up": false,
        "pending": false,
        "available": false,
        "autostart": true,
        "proto": "openconnect",
        "data": {

        }
}

# ifup moooc1
# ifstatus moooc1
{
        "up": false,
        "pending": false,
        "available": false,
        "autostart": true,
        "proto": "openconnect",
        "data": {

        },
        "errors": [
                {
                        "subsystem": "interface",
                        "code": "NO_DEVICE"
                }
        ]
}
```
moooc1 interface can no longer be bring up unless i change the name to something else in /etc/config/network

nmav, 2015-01-17:
The best is to check your log for any issues preventing openconnect from going up.
It could be related to the newly applied fix at: https://github.com/openwrt/packages/pull/790

xuefer, 2015-01-17:
```
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' has lost the connection
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Network device 'vpn-moooc3' link is down
Sat Jan 17 18:52:38 2015 user.notice openconnect: bringing down openconnect
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now down
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is setting up now
Sat Jan 17 18:52:38 2015 user.notice openconnect: initializing...
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Terminating on signal 15
Sat Jan 17 18:52:38 2015 user.notice openconnect: executing 'openconnect npv.ladio.me:443 -i vpn-moooc3 --non-inter --syslog --script /lib/netifd/vpnc-script  --servercert=sha1:e0de8dd5e38c60fbbe223d644391d0caca283d5f --no-system-trust -u xuefer --passwd-on-stdin'
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Connect time 1762.3 minutes.
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Sent 3512652818 bytes, received 2523267735 bytes.
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Network device 'pppoe-wan' link is down
Sat Jan 17 18:52:38 2015 daemon.err miniupnpd[5182]: ioctl(s, SIOCGIFADDR, ...): Cannot assign requested address
Sat Jan 17 18:52:38 2015 daemon.err miniupnpd[5182]: Failed to get IP for interface pppoe-wan
Sat Jan 17 18:52:38 2015 daemon.warn miniupnpd[5182]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Sat Jan 17 18:52:38 2015 daemon.notice pppd[3722]: Connection terminated.
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Connect time 1762.3 minutes.
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Sent 3512656269 bytes, received 2523271067 bytes.
Sat Jan 17 18:52:38 2015 daemon.info pppd[3722]: Exit.
Sat Jan 17 18:52:38 2015 daemon.info openconnect[5998]: POST https://npv.ladio.me/
Sat Jan 17 18:52:38 2015 daemon.info openconnect[5998]: Attempting to connect to server 50.116.8.145:443
Sat Jan 17 18:52:38 2015 daemon.notice openconnect[5998]: Failed to connect to host npv.ladio.me
Sat Jan 17 18:52:38 2015 daemon.notice openconnect[5998]: Failed to open HTTPS connection to npv.ladio.me
Sat Jan 17 18:52:38 2015 daemon.notice netifd: moooc3 (5994): Failed to obtain WebVPN cookie
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'wan' is now down
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'wan' is setting up now
Sat Jan 17 18:52:38 2015 user.notice openconnect: bringing down openconnect
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now down
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is setting up now
Sat Jan 17 18:52:38 2015 daemon.info pppd[6057]: Plugin rp-pppoe.so loaded.
Sat Jan 17 18:52:38 2015 daemon.info pppd[6057]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.7
Sat Jan 17 18:52:38 2015 user.notice openconnect: initializing...
Sat Jan 17 18:52:38 2015 user.notice openconnect: bringing down openconnect
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now down
Sat Jan 17 18:52:38 2015 daemon.notice pppd[6057]: pppd 2.4.7 started by root, uid 0
Sat Jan 17 18:52:38 2015 daemon.info pppd[6057]: PPP session is 34546
Sat Jan 17 18:52:38 2015 daemon.warn pppd[6057]: Connected to 76:91:2b:0d:00:02 via interface eth1
Sat Jan 17 18:52:38 2015 daemon.info pppd[6057]: Using interface pppoe-wan
Sat Jan 17 18:52:38 2015 daemon.notice pppd[6057]: Connect: pppoe-wan <--> eth1
Sat Jan 17 18:52:38 2015 daemon.info pppd[6057]: Remote message: Authentication success,Welcome!
Sat Jan 17 18:52:38 2015 daemon.notice pppd[6057]: PAP authentication succeeded
Sat Jan 17 18:52:38 2015 daemon.notice pppd[6057]: peer from calling number 76:91:2B:0D:00:02 authorized
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: local  LL address fe80::a407:8459:2c5d:883b
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: remote LL address fe80::7691:2bff:fe0d:0002
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: local  IP address 113.87.215.249
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: remote IP address 113.87.212.1
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: primary   DNS address 202.96.134.133
Sat Jan 17 18:52:39 2015 daemon.notice pppd[6057]: secondary DNS address 202.96.128.166
Sat Jan 17 18:52:39 2015 daemon.notice netifd: Network device 'pppoe-wan' link is up
Sat Jan 17 18:52:39 2015 daemon.notice netifd: Interface 'wan' is now up
Sat Jan 17 18:52:39 2015 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Sat Jan 17 18:52:40 2015 user.notice ddns-scripts[6592]: myddns: PID '6592' started at 2015-01-17 18:52
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: Listen normally on 15 pppoe-wan 113.87.215.249 UDP 123
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: Listen normally on 16 pppoe-wan fe80::a407:8459:2c5d:883b UDP 123
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: Deleting interface #14 vpn-moooc3, 192.168.99.111#123, interface stats: received=0, sent=0, dropped=0, active_time=39792 secs
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: Deleting interface #10 pppoe-wan, fe80::551d:a71b:636c:fba4#123, interface stats: received=0, sent=0, dropped=0, active_time=105733 secs
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: Deleting interface #8 pppoe-wan, 183.49.116.66#123, interface stats: received=441, sent=538, dropped=0, active_time=105733 secs
Sat Jan 17 18:52:41 2015 daemon.info ntpd[3898]: 202.112.10.36 interface 183.49.116.66 -> (none)
[... cut ...]
Sat Jan 17 18:52:41 2015 daemon.debug ntpd[3898]: new interface(s) found: waking up resolver
Sat Jan 17 18:52:41 2015 user.warn ddns-scripts[6592]: myddns: Service section disabled! - TERMINATE
Sat Jan 17 18:52:41 2015 user.warn ddns-scripts[6592]: myddns: PID '6592' exit WITH ERROR '1' at 2015-01-17 18:52\n
Sat Jan 17 18:52:55 2015 daemon.notice vnstatd[4669]: Traffic rate for "pppoe-wan" higher than set maximum 100 Mbit (60->825, r17592186037915 t746), syncing.
```

i tried multiple `ifup moooc3`

nmav, 2015-01-17:
The log is pretty short. The openconnect connection failed because at the moment ppp was down as you see in the log. I'd expect netifd to restart openconnect at some point later. Isn't that the case?

xuefer, 2015-01-18:
```
Sat Jan 17 18:52:38 2015 daemon.info openconnect[5998]: POST https://npv.ladio.me/
Sat Jan 17 18:52:38 2015 daemon.info openconnect[5998]: Attempting to connect to server 50.116.8.145:443
```
this is a reconnect, i'm not sure if it is reconnected by openconnect itself or by netifd

but anyway, openconnect process is down finally and can isn't started anymore. not until reboot or change network name of the vpn. there must be some state that persistent the failure but not affecting a new (after rename) interface name

nmav, 2015-01-18:
If I kill openconnect manually in my system (14.07) I see the following from netifd:
Sun Jan 18 08:31:08 2015 daemon.notice netifd: Network device 'vpn-AVPN' link is down
Sun Jan 18 08:31:08 2015 daemon.notice netifd: Interface 'AVPN' has lost the connection
Sun Jan 18 08:31:08 2015 daemon.notice netifd: Interface 'AVPN' is now down
Sun Jan 18 08:31:08 2015 daemon.notice netifd: Interface 'AVPN' is setting up now
Sun Jan 18 08:31:12 2015 daemon.notice netifd: Interface 'AVPN' is now up
Sun Jan 18 08:31:12 2015 daemon.notice netifd: Network device 'vpn-AVPN' link is up

So netifd normally tries to reconnect openconnect. In your logs I see some strangeness:
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now down
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is setting up now
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now down

So I bring down wan, kill openconnect and bring it up again:
# ifdown wan
# kill openconnect pid
# ifup wan

netifd doesn't bring the openconnect connection up again. It doesn't even detect (log) that my AVPN connection is down.

This does seem to be an issue in netifd.

nmav, 2015-01-18:
@nbd168 feel free to reassign back to openconnect, if that is not a netifd issue.

xuefer, 2015-01-18:
> So netifd normally tries to reconnect openconnect. In your logs I see
some strangeness:
```
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now
down
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is
setting up now
Sat Jan 17 18:52:38 2015 daemon.notice netifd: Interface 'moooc3' is now
down
```

sorry, please ignore this chunk. because i was trying to use hotplug.d
script to ifdown moooc3 when wan is down and ifup it again when wan is up.
the problem, as you reproduced, happen before i have this script.

champtar, 2015-03-11:
@xuefer, is it still an issue?

champtar, 2015-07-27:
feel free to reopen if it's still an issue

closed 2015-07-27 by champtar