openwrt/packages #782: impossible to install asterisk11-res-timing-dahdi

andr2k, 2015-01-15:
Hardware : tp-ling 1043nd v1
OpenWrt version: trunk 2015-01-15 09:56:51 UTC

Something is wrong with package dependencies.

When I try to install asterisk11-res-timing-dahdi, I get an error:

root@OpenWrt:~# opkg install asterisk11-res-timing-dahdi
Installing asterisk11-res-timing-dahdi (11.15.0-2) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/ar71xx/packages/telephony/asterisk11-res-timing-dahdi_11.15.0-2_ar71xx.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for asterisk11-res-timing-dahdi:
 * 	kmod-dahdi-linux * 
 * opkg_install_cmd: Cannot install package asterisk11-res-timing-dahdi.

I suppose, kmod-dahdi should be used instead.

sbyx, 2015-01-16:
Pleae head over to https://github.com/openwrt/telephony
We are not in charge here ;)

closed 2015-01-16 by sbyx

jslachta, 2015-01-16:
This should be fixed. :-)