openwrt/packages #795: cifs-utils ?

ambrosa, 2015-01-18:
Chaos Calmer trunk@43982
arch. Lantiq , target device TP-LINK W8970 (LANTIQ->XRX200)

I'm sorry. This is not a real 'issue' but I've looked for cifs-utils (former cifsmount) without found it anywhere.
I need to mount into my OpenWRT a remote Samba share.
I've enabled in my compile the kmod-fs-cifs and Busybox has BUSYBOX_CONFIG_FEATURE_MOUNT_CIFS = y

But I cannot mount any remote samba share.
mount -t cifs //IP/SHARE /mnt/cifs /mnt/cifs -o username=...,password=...  fails
Perhaps do I need the cifs-utils (former cifsmount) ? But it's not anymore in repo.

I'm a bit confused....

ambrosa, 2015-01-18:
Forget:  
smbclient -U ambrosa \\\\192.168.1.70\\ambrosa\\ pass
works fine.
So it's not a problem about CIFS protocol (and kernel module).

There is something strange with mount ....

hnyman, 2015-01-23:
cifs-utils is apparently still in the "oldpackages" feed: https://dev.openwrt.org/browser/packages/net/cifs-utils/Makefile

But you shouldn't really need it.

However, the needed cifs mount command options depend on the kernel version. I noticed that two years ago...
https://forum.openwrt.org/viewtopic.php?pid=202745#p202745
Since kernel 3.5 you need "unc" option
Since kernel 3.8 you need also "sec" option.
````
mount -t cifs //192.168.1.199/ftp-nas /mnt -o username=admin,password=pwww,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.199\\ftp-nas
````
You might test with additional options.
(I haven't really checked this recently, so I am not quite sure about kernel 3.14+ )

luizluca, 2015-01-23:
cifs-utils is the "correct way" to mount it. On a normal system, it even blame you from trying to use cifs directly. Passing all cifs options "by hand" is too hardcore.

However, the easier way to retrieve all cifs arguments is to use strace in a similar system (specially kernel version).

strace -s 9999 mount ...

Check for the mount syscall.

ambrosa, 2015-01-25:
Thanks for reply.
My kernel is 3.14.28

Some days ago I've found the official description about Busybox mount and cifs:
```
mount -t cifs <dummy argument=whatever you want> <mountpoint> -o user=<username>,password=<password>,unc=<path of the remote share>
```
more or less as suggested.

I've made many tests without ANY SUCCESS. 
The same commands below work fine into an Ubuntu PC
```
root@openwrt:/# ls -la /mnt
drwxr-xr-x    1 root     root          4096 Jan 25 16:54 .
drwxr-xr-x    1 root     root          4096 Jan 17 15:21 ..
drwxrwxrwx    2 root     root          4096 Jan 18 07:36 cifs

root@openwrt:~# mount -t cifs "//192.168.1.70/VIDEO/" /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc="//192.168.1.70/VIDEO/"
mount: mounting //192.168.1.70/VIDEO/ on /mnt/cifs failed: Invalid argument
root@openwrt:~# mount -t cifs \\\\192.168.1.70\\VIDEO\\ /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.70\\VIDEO\\
mount: mounting \\192.168.1.70\VIDEO\ on /mnt/cifs failed: Invalid argument
root@openwrt:~# mount -t cifs dummy /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.70\\VIDEO\\
mount: mounting dummy on /mnt/cifs failed: Invalid argument
```

Using strace as suggested above (thanks luizluca) show the same problem: mount has an invalid argument

```
root@openwrt:~# strace -s 9999 mount -t cifs "//192.168.1.70/VIDEO/" /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc="//192.168.1.70/VIDEO/"
execve("/bin/mount", ["mount", "-t", "cifs", "//192.168.1.70/VIDEO/", "/mnt/cifs", "-o", "username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=//192.168.1.70/VIDEO/"], [/* 11 vars */]) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eaf000
stat("/etc/ld.so.cache", 0x7ff206e0)    = -1 ENOENT (No such file or directory)
open("/lib/libcrypt.so.0", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=9296, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eae000
mmap(NULL, 147456, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77e76000
mmap(0x77e76000, 9228, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77e76000
mmap(0x77e88000, 1104, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0x77e88000
mmap(0x77e89000, 67280, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77e89000
close(3)                                = 0
munmap(0x77eae000, 4096)                = 0
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=89724, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eae000
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77e50000
mmap(0x77e50000, 89240, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77e50000
mmap(0x77e75000, 3708, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x77e75000
close(3)                                = 0
munmap(0x77eae000, 4096)                = 0
open("/lib/libgcc_s.so.1", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=87555, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eae000
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77e2a000
mmap(0x77e2a000, 87172, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77e2a000
mmap(0x77e4f000, 1539, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x77e4f000
close(3)                                = 0
munmap(0x77eae000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eae000
mmap(NULL, 442368, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77dbe000
mmap(0x77dbe000, 351072, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77dbe000
mmap(0x77e23000, 7332, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x55000) = 0x77e23000
mmap(0x77e25000, 17492, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77e25000
close(3)                                = 0
munmap(0x77eae000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
[... 4 lines cut ...]
close(3)                                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=28968, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77eae000
set_thread_area(0x77eb5440)             = 0
mprotect(0x77e23000, 4096, PROT_READ)   = 0
mprotect(0x77eb0000, 4096, PROT_READ)   = 0
ioctl(0, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
getuid()                                = 0
brk(0)                                  = 0xb01000
brk(0xb02000)                           = 0xb02000
getuid()                                = 0
geteuid()                               = 0
brk(0xb03000)                           = 0xb03000
mount("//192.168.1.70/VIDEO/", "/mnt/cifs", "cifs", MS_SILENT, "username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=//192.168.1.70/VIDEO/,unc=\\\\192.168.1.70\\VIDEO,prefixpath=,ip=192.168.1.70") = -1 EINVAL (Invalid argument)
clone(child_stack=0x7ff206e8, flags=CLONE_VM|CLONE_VFORK|SIGCHLD) = 3310
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=3310, si_status=111, si_utime=0, si_stime=0} ---
waitpid(3310, NULL, 0)                  = 3310
write(2, "mount: mounting //192.168.1.70/VIDEO/ on /mnt/cifs failed: Invalid argument\n", 76mount: mounting //192.168.1.70/VIDEO/ on /mnt/cifs failed: Invalid argument
) = 76
exit_group(-1)                          = ?
+++ exited with 255 +++

```
Any idea ?

luizluca, 2015-01-25:
You are missing the ip option. It is important. Try with all options ubuntu used.

ambrosa, 2015-01-26:
IP option ??

*ip=arg*
*sets the destination IP address. This option is set automatically if the server name portion of the requested UNC name can be resolved so rarely needs to be specified by the user.*

Never used before but nothing change:
```
mount -t cifs \\\\192.168.1.70\\VIDEO\\ /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\
\192.168.1.70\\VIDEO\\
mount: mounting \\192.168.1.70\VIDEO\ on /mnt/cifs failed: Invalid argument
```
The SAME COMMAND, write as-is in an Ubuntu machine, works (but there is not Busybox but standard mount.cifs).

ambrosa, 2015-01-26:
Strange. if I remove ending \\ from the first  \\\\192.168.1.70\\VIDEO\\
errore message change: from "invalid argument" to "No such file or directory"
But share is correct.
```
root@openwrt:~# mount -t cifs \\\\192.168.1.70\\VIDEO /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\\1
92.168.1.70\\VIDEO\\
mount: mounting \\192.168.1.70\VIDEO on /mnt/cifs failed: No such file or directory
```

luizluca, 2015-01-26:
Try a strace inside openwrt to see if it is really passing your args just like ubuntu does. The backslashes might be the problem.

ambrosa, 2015-01-26:
The strace posted by me some msg above is done inside openwrt.
I post again a new strace made inside openwrt (I've only omitted some unuseful data).
Bu I don't see anything strange. But I'm not a developer :-(
```
root@openwrt:~# strace -s 9999 mount -t cifs \\\\192.168.1.70\\VIDEO /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,ip=192.168
.1.70,unc=\\\\192.168.1.70\\VIDEO
execve("/bin/mount", ["mount", "-t", "cifs", "\\\\192.168.1.70\\VIDEO", "/mnt/cifs", "-o", "username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\\192.168.1.70\\VIDEO"], [/* 11 vars */]) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728b000
stat("/etc/ld.so.cache", 0x7fdee5f0)    = -1 ENOENT (No such file or directory)
open("/lib/libcrypt.so.0", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=9296, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728a000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\4\0\0\0\0004
(...omissis...)
\0\0\17x\200%\316\3\0", 4096) = 4096
mmap(NULL, 147456, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77252000
mmap(0x77252000, 9228, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77252000
mmap(0x77264000, 1104, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0x77264000
mmap(0x77265000, 67280, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77265000
close(3)                                = 0
munmap(0x7728a000, 4096)                = 0
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=89724, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728a000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\22\320\0\0\0004
(...omissis...)
\0log\0log2\0log10\0pow\0remainder\0drem\0scalb\0sqrt", 4096) = 4096
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7722c000
mmap(0x7722c000, 89240, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x7722c000
mmap(0x77251000, 3708, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x77251000
close(3)                                = 0
munmap(0x7728a000, 4096)                = 0
open("/lib/libgcc_s.so.1", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=87555, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728a000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0006\260\0\0\0004
(...omissis...)
\1\354\0\0<\20\0\0\0(\22\0\0\n", 4096) = 4096
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77206000
mmap(0x77206000, 87172, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77206000
mmap(0x7722b000, 1539, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x7722b000
close(3)                                = 0
munmap(0x7728a000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728a000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\234p\0\0\0004
(...omissi...)
\3\274", 4096) = 4096
mmap(NULL, 442368, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7719a000
mmap(0x7719a000, 351072, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x7719a000
mmap(0x771ff000, 7332, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x55000) = 0x771ff000
mmap(0x77201000, 17492, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x77201000
close(3)                                = 0
munmap(0x7728a000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=28968, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x7728a000
set_thread_area(0x77291440)             = 0
mprotect(0x771ff000, 4096, PROT_READ)   = 0
mprotect(0x7728c000, 4096, PROT_READ)   = 0
ioctl(0, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
getuid()                                = 0
brk(0)                                  = 0x782000
brk(0x783000)                           = 0x783000
getuid()                                = 0
geteuid()                               = 0
brk(0x784000)                           = 0x784000
mount("\\\\192.168.1.70\\VIDEO", "/mnt/cifs", "cifs", MS_SILENT, "username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\\192.168.1.70\\VIDEO,unc=\\\\192.168.1.70\\VIDEO,ip=192.168.1.70") = -1 ENOENT (No such file or directory)
clone(child_stack=0x7fdee5f8, flags=CLONE_VM|CLONE_VFORK|SIGCHLD) = 21581
waitpid(21581, NULL, 0)                 = 21581
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=21581, si_status=111, si_utime=0, si_stime=0} ---
write(2, "mount: mounting \\\\192.168.1.70\\VIDEO on /mnt/cifs failed: No such file or directory\n", 84mount: mounting \\192.168.1.70\VIDEO on /mnt/cifs failed: No such file or directory
) = 84
exit_group(-1)                          = ?
+++ exited with 255 +++

root@openwrt:~# 

```

ambrosa, 2015-01-26:
And this is with ending \\ , => Invalid argument error
Strange strange strange
```
root@openwrt:~# strace -s 9999 mount -t cifs \\\\192.168.1.70\\VIDEO\\ /mnt/cifs -o username=ambrosa,password=foo,sec=ntlm,ip=192.168.
1.70,unc=\\\\192.168.1.70\\VIDEO\\
execve("/bin/mount", ["mount", "-t", "cifs", "\\\\192.168.1.70\\VIDEO\\", "/mnt/cifs", "-o", "username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\\192.168.1.70\\VIDEO\\"], [/* 11 vars */]) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77668000
stat("/etc/ld.so.cache", 0x7f8ed590)    = -1 ENOENT (No such file or directory)
open("/lib/libcrypt.so.0", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=9296, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77665000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\4\0\0\0\0004
(...omissis...)
\0\0\17x\200%\316\3\0", 4096) = 4096
mmap(NULL, 147456, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7762c000
mmap(0x7762c000, 9228, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x7762c000
mmap(0x7763e000, 1104, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x2000) = 0x7763e000
mmap(0x7763f000, 67280, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7763f000
close(3)                                = 0
munmap(0x77665000, 4096)                = 0
open("/lib/libm.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=89724, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77665000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\22\320\0\0\0004
(...omissis...)
\0log\0log2\0log10\0pow\0remainder\0drem\0scalb\0sqrt", 4096) = 4096
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77606000
mmap(0x77606000, 89240, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77606000
mmap(0x7762b000, 3708, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x7762b000
close(3)                                = 0
munmap(0x77665000, 4096)                = 0
open("/lib/libgcc_s.so.1", O_RDONLY)    = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=87555, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77665000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0006\260\0\0\0004
(...omissis...)
\1\354\0\0<\20\0\0\0(\22\0\0\n", 4096) = 4096
mmap(NULL, 155648, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x775e0000
mmap(0x775e0000, 87172, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x775e0000
mmap(0x77605000, 1539, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x15000) = 0x77605000
close(3)                                = 0
munmap(0x77665000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77665000
read(3, "\177ELF\1\2\1\0\0\0\0\0\0\0\0\0\0\3\0\10\0\0\0\1\0\0\234p\0\0\0004
(...omissis...)
\3\274", 4096) = 4096
mmap(NULL, 442368, PROT_NONE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x77574000
mmap(0x77574000, 351072, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED, 3, 0) = 0x77574000
mmap(0x775d9000, 7332, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x55000) = 0x775d9000
mmap(0x775db000, 17492, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x775db000
close(3)                                = 0
munmap(0x77665000, 4096)                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
open("/lib/libc.so.0", O_RDONLY)        = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=355492, ...}) = 0
close(3)                                = 0
stat("/lib/ld-uClibc.so.0", {st_mode=S_IFREG|0755, st_size=28968, ...}) = 0
mmap(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS|0x4000000, -1, 0) = 0x77665000
set_thread_area(0x7766c440)             = 0
mprotect(0x775d9000, 4096, PROT_READ)   = 0
mprotect(0x77666000, 4096, PROT_READ)   = 0
ioctl(0, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
ioctl(1, TIOCNXCL, {B38400 opost isig icanon echo ...}) = 0
getuid()                                = 0
brk(0)                                  = 0xa27000
brk(0xa28000)                           = 0xa28000
getuid()                                = 0
geteuid()                               = 0
brk(0xa29000)                           = 0xa29000
mount("\\\\192.168.1.70\\VIDEO\\", "/mnt/cifs", "cifs", MS_SILENT, "username=ambrosa,password=foo,sec=ntlm,ip=192.168.1.70,unc=\\\\192.168.1.70\\VIDEO\\,unc=\\\\192.168.1.70\\VIDEO,prefixpath=,ip=192.168.1.70") = -1 EINVAL (Invalid argument)
clone(child_stack=0x7f8ed598, flags=CLONE_VM|CLONE_VFORK|SIGCHLD) = 21637
waitpid(21637, NULL, 0)                 = 21637
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_EXITED, si_pid=21637, si_status=111, si_utime=0, si_stime=0} ---
write(2, "mount: mounting \\\\192.168.1.70\\VIDEO\\ on /mnt/cifs failed: Invalid argument\n", 76mount: mounting \\192.168.1.70\VIDEO\ on /mnt/cifs failed: Invalid argument
) = 76
exit_group(-1)                          = ?
+++ exited with 255 +++

root@openwrt:~# 

```

ambrosa, 2015-01-26:
I suppose a bug with Busybox internal cifs mount.
Any chance to have a mount.cifs (old cifs-utils package) ??

hnyman, 2015-01-27:
Just for reference: I tested with myCC trunk r44153 with kernel 3.14 (ar71xx) and mounting my NAS share using the busybox mount worked ok:
````
mount -t cifs //192.168.1.195/ftp-nas /mnt -o username=admin,password=pwpwpw,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.195\\ftp-nas
````
Note: other slashes are normal "/", but unc has "\"

And strace shows that the ip was added automatically there:
````
mount("//192.168.1.195/ftp-nas", "/mnt", "cifs", MS_SILENT, "username=admin,password=pwpwpw,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.195\\ftp-nas,unc=\\\\192.168.1.195\\ftp-nas,ip=192.168.1.195") 
````

ambrosa, 2015-01-27:
Dear hnyman , 
can you give me your .config ?
I've tried again but no success and I'm confused. Error change if I use or not the ending "/" (or "\\"
```
root@openwrt:/# mount -t cifs //192.168.1.70/VIDEO /mnt -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.70\\VIDEO
mount: mounting //192.168.1.70/VIDEO on /mnt failed: No such file or directory
root@openwrt:/# mount -t cifs //192.168.1.70/VIDEO /mnt -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.70\\VIDEO\\
mount: mounting //192.168.1.70/VIDEO on /mnt failed: No such file or directory
root@openwrt:/# mount -t cifs //192.168.1.70/VIDEO/ /mnt -o username=ambrosa,password=foo,sec=ntlm,file_mode=0644,unc=\\\\192.168.1.70\\VIDEO\\
mount: mounting //192.168.1.70/VIDEO/ on /mnt failed: Invalid argument

```

Perhaps can be some strange behaviours with my .config ....

I user pivot overlay : I expand my root storage with a 8GB USB stick.
Can be overlayfs a problem ? /mnt (the local mount point) is under overlayfs .....

hnyman, 2015-01-27:
Sure. I am building a community build for ar71xx/WNDR3700 (https://forum.openwrt.org/viewtopic.php?id=28392 ) and the full .config etc. is downloadable from https://www.dropbox.com/sh/t52c02rm20y8x9p/khFGAJu3gc
You can pick the trunk config from r44153 that I used above.

If you want just the config changes that I actually do, you may use the .config.init to see them. It is a short recipe that expand to full .config via "make defconfig".

ambrosa, 2015-02-01:
Dear hnyman,

today I've checked your .config against mine.
Nothing different about BusyBox. I'm very confused.

Now I'm building 44234 and I will try with a new OpenWrt version.

tincanfury, 2015-09-20:
I'm having this issue, I can't mount from my NAS over CIFS either,

#mount -t cifs //192.168.1.151/Home home/ -o user=<user>,password=<pass>
mount: mounting //192.168.1.151/Home on home/ failed: No such device

pyrovski, 2015-10-03:
This affects me, too. From a fresh install of Chaos Calmer (OpenWrt Chaos Calmer 15.05 / LuCI Master (git-15.233.47308-791ca8b)), I get either hangs or "No such file or directory".

ffainelli, 2016-01-03:
Same here, will try to look deeper into this. FWIW, my mounts are with "guest" not using usernames

ffainelli, 2016-01-06:
Ok, so after enabling CONFIG_CIFS_DEBUG=y in the kernel, I could finally start seeing some debug messages that were helpful:

<3>[  269.894495] CIFS VFS: No username specified

The syntax at https://wiki.openwrt.org/doc/howto/cifs.client is wrong for guest mounts:

"-o guest", when passed down to fs/cifs/connect.c is ignored as an option, the correct way would be:

"-o user=guest, or -o user=,<additional options>"

Now, the weird part is that I built two images for RPi, and one does not expose the problem, while the other does. Arguably the Samba servers are different. One is Ubuntu 14.10 4.1.11+dfsg-1ubuntu2.2, the other one is an OpenWrt 3.6.24-1....

g00dy22, 2016-01-13:
Thanks, ffainelli.
This is working for me: mount -t cifs //servername/share /mnt/share -o user=,file_mode=0777,dir_mode=0777,unc=\\\\1.1.1.27\\SHARE

jow-, 2016-01-15:
This can be a musl problem, specifically getopt incompatibilities and the way "mount" passes args down to the respective mount helpers.

petvoigt, 2016-01-16:
Yeah, the same with me. I would like a mount.cifs in favor of the the busybox mount command. It took me numerous attempts to mount my FreeNAS shares. I am still not successful because my password contains special characters. In particular a comma is misinterpreted in terms of parameter separation and no quoting helps. Another Linux share could be mounted, because of a "simpler" password. And why does the busybox mount command not offer any chance to enter the password interactively? Under 14.07 there was a useful cifsmount package that I would like to see under 15.05 as well.

ffainelli, 2016-01-16:
cifs-utils is in the oldpackages, but it should not be too hard to get it up to date and merged in packages, if no maintainer shows up, feel free to assign it to myself.

hnyman, 2016-03-15:
cifs-utils has been imported with https://github.com/openwrt/packages/commit/89cce919ecd9eb9396d09994084123532cba1a17

closed 2016-03-15 by hnyman

dimoni68, 2017-10-21:
with this is show fine
root@OpenWrt-AR-5387un:/mnt#  mount -t cifs //192.168.1.12/Harddisk /mnt/ultimo -o user=guest

and mount shows like that
`root@OpenWrt-AR-5387un:/mnt# mount
rootfs on / type rootfs (rw)
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mtdblock3 on /overlay type jffs2 (rw,noatime)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)

//192.168.1.12/Harddisk on /mnt/ultimo type cifs (rw,relatime,vers=1.0,cache=strict,username=guest,domain=WORKGROUP,uid=0,noforceuid,gid=0,noforcegid,addr=192.168.1.12,unix,posixpaths,serverino,acl,rsize=61440,wsize=65536,actimeo=1)
root@OpenWrt-AR-5387un:/mnt#`

Many thanks ffainelli