openwrt/packages #796: DDNS - Luci - no option to secure connections

phoenix-nz, 2015-01-18:
It is not possible to configure dynamic DNS securely using the Luci-Web interface.
The option use_https is not displayed, nor is there any way to add cacert files.

Preferably https should be the default anyway, as it took me over a year of running OpenWRT to realise that my dyndns password is being broadcasted openly across the web every 10 minutes :-(

chris5560, 2015-01-18:
Which Version of ddns-scripts are you using ? Click on "Dynamic DNS" header on the Overview page.
Dynamic DNS together with secure connection requires GNU Wget or cURL package to be installed.
Additional the ca-certificate package is a good idea.
Thats the reason why secure http is not the default.
If your LuCI application don't offer you to select HTTPS then neither GNU Wget nor cURL is installed on your box.
There should be "Hints" showing you details.

phoenix-nz, 2015-01-18:
Sorry if my post wasn't precise enough. The problem isn't with ddns-scripts (although I have to admit I cannot find the "Dynamic DNS" header on my Overview page).
The wiki page on OpenWRT explains quite well how to set everything up using the command line. (Which works just fine.)

My bug report is about the web interface LuCI which in my version - LuCI Trunk (0.12+svn-r10530) OpenWrt Barrier Breaker 14.07 - does not expose the https functionality. Thus it is not possible to set up a secure DynDNS connection using just the web interface.

This should (imho) be the default, or at least possible!

phoenix-nz, 2015-01-18:
[image]

I've attached a screenshot of my config page on LuCI

chris5560, 2015-01-19:
Hi,

for BB 14.07 currently only ddns-scripts v1.0 and according LuCI interface are available.
Normally the trunk packages should work but the postinst procedure inside trunk was changed,
Installation of trunk packages failed with "postinst script returned status 127".
Here a backport is currently missing from my knowledge.

The new LuCI interface require ddns-scripts 2.x and support all configuration settings to configure a service section.
I tested ddns-scripts 2.x and corresponding LuCI app with BB installing file by file.
ddns-scripts are shell scripts and LuCI lua scripts. So copy from github file by file should work.

Christian

champtar, 2015-03-11:
@phoenix-nz, can i close this issue?

chris5560, 2015-03-16:
I think you should close this.
Thanks
Christian

phoenix-nz, 2015-03-16:
Sorry about not replying earlier: I am using ddns-scripts v1.0 and luci-app-ddns v0.12 as listed by the output of:

>opkg list | grep ddns-scripts
ddns-scripts - 1.0.0-23 - A highly configurable set of scripts for doing dynamic dns updates

> opkg list | grep luci-app-ddns
luci-app-ddns - 0.12+svn-r10530-1 - Dynamic DNS configuration module

So I'm not sure what Christian meant with "for BB 14.07 currently only ddns-scripts v1.0 and according LuCI interface are available." That is fine with me, after all, ddns-scripts v1.0 seems to support https, yet the corresponding UI luci-app-ddns does not. 

Thus the error still stands?

phoenix-nz, 2015-03-16:
Note: here is the ddns config file for BB14.07:
https://github.com/openwrt/packages/blob/for-14.07/net/ddns-scripts/files/etc/config/ddns

Clearly there is an https option there.

chris5560, 2015-03-16:
There is currently no plan from my side to backport luci-app-ddns and ddns-scripts to LuCI 0.12 / BB 14.07 or to modify the existing inside current BB14.07 release.

phoenix-nz, 2015-03-16:
If I try and get the option(s) into luci-app-ddns would you review the patch? Or is there a new release of openwrt available (soon) which means it won't be worth the effort?

Because I can't find any information of the sort on the main page.

chris5560, 2015-03-17:
Hi,
there where other options missing like setting the event interface or ip_script or ip_interface.
It's unchanged since 10.03. The reason why I rebuild the scripts and app, but when I started BB 14.07 was already released.
Normally this parameters are only set once.
Give me some days I will have a look.
If you contact me by direct mail I can send you a modified file for testing
before modifying inside luci 0.12 on the released side.
Christian

champtar, 2015-07-27:
CC is almost there

closed 2015-07-27 by champtar