openwrt/packages #803: python-pip depends on "python" package, not "python-base"

bmurphy1976, 2015-01-21:
python-pip requires the "python" package, however the python package is named "python-base".

```
/ # opkg-install python-base
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/base.
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/packages.
Installing python-base (2.7.9-4) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/python-base_2.7.9-4_x86_64.ipk.
Installing libpthread (0.9.33.2-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/libpthread_0.9.33.2-1_x86_64.ipk.
Installing zlib (1.2.8-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/zlib_1.2.8-1_x86_64.ipk.
Installing libffi (3.0.13-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/libffi_3.0.13-1_x86_64.ipk.
Installing libopenssl (1.0.1l-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/libopenssl_1.0.1l-1_x86_64.ipk.
Configuring libpthread.
Configuring zlib.
Configuring libffi.
Configuring libopenssl.
Configuring python-base.
/ # python --version
Python 2.7.9
/ # opkg-install python-pip
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/base.
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/packages.
Installing python-pip (1.5.6-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/python-pip_1.5.6-1_x86_64.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for python-pip:
 * 	python * 	python *
 * opkg_install_cmd: Cannot install package python-pip.
/ #
```

If I force install the package, I get an io module error.

```
/ # pip
/bin/sh: pip: not found
/ # opkg-install --force-depends python-pip
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/base.
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Inflating http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/packages.
Installing python-pip (1.5.6-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/python-pip_1.5.6-1_x86_64.ipk.
Installing python-setuptools (7.0-1) to root...
Downloading http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/python-setuptools_7.0-1_x86_64.ipk.
Configuring python-setuptools.
Configuring python-pip.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for python-pip:
 * 	python * 	python *
/ # pip
Could not find platform dependent libraries <exec_prefix>
Consider setting $PYTHONHOME to <prefix>[:<exec_prefix>]
Traceback (most recent call last):
  File "/usr/bin/pip", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python2.7/site-packages/setuptools-7.0-py2.7.egg/pkg_resources.py", line 19, in <module>
ImportError: No module named io
/ #
```

This was working this morning.  I have an image I created that installed "python" and "python-pip" and everything worked great.  It appears to have changed sometime this afternoon.

bmurphy1976, 2015-01-21:
Another possibility, it may just be that the python package is missing.  I see a python3 package but no python package here:

http://downloads.openwrt.org/snapshots/trunk/x86_64/packages/packages/

commodo, 2015-01-21:
That would be my fault.
I broke it 2 days ago [I guess the breakage was caught by the buildbot], but I fixed it yesterday.
I wonder how long it takes for the package to be back.

bmurphy1976, 2015-01-23:
Any idea when the next build will go out?  It's been a couple days now and python is still broken.

commodo, 2015-01-23:
@sbyx  or @hnyman ?

hnyman, 2015-01-23:
> Any idea when the next build will go out?

Depends on your platform. Buildbot automatically builds everything roughly daily, but a new "ola" package was broken on three days in a row in such a way that the whole buildbot build stopped, so there haven't been succesful builds for three days. (see e.g. #808)
Check the build status here: http://buildbot.openwrt.org:8010/grid
(The most common platform, ar71xx, is currently being built.)

bmurphy1976, 2015-01-23:
OK, thanks for the info.  That gives me visibility into the process.  I'll close this ticket if and when I notice the build completed and the package was pushed.

hnyman, 2015-01-23:
Well, if you are talking about x86_64, the packages have been there for some hours, but I think that the snapshot download directory structure has changed a bit, so you may need to manually download the packages (it is now .../x86_64/generic/... ):
https://downloads.openwrt.org/snapshots/trunk/x86_64/generic/packages/packages/
python-pip_1.5.6-1_x86_64.ipk                      23-Jan-2015 12:23              727386
python3_3.4.2-1_x86_64.ipk                         23-Jan-2015 12:28             4822355
python_2.7.9-4_x86_64.ipk                          23-Jan-2015 10:53             4559859

bmurphy1976, 2015-01-23:
Boom!  Problem solved.  I was able to build my image.  Thanks!

closed 2015-01-23 by bmurphy1976