openwrt/packages #808: "ola" fails all buildbot runs

jow-, 2015-01-22:
The recently introduced "ola" package appears to have impicent dependencies on protobuf devlopment files on the host system, this is not acceptable for OpenWrt, it seems we need a host build of the protobuf package and add a host build dependencies to the ola package.

Until this is sorted out I recommend marking the package as @BROKEN, it is the 2nd issue that broke the builders. Please also test compiling "ola" in a clean system, with just the minimal build tools installed and a fresh clone of OpenWrt, to rule out other implicit dependencies.

One failing build log can be seen here: http://buildbot.openwrt.org:8010/builders/atheros/builds/866/steps/compile_5/logs/stdio

mhei, 2015-01-22:
I think these compile tests *must* be performed before the PR is created, and *should* be performed a second time by the developer who merges the PR (aka two-man rule). I recommend to first assign a PR to a developer to mark it as "merge-test-in-progress", and finally only merge it if such basic tests are solved. The developer then could add a second SoB line to indicate this review.

thess, 2015-01-22:
Yes, I agree that PR's should be tested before merging. I've been trying to do some of them however, they are usually merged (time-zone differences, etc) before I get a chance. I would be more than willing to take on some of the build-testing before merging. This is probably more necessary for new package submissions.

mhei, 2015-01-22:
That's why I propose to mark a PR by assigning. This way you show that you're working on it and nobody else needs to care. BTW I think that we do not need to hurry when merging: quality before speed.

hnyman, 2015-01-22:
Somebody with merge rights should quickly mark this ola package broken, as the third day of buildbot runs has already been destroyed by this one failing package. Afaik the faulty code is still active... I am not sure if @bk138 (author of the package) or @sbyx (did the merge) are yet even aware of the problem.

I proposed yesterday on the openwrt-devel mailing that the buildbot logic should be changed so that failing packages from feeds would not kill the whole buildbot build any more. It seems rather wasteful that one misbehaving non-essential package, like ola now, can kill the whole buildbot run. Apparently having either a host section or a kmod definition in the failing package is enough to kill the whole buildbot build.
https://lists.openwrt.org/pipermail/openwrt-devel/2015-January/030719.html

sbyx, 2015-01-22:
it's broken for now. another problem here is that spurious host-dependencies often just aren't noticed by developers even the ones who merge it as we usually tend to have all sorts of development headers and tools installed.

bk138, 2015-01-22:
Sorry for the trouble caused - yes ola depends on a host build of protoc. I was modeling the package after the vala package with seems to depend on a host build as well. What do I need to change?

mhei, 2015-01-22:
Please have a look into the branch I pushed just now. But it's not yet fully tested... just a starting point.

luizluca, 2015-01-23:
It would save some headache if there were some buildbot that compiles, for all targets, all openwrt/packages on top of current working SDK. Some projects do just like that (and also do some "make test"). However, I guess buildbot are already too busy on their current job.

bk138, 2015-01-27:
Can someone point me at the docs where dependency handling is described? Grepping for HOST_BUILD_DEPENDS in the whole SDK (barrier breaker) give me no hints on what that tag actually means... 

Package lang/vala has a HOST_BUILD_DEPENDS, for instance, and this also fails to build on my machine since I don't have the glib and gobject dev packages installed on the host.

Thanks for your help!

hnyman, 2015-01-27:
vala probably fails just now because glib seems to be broken right now (issue #826). Missing glib breaks lots of other modules...

But the (host) dependency handling is not really explained anywhere to my knowledge. Best source that I have found is http://wiki.openwrt.org/doc/devel/packages (in which host section is rather short, but references also this message: https://lists.openwrt.org/pipermail/openwrt-devel/2014-February/023970.html )

bk138, 2015-01-29:
Yeah I've seen  http://wiki.openwrt.org/doc/devel/packages and tried to model the package after this and the other ones in the feed. Are there any docs on how to setup an own buildbot environment to test dependency handling. AFAICT, a simple make in the SDK doesn't pull in any _DEPENDS, or am I doing sth wrong?

bk138, 2015-03-16:
Hey there again, I've incorporated https://github.com/mhei/packages/commit/359d78fd9e3c1c8e2dfbfc246c2ce8b1ff3dc355 and updated OLA to 0.9.5 in https://github.com/bk138/packages/tree/ola-squashed. It now builds without protobuf installed on the host system and a fresh and empty SDK - *however* I still have to issue a manual ```make package/protobuf/compile``` before being able to build ola via ```make package/ola/compile``` - dunno if the build system pulls in deps automagically...

bk138, 2015-05-02:
Is there anything I can do - shall I open a new PR?

closed 2015-07-27 by champtar