openwrt/packages #812: CC/Luci - all-zero mac address for eth0 reported on Archer C7v2

stangri, 2015-01-22:
[image]
As seen above luci reports an incorrect mac address for eth0. ifconfig reports the actual mac address tho.

Not sure if it's limited to Archer C7v2. 

Here's the Luci version from the footer: Powered by LuCI (git-15.017.51462-30d45d0).

jow-, 2015-01-22:
Please paste the full output of "ubus call network.interface dump" and "ubus call network.device status"

stangri, 2015-01-22:
Uhm, you said full, but I've sanitized it a little for obvious reason.

ubus call network.interface dump:
{
	"interface": [
		{
			"interface": "cfg0c4d8f",
			"up": true,
			"pending": false,
			"available": true,
			"autostart": true,
			"uptime": 65665,
			"l3_device": "br-lan",
			"proto": "static",
			"device": "br-lan",
			"updated": [
				"addresses"
			],
			"metric": 0,
			"delegation": true,
			"ipv4-address": [
				{
					"address": "192.168.1.11",
					"mask": 24
				}
			],
			"ipv6-address": [

			],
			"ipv6-prefix": [

			],
			"ipv6-prefix-assignment": [

			],
			"route": [

			],
			"dns-server": [

			],
			"dns-search": [

			],
			"inactive": {
				"ipv4-address": [

				],
				"ipv6-address": [

				],
				"route": [

				],
				"dns-server": [

				],
				"dns-search": [

				]
			},
			"data": {

			}
		},
		{
			"interface": "lan",
			"up": true,
			"pending": false,
			"available": true,
[... 371 lines cut ...]
			"rx_frame_errors": 0,
			"tx_compressed": 0,
			"multicast": 0,
			"rx_length_errors": 0,
			"tx_dropped": 0,
			"rx_bytes": 104425265,
			"rx_missed_errors": 0,
			"tx_errors": 0,
			"rx_compressed": 0,
			"rx_over_errors": 0,
			"tx_fifo_errors": 0,
			"rx_crc_errors": 0,
			"rx_packets": 1380749,
			"tx_heartbeat_errors": 0,
			"rx_dropped": 0,
			"tx_aborted_errors": 0,
			"tx_packets": 4022739,
			"rx_errors": 0,
			"tx_bytes": 1398958204,
			"tx_window_errors": 0,
			"rx_fifo_errors": 0,
			"tx_carrier_errors": 0
		}
	}
}

jow-, 2015-01-23:
Ok, so this problem is not LuCI related, the mac address information is not exposed via ubus for eth0/eth1.

stangri, 2015-01-23:
As I have no idea about how ubus is supposed to work, could you then please open the bug in openwrt project tracker when you close this one?

champtar, 2015-03-11:
@stangri @jow-, can we close this one?

stangri, 2015-03-11:
It's not a Luci bug, but it's still a bug -- the mac address information is not exposed via ubus for eth0/eth1. Either a new one needs to be opened (and I do not understand ubus stuff so it would be great if developers opened it) and this one closed or this one needs to be adjusted and kept open. Your call.

FedericoVaga, 2015-06-23:
I stumbled upon the same issue. From *sysfs* and *ifconfig* I get the correct MAC address. But from *ubus* (so luci) the MAC address is not there. It's not clear to me how *ubus* get the information from the system.

stangri, 2017-02-13:
I no longer own an Archer C7v2. If the problem persists for other users, maybe they could open an issue against ubus, but I'm closing this as I can't even confirm it's still an issue.

closed 2017-02-13 by stangri