openwrt/packages #826: glib2 fails to build after 2.43.3 update

kevinmehall, 2015-01-27:
After b490ad69, glib2 fails to build with the error:

```
make[7]: Entering directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3/gio'
/bin/bash ../libtool  --tag=CC   --mode=link mipsel-openwrt-linux-uclibc-gcc -Wall -Wstrict-prototypes -Werror=declaration-after-statement -Werror=missing-prototypes -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format=2 -Werror=missing-include-dirs -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -I/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libiconv-stub/include -I/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libintl-stub/include   -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lib -L/vol/wrtnode/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/vol/wrtnode/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libiconv-stub/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libintl-stub/lib  -o glib-compile-resources gvdb-builder.o glib-compile-resources.o libgio-2.0.la ../gobject/libgobject-2.0.la ../glib/libglib-2.0.la  
libtool: link: mipsel-openwrt-linux-uclibc-gcc -Wall -Wstrict-prototypes -Werror=declaration-after-statement -Werror=missing-prototypes -Werror=implicit-function-declaration -Werror=pointer-arith -Werror=init-self -Werror=format=2 -Werror=missing-include-dirs -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -I/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libiconv-stub/include -I/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libintl-stub/include -o .libs/glib-compile-resources gvdb-builder.o glib-compile-resources.o  -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/lib -L/vol/wrtnode/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/vol/wrtnode/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libiconv-stub/lib -L/vol/wrtnode/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/usr/lib/libintl-stub/lib ./.libs/libgio-2.0.so ../gobject/.libs/libgobject-2.0.so ../glib/.libs/libglib-2.0.so
../glib/.libs/libglib-2.0.so: warning: the use of OBSOLESCENT `utime' is discouraged, use `utimes'
/vol/wrtnode/staging_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/lib/gcc/mipsel-openwrt-linux-uclibc/4.8.3/../../../../mipsel-openwrt-linux-uclibc/bin/ld: warning: libgmodule-2.0.so.0, needed by ./.libs/libgio-2.0.so, not found (try using -rpath or -rpath-link)
./.libs/libgio-2.0.so: undefined reference to `g_module_error'
./.libs/libgio-2.0.so: undefined reference to `g_module_close'
./.libs/libgio-2.0.so: undefined reference to `g_module_open'
./.libs/libgio-2.0.so: undefined reference to `g_module_supported'
./.libs/libgio-2.0.so: undefined reference to `g_module_symbol'
collect2: error: ld returned 1 exit status
make[7]: *** [glib-compile-resources] Error 1
make[7]: Leaving directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3/gio'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3/gio'
make[5]: *** [all] Error 2
make[5]: Leaving directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3/gio'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3'
make[3]: *** [all] Error 2
make[3]: Leaving directory `/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3'
make[2]: *** [/vol/wrtnode/build_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/glib-2.43.3/.built] Error 2
make[2]: Leaving directory `/vol/wrtnode/feeds/packages/libs/glib2'
make[1]: *** [package/feeds/packages/glib2/compile] Error 2
make[1]: Leaving directory `/vol/wrtnode'
make: *** [package/glib2/compile] Error 2
```
When that commit is reverted (downgrading glib to 2.42.1), the package builds successfully.

tripolar, 2015-01-27:
compiles here without a problem. try to rebuild openwrt

hnyman, 2015-01-27:
glib2 fails also in buildbot for most platforms...
E.g. ar71xx, lantiq and x86 have failed in builds just completed (r44152+) .
http://buildbot.openwrt.org:8010/broken_packages/ar71xx.nand/glib2/compile.txt
http://buildbot.openwrt.org:8010/broken_packages/lantiq/glib2/compile.txt
http://buildbot.openwrt.org:8010/broken_packages/x86/glib2/compile.txt

tripolar, 2015-01-27:
thanks jow. should be fixed now.

closed 2015-01-27 by tripolar