openwrt/packages #829: luaposix: module 'posix.deprecated' not found

karlp, 2015-01-29:
luaposix module fails to load on current trunk.

```
root@eg-018C06:/# opkg list-installed | grep posix
luaposix - v33.2.1-1
root@eg-018C06:/# lua
Lua 5.1.5  Copyright (C) 1994-2012 Lua.org, PUC-Rio (double int32)
> require("posix")
/usr/lib/lua/posix.lua:40: module 'posix.deprecated' not found:
	no field package.preload['posix.deprecated']
	no file './posix/deprecated.lua'
	no file '/usr/share/lua/posix/deprecated.lua'
	no file '/usr/share/lua/posix/deprecated/init.lua'
	no file '/usr/lib/lua/posix/deprecated.lua'
	no file '/usr/lib/lua/posix/deprecated/init.lua'
	no file './posix/deprecated.so'
	no file '/usr/lib/lua/posix/deprecated.so'
	no file '/usr/lib/lua/loadall.so'
	no module 'posix.deprecated' in file '/usr/lib/lua/posix.so'
stack traceback:
	[C]: in function 'require'
	/usr/lib/lua/posix.lua:40: in main chunk
	[C]: in function 'require'
	stdin:1: in main chunk
	[C]: ?
> 
```

mstorchak, 2015-01-29:
Thanks for the notice, will take a look asap.

mstorchak, 2015-01-29:
@karlp , could you please test this patch?
```diff
diff --git a/lang/luaposix/Makefile b/lang/luaposix/Makefile
index 7294943..94abf14 100644
--- a/lang/luaposix/Makefile
+++ b/lang/luaposix/Makefile
@@ -9,7 +9,7 @@ include $(TOPDIR)/rules.mk

 PKG_NAME:=luaposix
 PKG_VERSION:=v33.2.1
-PKG_RELEASE:=1
+PKG_RELEASE:=2

 PKG_SOURCE:=release-$(PKG_VERSION).tar.gz
 PKG_SOURCE_URL:=https://github.com/luaposix/luaposix/archive/
@@ -48,9 +48,10 @@ endif

 define Package/luaposix/install
-       $(INSTALL_DIR) $(1)/usr/lib/lua
+       $(INSTALL_DIR) $(1)/usr/lib/lua/posix
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/ext/posix/.libs/posix.so $(1)/usr/lib/lua
        $(INSTALL_BIN) $(PKG_BUILD_DIR)/lib/posix.lua $(1)/usr/lib/lua
+       $(INSTALL_BIN) $(PKG_BUILD_DIR)/lib/posix/*.lua $(1)/usr/lib/lua/posix
 endef

 $(eval $(call BuildPackage,luaposix))
```

karlp, 2015-01-29:
I would imagine that would work, after looking at how it's built myself.  Seems kinda gross to have to add 40k of compat and deprecated libs of course :)  Should we perhaps file a bug upstream?

karlp, 2015-01-29:
(putting a patch in text in github comments isn't very amenable to actually applying it, fwiw, your own branch might have been easier for me to try pulling it in?)

mstorchak, 2015-01-29:
> Seems kinda gross to have to add 40k of compat and deprecated libs of course :)
Oops, didn't notice the size :). I'd like to leave it as is for now to unbreak the package and debloat it later though.
> Should we perhaps file a bug upstream?
I'm not sure :-/. Luaposix may have to maintain compatibility with old releases. I'll try to see how to get rid of legacy stuff.

> putting a patch in text in github comments isn't very amenable to actually applying it, fwiw, your own branch might have been easier for me to try pulling it in?
you are right, thanks :). I tried to get confirmation ASAP, so adding a patch to the discussion looked like a good idea, but maybe it's not :).

karlp, 2015-01-30:
fix merged

closed 2015-01-30 by karlp

karlp, 2015-08-27:
heh. And now I've come across the problem of missing the compat includes :|  I guess I can't complain about it.  Had some BB code that fails to run on CC because we trimmed out the compat libs.  Given I even suggested dropping it myself here though, I'll just bit the bullet and work around them :)