openwrt/packages #847: seafile-seahub failed to build on buildbot (ar71xx + other)

plntyk, 2015-02-02:
from http://buildbot.openwrt.org:8010/broken_packages/ar71xx/seafile-seahub/compile.txt
```
ERROR:root:code for hash sha224 was not found.
...
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
...
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
...
raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
...
Downloading https://www.djangoproject.com/m/releases/1.5/Django-1.5.8.tar.gz
error: Download error for https://www.djangoproject.com/m/releases/1.5/Django-1.5.8.tar.gz: unknown url type: https
```

commodo, 2015-02-02:
I'm wondering if this is my fault, since Python may not build with SSL support.
I'll check it out this week; at the moment I'm on the road.

plntyk, 2015-02-03:
buildroot project does not build host python with ssl
have a look at their patches/patch series and what is disabled for their host-build
host python ssl change seems trivial

http://git.buildroot.net/buildroot/tree/package/python

commodo, 2015-02-03:
I did not know about buildroot's Python package.
Seems sensible enough to incorporate a few of their patches in the Python package.
Thing is: the OpenWRT Python package also builds a Host version; I guess in case of the buildbot case, buildroot's Python is used.

xfguo, 2015-02-06:
I got the same problem, my temporary solution is 

```
mv staging_dir/host/bin/python2.7 staging_dir/host/bin/python2.7.bak
ln -s /usr/bin/python2.7 `pwd`/staging_dir/host/bin/python2.7
````

commodo, 2015-02-08:
I tried a local build of seafile. This is without buildroot.
It built for me.
Looking through the `seafile-seahub` Makefile, it seems to call `HostPython` which should be the Python executable that is built by the `python` package.

So, I'm guessing that buildroot may add it's Python somewhere.

kissg1988, 2015-02-09:
Package maintainer here.

Is there anything I could do to fix this? I'm not sure how this got broken, it was working fine earlier. Could you please find out which change of the python package might have caused this and revert the change or create a patch?

If there's anything I can help please let me know.

xfguo, 2015-02-10:
hi, @kissg1988

My problem is, I have a package (smartsnmp) which is use _SCons_ as construct tool, it is failed when use latest trunk SDK to build it. It can find some `SHA` and so on libs in python.

commodo, 2015-02-10:
Helllo,

@kissg1988 seafile-seahub builds for me [locally], so it may be some build-environment issue,

Let's pull in some packages maintainers.
@sbyx  @hnyman  @thess 

Any idea if something changed recently in buildbot that would have caused this ?
At the moment it seems that the issue may be what @plntyk found.
Any input ?

Thanks
Alex

hnyman, 2015-02-10:
Why not download manually from djangoproject? Why does python need to do the https download?

kissg1988, 2015-02-10:
@hnyman this is not simply about downloading django. When I created the package's Makefile, I wanted to use a simple and flexible way to install the required python dependencies. If you take a look at the [Makefile](https://github.com/openwrt/packages/blob/master/net/seafile-seahub/Makefile), you'll see that all of the python dependencies are installed this way. Unfortunately, dependencies are only available through HTTPS (the HTTP links are redirecting) so using http links wouldn't help here.

Any ideas?

hnyman, 2015-02-10:
Well, the define a download step for all of those dependent packages. Download them first with normal download routines, and then easy-install (?) them at the compile step.

I haven't crafted download sections myself, but e.g. usbmode package has a separate data download step. To my knowledge data files can be downloaded and then possibly manipulated at the "prepare" step.
Examples:
https://dev.openwrt.org/browser/trunk/package/utils/usbmode/Makefile
https://dev.openwrt.org/browser/trunk/package/utils/usbutils/Makefile

commodo, 2015-02-11:
[ My personal input here ]
The `easy_install` script offers a way to install packages through python.
I will agree it's a bit un-standard for packaging (since no other package uses it, to my knowledge).
That said, I see no reason not to make it a standard/accepted packaging method, unless someone offers some convincing arguments.

My only take on this: why would python allow HTTP and not allow HTTPS ? (Or, does python allow HTTP ?)
It is the reason we're having this conversation in the first place, and my opinion is that if python disallows HTTPS, it should also disallow HTTP and other networking functions, since it would only be disallowed as a security measure (to not access any external stuff on the net).
Also, disallowing HTTPS and allowing HTTP (if it is allowed) is unsecure to begin with.

And I still haven't got a clue to this: which host Python is used ? buildbot's or the one in the packages feed ?

Thanks
Alex

kissg1988, 2015-02-11:
@commodo  This is really strange. The HostPython macro calls the version of python built by the buildroot and not the one installed on the host system as seen [here](https://github.com/openwrt/packages/blob/master/lang/python/files/python-package.mk):

> HOST_PYTHON_BIN:=$(STAGING_DIR_HOST)/bin/python2

> define HostPython
> ( export PYTHONPATH="$(PYTHONPATH)"; \
> export PYTHONOPTIMIZE=""; \
> export PYTHONDONTWRITEBYTECODE=1; \
> $(1) \
> $(HOST_PYTHON_BIN) $(2); \
> )
> endef

I believe buildbot should behave exactly the same as a local buildroot on a PC, then but it doesn't. The only possible reason I can imagine is that buildbot builds python without SSL support included or the python package has some dependency that lacks support for HTTPS. Could someone please check that?

plntyk, 2015-02-15:
so to get this / future errors tracked down
- config.log files in build_dir/host should be packaged into logs.tar.gz on buildbots
- Compile log of host packages should be generated (its missing from host-build.mk afaics)

champtar, 2015-03-11:
can this be closed (this is now building on buildbots)?

kissg1988, 2015-03-13:
Sure, please do so.