openwrt/packages #858: miniupnpd prints help message and don't want to start

andrew-aladev, 2015-02-05:
I've installed miniupnpd on MR3020 from 14.07 release

```
sudo /etc/init.d/miniupnpd start
> Usage: ...
sudo /usr/sbin/miniupnpd -f /var/etc/miniupnpd.conf -d
> Usage: ...
```

What is going on? It looks like miniupnpd build is completely broken.

thess, 2015-02-11:
It seems from this output that the internal lan interface is not available. There is no listening_ip (interface) in the output you provided.

I suspect the problem lies in your network config and/or miniupnpd defaults. Could you please supply some additional data:

* Output from 'ifconfig'
* Contents of /etc/config/upnpd
* Contents of /etc/config/network and /etc/config/wireless.

andrew-aladev, 2015-02-13:
Oh. The problem was `option internal_iface 'lan'`. I've changed it to `option internal_iface 'LAN'` and miniupnpd started working.

closed 2015-02-13 by andrew-aladev