openwrt/packages #859: Netatalk upgrade to 3.1.7 (or the latest)

YasharF, 2015-02-05:
The current version in OpwenWRT Packages is version 2.2.4 ( https://github.com/openwrt/packages/blob/c3cfd145356e3461eeb8ae2c3a4ba3386ef312c7/net/netatalk/Makefile#L11 ); while the latest version of Netatalk is 3.1.7.  There have been a slew of bug fixes and new features added to this version. See http://netatalk.sourceforge.net/

champtar, 2015-07-27:
#1550

closed 2015-07-27 by champtar

Neustradamus, 2024-11-25:
@YasharF: For the history:

Netatalk 3.1.7 (not merged):
- https://github.com/openwrt/packages/pull/1550

Netatalk 3.1.9:
- https://github.com/openwrt/packages/pull/3135

Netatalk 3.1.11:
- https://github.com/openwrt/packages/pull/4402

Netatalk 3.1.12:
- https://github.com/openwrt/packages/pull/7850

netatalk: drop package:
- https://github.com/openwrt/packages/pull/15873

Netatalk 3.1.13:
- https://github.com/openwrt/packages/pull/18687
- Backport: https://github.com/openwrt/packages/pull/18696

Netatalk 3.1.18 (not merged):
- https://github.com/openwrt/packages/pull/24206
- https://github.com/openwrt/packages/pull/24249

Netatalk 3.2.0:
- https://github.com/openwrt/packages/pull/24257

Netatalk 3.2.5:
- https://github.com/openwrt/packages/pull/24762

Netatalk 4.0.2 (not merged):
- https://github.com/openwrt/packages/pull/25195

Netatalk 4.0.4:
- Master: https://github.com/openwrt/packages/pull/25200 + https://github.com/openwrt/packages/pull/25251
- 24.10 backport: https://github.com/openwrt/packages/pull/25246 + https://github.com/openwrt/packages/pull/25260

YasharF, 2025-01-12:
@Neustradamus I'm not sure why you are tagging me on this github issue that was created and resolved 9 years ago.  I don't even use OpenWRT at the moment.  If you need support, please create an issue in the appropriate repo.  Thank you.

Neustradamus, 2025-01-12:
@YasharF: I was for history reason.

Netatalk has been removed in the past, and have done a comeback :)