openwrt/packages #861: mt7601 temporary package implementation

cusspvz, 2015-02-05:
Currently mt7601 isn't part of kernel wireless drivers, but would be nice to have on **OpenWRT** since:
* it is a "recent" chip, it may take a while to merge into kernel's repo.
* it is cheap, just a couple of dollars on ebay, i have no doubt that educational organizations would select them if they appear as a choice.

I brought 4 usb pens with that chip for messing around, as a user of OpenWRT, I would be sad to switch to raspbian on my raspberry pi's.

Long story short, @olivluca already made a [mt7601 package](https://github.com/olivluca/mt7601-openwrt), could it be part of openwrt/packages?

olivluca, 2015-02-05:
El 05/02/15 a les 18:45, José Moreira ha escrit:
> Currently mt7601 isn't part of kernel wireless drivers, but would be
> nice to have on *OpenWRT* since:
>
>   * it is a "recent" chip, it may take a while to merge into kernel's repo.
>   * it is cheap, [...]

Note that, while the package worked, I won't be maintaining it any more,
since the cheap adapter I bought broke after a couple of uses.
I have to say that, while it worked, it did so remarkably well, but
there is probably a reason why it is so cheap ;-)

Bye
-- 
Luca

cusspvz, 2015-02-22:
Hi @olivluca , thanks for your reply!

What equipment where you using with OpenWRT?
I'm getting issues with Wifi connections on RPi, first i tough it was related with `mt7601` but tried other chips as: `zd1211rw` and `carl9170` and they didn't connect.

olivluca, 2015-02-22:
This one
http://wiki.openwrt.org/toh/astoria/arv7518pw

tandukion, 2015-03-16:
Hi @olivluca , I am also having trouble with MT7601.
The problem is I am going to use it for my Arduino Yun, and I am still newbie to OpenWRT. I am still confused how to make the packages.

Is the .ipk file is made from creating the package?  If so, can i get it directly (since I can't create it)?

olivluca, 2015-03-16:
The ipk is made by the build system and it depends on the platform.
See the section "Building Openwrt" here: http://wiki.openwrt.org/doc/start

cusspvz, 2015-03-16:
I've created the package, but it seems that wifi isn't working on RPi, neither with other cards. I did not went deeper, so I've installed Raspbian and configured everything by hand.

cusspvz, 2015-03-16:
@tandukion after following building steps as @olivluca said, you must clone repo somewhere on `packages/` folder, then `make menuconfig` and activate that package.

cusspvz, 2015-03-31:
UPDATE:

It seems that this devices have a change on Linux kernel. @kuba-moo written it from scratch attempting to merge it onto kernel.

https://github.com/kuba-moo/mt7601u

hoatienii, 2015-04-03:
How to add to openwrt package?

cusspvz, 2015-04-03:
* Prepare your build system
* Once you have it, do from your folder:
  * `git clone https://github.com/olivluca/mt7601-openwrt packages/kernel/mt7601-olivluca`
  * `make package/kernel/mt7601/{clean,compile,install}`

cusspvz, 2015-04-03:
UPDATE:
I've created a package for @kuba-moo 's driver version, but it won't work until **OpenWRT** gets into `Linux 3.19`.

https://github.com/cusspvz/openwrt-mt7601u

hoatienii, 2015-04-04:
Thanks! Now, openwrt still 3.18. I will wait.

cusspvz, 2015-08-31:
kernel 4.2 has now an [mt7601](https://github.com/kuba-moo/mt7601u) driver merged.

plntyk, 2015-08-31:
mt7601u driver is in OpenWrt since:
https://dev.openwrt.org/changeset/46682

this ticket can be closed

closed 2015-09-01 by cusspvz