openwrt/packages #864: nginx isn't compile with SSL

foxy82, 2015-02-06:
I'm on Barrier Breaker and have installed nginx but the SSL options don't work as it hasn't been compile with SSL. Is there a way I can install SSL?

heil, 2015-02-06:
Hi,

SSL is not the default for Barrier Breaker. So you need to take the
ImageBuilder and recompile  the nginx package with SSL Support. You can
also try
to use haproxy  which has SSL Support allready.

cheers
thomas

dangowrt, 2015-02-06:
Small correction: To re-build nginx with SSL (or any other package) for a specific release, you will need the OpenWrt-SDK-* rather than OpenWrt-ImageBuilder-*

heil, 2015-02-06:
Hi,

> > Hi,
> > >
> > > I'm on Barrier Breaker and have installed nginx but the SSL options
> > > don't work as it hasn't been compile with SSL. Is there a way I can
> > > install SSL?
> > >
> > >
> > SSL is not the default for Barrier Breaker. So you need to take the
> > ImageBuilder and recompile the nginx package with SSL Support. You can
> > also try
> > to use haproxy which has SSL Support allready.
>
> Small correction: To re-build nginx with SSL (or any other package)
> for a specific release, you will need the OpenWrt-SDK-* rather than
> OpenWrt-ImageBuilder-*
Thats right, my fault.

champtar, 2015-03-19:
Hi @heil,
your answer is a bit misleading, because SSL isn't the default in CC, not only in BB

closed 2015-03-19 by champtar

praeluceo, 2016-02-29:
I can confirm that this also affects CC:
`root@Helepolis:/etc/nginx# nginx -V
nginx version: nginx/1.4.7
built by gcc 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r46450) 
configure arguments: --crossbuild=Linux::arm --prefix=/usr --conf-path=/etc/nginx/nginx.conf --add-module=/home/buildbot/slave-local/mvebu_generic/build/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/nginx-1.4.7/nginx-naxsi/naxsi_src --with-ipv6 --without-http-cache --add-module=/home/buildbot/slave-local/mvebu_generic/build/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/nginx-1.4.7/nginx-syslog --error-log-path=/var/log/nginx/error.log --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=ccache_cc --with-cc-opt='-I/home/buildbot/slave-local/mvebu_generic/build/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include -I/home/buildbot/slave-local/mvebu_generic/build/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/include -I/home/buildbot/slave-local/mvebu_generic/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/include -I/home/buildbot/slave-local/mvebu_generic/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include -Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft' --with-ld-opt='-L/home/buildbot/slave-local/mvebu_generic/build/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/lib -L/home/buildbot/slave-local/mvebu_generic/build/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/lib -L/home/buildbot/slave-local/mvebu_generic/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/lib -L/home/buildbot/slave-local/mvebu_generic/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/lib'
root@Helepolis:/etc/nginx# cat /etc/openwrt_*
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Chaos Calmer'
DISTRIB_REVISION='r47608'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05'
DISTRIB_TAINTS='no-all'
15.05`

When the build servers are back up, could nginx be compiled with common stock options built-in by default?

For comparison, these are the modules compiled into nginx by default in Ubuntu:

`root@trebuchet:/$ nginx -V
nginx version: nginx/1.1.19
TLS SNI support enabled
configure arguments: --prefix=/etc/nginx --conf-path=/etc/nginx/nginx.conf --error-log-path=/var/log/nginx/error.log --http-client-body-temp-path=/var/lib/nginx/body --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --http-log-path=/var/log/nginx/access.log --http-proxy-temp-path=/var/lib/nginx/proxy --http-scgi-temp-path=/var/lib/nginx/scgi --http-uwsgi-temp-path=/var/lib/nginx/uwsgi --lock-path=/var/lock/nginx.lock --pid-path=/var/run/nginx.pid --with-debug --with-http_addition_module --with-http_dav_module --with-http_geoip_module --with-http_gzip_static_module --with-http_image_filter_module --with-http_realip_module --with-http_stub_status_module --with-http_ssl_module --with-http_sub_module --with-http_xslt_module --with-ipv6 --with-sha1=/usr/include/openssl --with-md5=/usr/include/openssl --with-mail --with-mail_ssl_module --add-module=/build/buildd/nginx-1.1.19/debian/modules/nginx-auth-pam --add-module=/build/buildd/nginx-1.1.19/debian/modules/nginx-echo --add-module=/build/buildd/nginx-1.1.19/debian/modules/nginx-upstream-fair --add-module=/build/buildd/nginx-1.1.19/debian/modules/nginx-dav-ext-module`