openwrt/packages #865: lvm2 compile error

love4taylor, 2015-02-06:
make[5]: Entering directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.116/tools'
arm-openwrt-linux-uclibcgnueabi-gcc -Os -pipe -march=armv6k -mtune=mpcore -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft -I/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/usr/include -I/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/include -I/home/love4taylor/openwrt/trunk/staging_dir/toolchain-arm_mpcore_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/include -I/home/love4taylor/openwrt/trunk/staging_dir/toolchain-arm_mpcore_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/include -fpic -L/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/usr/lib -L/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/lib -L/home/love4taylor/openwrt/trunk/staging_dir/toolchain-arm_mpcore_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/usr/lib -L/home/love4taylor/openwrt/trunk/staging_dir/toolchain-arm_mpcore_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/lib  -O2 -L../libdm -L../lib -L../libdaemon/client -Wl,-z,relro,-z,now -pie -fPIE -Wl,--export-dynamic -o lvm dumpconfig.o formats.o lvchange.o lvconvert.o lvcreate.o lvdisplay.o lvextend.o lvmchange.o lvmcmdline.o lvmdiskscan.o lvreduce.o lvremove.o lvrename.o lvresize.o lvscan.o polldaemon.o pvchange.o pvck.o pvcreate.o pvdisplay.o pvmove.o pvremove.o pvresize.o pvscan.o reporter.o segtypes.o tags.o toollib.o vgcfgbackup.o vgcfgrestore.o vgchange.o vgck.o vgcreate.o vgconvert.o vgdisplay.o vgexport.o vgextend.o vgimport.o vgmerge.o vgmknodes.o vgreduce.o vgremove.o vgrename.o vgscan.o vgsplit.o lvm.o \
		-llvm-internal -ldaemonclient  -ldl -lblkid  -ldevmapper -lreadline -lncurses  -rdynamic
/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/usr/lib/libblkid.so: warning: This version of __fpending returns bytes remaining in buffer for both narrow and wide streams.  glibc's version returns wide chars in buffer for the wide stream case.
reporter.o: In function `_report':
reporter.c:(.text+0x1430): undefined reference to `dm_report_compact_fields'
../lib/liblvm-internal.a(thin.o): In function `_thin_pool_add_target_line':
thin.c:(.text+0x10d8): undefined reference to `dm_tree_node_set_thin_pool_error_if_no_space'
collect2: error: ld returned 1 exit status
make[5]: *** [lvm] Error 1
make[5]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.116/tools'
make[4]: *** [tools] Error 2
make[4]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.116'
make[3]: *** [/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.116/.built] Error 2
make[3]: Leaving directory `/home/love4taylor/openwrt/trunk/feeds/packages/utils/lvm2'
make[2]: *** [package/feeds/packages/lvm2/compile] Error 2
make[2]: Leaving directory `/home/love4taylor/openwrt/trunk'
make[1]: *** [/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/love4taylor/openwrt/trunk'

dangowrt, 2015-02-06:
This happens due to left-overs from previous builds. Please make clean and try again.

closed 2015-02-06 by dangowrt

love4taylor, 2015-02-06:
always errors

dangowrt, 2015-02-06:
I'm sorry, but really, I'm using it (on my NAS with actual volumes),
all buildbots do emmit the package properly, see e.g.
ARMv5
http://downloads.openwrt.org/snapshots/trunk/kirkwood/generic/packages/packages/lvm2_2.02.116-1_kirkwood.ipk
MIPS64
http://downloads.openwrt.org/snapshots/trunk/netlogic/generic/packages/packages/lvm2_2.02.116-1_netlogic.ipk
MIPS32
http://downloads.openwrt.org/snapshots/trunk/ramips/mt7628/packages/packages/lvm2_2.02.116-1_ramips_24kec.ipk
PowerPC
http://downloads.openwrt.org/snapshots/trunk/mpc85xx/generic/packages/packages/lvm2_2.02.116-1_mpc85xx.ipk
...

So please clean your source tree and try again.
It must be some leftover libdevmapper stuck in staging_dir or something
like that. (you are not the first one coming along with that problem,
however, properly cleaning the tree fixed it for everyone by now)