openwrt/packages #883: Openconnect route cleanup issue

sdysj, 2015-02-09:
Whenever disconnected an openconnect connection, there is still one route ( to server) did not remove. Also after removed that route manually and reconnected the wan, that route just regenered again.

I can't figure out  the issue is related with netifid or openconnect itself.

sdysj, 2015-02-11:
Ok, so i have tracked down the issue cause by function proto_add_host_dependency.

When VPN service started, the function linked with output interface, it will be great if netifd provide remove dependency function.

nmav, 2015-02-27:
Deja vu, but I don't remember what I did with that issue. Shouldn't that be reported in https://dev.openwrt.org/wiki ? It's a base component issue.

closed 2016-04-21 by nmav