openwrt/packages #892: expat 2.1.0 doesn't exist on sources

cusspvz, 2015-02-10:
https://github.com/openwrt/packages/blob/master/libs/expat/Makefile#L11

Current available versions on `/sources`: 
[image]

thess, 2015-02-11:
Wiki says: "Source tar-balls in trunk are sometimes only found at the PKG_SOURCE_URL from the Makefile.". This is true more often than not.

@sbyx -- How/when is this repo updated? It seems pretty out-of-date.

sbyx, 2015-02-11:
Actually rarely these days, but i don't consider this an issue since the sources are identical to those from the original authors.

Usually someone adds a tarball to our own mirror if the original servers are (temporarily) down or if the original tarball was removed for some other reason. Sometimes sources are added if requested by a user (i.e. he/she has trouble reaching the original servers) or a .tar.gz is added when the original source is a (slow) svn or git repo especially close to a release.

closed 2015-02-11 by sbyx

cusspvz, 2015-02-11:
So shouldn't `PKG_SOURCE_URL` point to another location with some kind of fallback to `downloads.openwrt.org/sources` ?

sbyx, 2015-02-11:
The PKG_SOURCE_URL of expat is @SF/expat which is a macro which resolves to the sourceforge mirrors.

cusspvz, 2015-02-11:
If I'm not mistaken, my local repo is currently synced with last revision of `trunk/barrier_breaker` and it is pointing to `downloads.openwrt.org` instead on `expat` 's `Makefile`.

cusspvz, 2015-02-11:
I tough it was some typo error and as a temporary workaround for building I've changed `PKG_SOURCE_URL` to `2.0.1` as also checksum.

cusspvz, 2015-02-11:
As I don't have any experience related with OpenWRT compiling, yesterday was my first time on it, is `trunk/barrier_breaker` checking out a specific commit of this repo?