openwrt/packages #897: Compiling dvb_usb_lmedm04 kernel module as a KernelPackage

cusspvz, 2015-02-11:
I use OpenWRT since `White Russian` release, but I'm only giving my first steps building OpenWRT since about a week ago.

Lately i've been using a Fon 2b with OpenWRT but i'm changing to a Raspberry Pi for handling: DVB-S stream decoding (CA+ attached to it) and pipelining to Android/Kodi boxes on my house, mwan3 load balancing and proxy caching. I didn't wanted to use `Raspbian` since i'm used to OpenWRT and it is light as hell!! :+1: 

It happens that my last need is to compile `dvb_usb_lmedm04` kernel module.
Also it would be great if some of you give me some instructions about how to create it, maybe we could add it on this repo since DM04 (a shitty but cheap DVB-S receiver) is USB.

plntyk, 2015-02-12:
2 different solutions are possible:

1. Add config symbols to Kernel config / use make kernel_menuconfig
unclean - will probably not be integrated in releases

2. create new module build rules in
```
package/kernel/linux/modules/media.mk
```
media.mk because the modules are in drivers/media/usb/dvb-usb-v2 in Linux Kernel.
Look at other examples or submits in that directory for guidance.

Some quick research results (look at Kconfig and Makefiles) of Kernel
I extracted (most?) of the necessary information
```
CONFIG_DVB_USB_V2
depends on DVB_CORE && USB && I2C && (RC_CORE || RC_CORE=n)

if I2C && MEDIA_DIGITAL_TV_SUPPORT
	comment "Digital TV USB devices"
source "drivers/media/usb/dvb-usb-v2/Kconfig"

config DVB_USB_LME2510
	tristate "LME DM04/QQBOX DVB-S USB2.0 support"
	depends on DVB_USB_V2
	depends on RC_CORE
	select DVB_TDA10086 if MEDIA_SUBDRV_AUTOSELECT
	select DVB_TDA826X if MEDIA_SUBDRV_AUTOSELECT
	select DVB_STV0288 if MEDIA_SUBDRV_AUTOSELECT
	select DVB_IX2505V if MEDIA_SUBDRV_AUTOSELECT
	select DVB_STV0299 if MEDIA_SUBDRV_AUTOSELECT
	select DVB_PLL if MEDIA_SUBDRV_AUTOSELECT
	select DVB_M88RS2000 if MEDIA_SUBDRV_AUTOSELECT
	select DVB_TS2020 if MEDIA_SUBDRV_AUTOSELECT
	help
	  Say Y here to support the LME DM04/QQBOX DVB-S USB2.0

from Makefiles:
dvb-usb-lmedm04-objs := lmedm04.o
dvb_usb_v2-objs := dvb_usb_core.o dvb_usb_urb.o usb_urb.o
obj-$(CONFIG_DVB_USB_V2) += dvb_usb_v2.o
dvb-usb-lmedm04-objs := lmedm04.o
obj-$(CONFIG_DVB_USB_LME2510) += dvb-usb-lmedm04.o
```
So that media.mk should contain packages setting CONFIG_DVB_TS2020 etc...
Some symbols for Raspi are set via patches (bad practise) instead of trying to add these features for other platforms. Check the target/linux ... directory for differences.

cusspvz, 2015-02-13:
@plntyk thanks for your time!

Following your instructions i've made this:
```Makefile
MEDIA_MENU:=Multimedia

define KernelPackage/dvb-usb-lme2510
  SUBMENU:=$(MEDIA_MENU)
  TITLE:=LME DM04/QQBOX DVB-S USB2.0 support
  DEPENDS:=+kmod-usb-core
  KCONFIG:= \
	CONFIG_DVB_USB_V2=y \
	CONFIG_RC_CORE=y \
	CONFIG_DVB_USB_LME2510=y \
	CONFIG_DVB_TDA10086=y \
	CONFIG_DVB_TDA826X=y \
	CONFIG_DVB_STV0288=y \
	CONFIG_DVB_IX2505V=y \
	CONFIG_DVB_STV0299=y \
	CONFIG_DVB_PLL=y \
	CONFIG_DVB_M88RS2000=y \
	CONFIG_DVB_TS2020=y

  FILES:= \
	$(LINUX_DIR)/drivers/media/dvb-frontends/tda10086.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/tda826x.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/stv0288.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/ix2505v.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/stv0299.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/dvb-pll.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/m88rs2000.ko \
	$(LINUX_DIR)/drivers/media/dvb-frontends/ts2020.ko \
	\
	$(LINUX_DIR)/drivers/media/rc/rc-core.ko \
	$(LINUX_DIR)/drivers/media/rc/lirc_dev.ko \
	$(LINUX_DIR)/drivers/media/rc/keymaps/rc-lme2510.ko \
	\
	$(LINUX_DIR)/kernel/drivers/media/usb/dvb-usb-v2/dvb_usb_v2.ko \
	$(LINUX_DIR)/drivers/media/usb/dvb-usb-v2/dvb-usb-lmedm04.ko
endef

$(eval $(call KernelPackage,dvb-usb-lme2510))
```

> Some symbols for Raspi are set via patches (bad practise) instead of trying to add these features for other platforms. Check the target/linux ... directory for differences.

Thats what i've found, but don't know how to solve it. Any clue?

cusspvz, 2015-02-22:
ping @plntyk

diizzyy, 2017-06-07:
@thess @hnyman 
Close this, obsolete. 
If this needs to be fixed, submitter needs to open a new issue and/or a pull request.

closed 2017-06-07 by hnyman