openwrt/packages #899: no-ip.com missing in the ddns service

tedaz, 2015-02-12:
no-ip.com was in the service file previously, but not it is missing.
File: packages/net/ddns-scripts/files/services

hnyman, 2015-02-12:
Since a few days ago ( #862 ) , that service is in a separate packages that you have to install, too.
... ddns-scripts_No-IP_com  is the name, I think.
Package is visible e.g. in http://downloads.openwrt.org/snapshots/trunk/ar71xx/generic/packages/

chris5560, 2015-02-12:
have a look at #893
To minimize size of default package the scripts and needed files
for the services of No-IP.com and CloudFlare are inside separate packages.
Run make menuconfig and/or install ddns-scripts_No-IP_com or ddns-scripts_CloudFlare package.

tedaz, 2015-02-12:
@chris5560 Thanks for your reply. Found it, rebuilding.

closed 2015-02-12 by tedaz