openwrt/packages #909: LVM2.2.02.116 fails to build: missing blkid.h

ambrosa, 2015-02-15:
A week ago I've tried to build CC trunk@44257 and today again to build from scratch 44455
I've always an issue about LVM
Compile looks for blkid.h but it's not found.
```
make[5]: Leaving directory `/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/build_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/LVM2.2.02.116/libdaemon'
make -C lib
make[5]: Entering directory `/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/build_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/LVM2.2.02.116/lib'
mips-openwrt-linux-uclibc-gcc -c -I. -I../include -DHAVE_CONFIG_H -DDM_IOCTLS -Wall -Wundef -Wshadow -Wcast-align -Wwrite-strings -Wmissing-prototypes -Wmissing-declarations -Wnested-externs -Winline -Wmissing-noreturn -Wformat-security -Wredundant-decls -Wpointer-arith -Wuninitialized -Wmissing-include-dirs -Wfloat-equal -Wstrict-prototypes -Wold-style-definition -Wmissing-format-attribute -Wjump-misses-init -Wclobbered -Wempty-body -Wignored-qualifiers -Wmissing-parameter-type -Wold-style-declaration -Woverride-init -Wtype-limits -Wsync-nand -Wlogical-op -O2 -pipe -mno-branch-likely -march=34kc -mdsp -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -I/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/staging_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/usr/include -I/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/staging_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/include -I/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/staging_dir/toolchain-mips_34kc+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/media/ambrosa/ARCHIVIO/openwrt-TPLINK8970/staging_dir/toolchain-mips_34kc+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/include -fpic  device/dev-type.c -o device/dev-type.o
device/dev-type.c:25:19: fatal error: blkid.h: No such file or directory
 #include <blkid.h>
                   ^
compilation terminated.
```
I HAVE blkid.h and it's located into *staging_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/usr/include/blkid/* but this directory is not into -I directive.
To solve the problem I've simply copied 
*staging_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/usr/include/blkid/blkid.h*
into 
*staging_dir/target-mips_34kc+dsp_uClibc-0.9.33.2/usr/include/*
and build is successful.

Past build (i.e. 43982) I had not this issue.

plntyk, 2015-02-19:
Something is wrong with your build environment.

if you look at
https://github.com/openwrt/packages/blob/master/utils/lvm2/patches/001-include_fix.patch
you can see, that this patch is not applied on your side.

It is applied at my end and on buildbots (no build error on lvm2)
btw - nfs-kernel-server, btrfs-progs  use blkid/blkid.h include too

plntyk, 2015-02-19:
use a new check out or clean up your buildroot

ticket can probably be closed

thess, 2015-03-07:
Closed -- not reproducible/no response

closed 2015-03-07 by thess