openwrt/packages #931: Python 2.7.9 fails to compile (some modules)

enigmagroup, 2015-02-20:
https://github.com/openwrt/packages/commit/999cc6a04045ca98fb9ba56b2461bf3dfcd6ab80 - this patch "causes" the following error message:

```
strecent/staging_dir/host/usr/lib -L/usr/local/lib -o build/lib.linux-x86_64-2.7/_ctypes.so

Python build finished, but the necessary bits to build these modules were not found:
_bsddb             _curses            _curses_panel   
_sqlite3           _ssl               _tkinter        
bsddb185           bz2                dbm             
dl                 gdbm               imageop         
readline           sunaudiodev        zlib            
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

Failed to build these modules:
crypt              nis                                

make[4]: *** [sharedmods] Error 1
make[4]: Leaving directory `/home/x/openwrt-32.mostrecent/build_dir/host/Python-2.7.9'
make[3]: *** [/home/x/openwrt-32.mostrecent/build_dir/host/Python-2.7.9/.built] Error 2
make[3]: Leaving directory `/home/x/openwrt-32.mostrecent/feeds/packages/lang/python'
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
make[2]: Leaving directory `/home/x/openwrt-32.mostrecent'
make[1]: *** [/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/x/openwrt-32.mostrecent'
make: *** [world] Error 2
```

commodo, 2015-02-20:
So, this is expected behaviour.

The idea is that Python build should fail if any modules fail to build.
I would have hoped no modules fail to build, but as I see, it seems to be on the host side.
The host Python build varies from machine to machine.

Let me look into this, and I'll see what solution I'll come up with.
Maybe I'll disable those modules. Seems that buildroot disables quite a few on the host side.

Thanks
Alex

commodo, 2015-02-20:
Solved by this: https://github.com/openwrt/packages/pull/934

I'll investigate if host Python actually needs to be built.
I'm finding out that host Python is installed on buildbot/buildroot, so we may want to use that for determinism.

commodo, 2015-03-02:
@enigmagroup or @sbyx 
Any reason to keep this issue open ?

closed 2015-03-02 by sbyx