openwrt/packages #935: tor fails to compile since seccomp was added (buildbot)

plntyk, 2015-02-20:
```
checking for seccomp.h... yes
checking for library containing seccomp_init... -lseccomp
...later...
src/common/sandbox.c: In function 'sb_mmap2':
src/common/sandbox.c:345:58: error: 'MAP_STACK' undeclared (first use in this function)
        SCMP_CMP(3, SCMP_CMP_EQ,MAP_PRIVATE|MAP_ANONYMOUS|MAP_STACK));
```
reported by:
http://buildbot.openwrt.org:8010/broken_packages/ar71xx/tor/compile.txt

This is some cross compile / build system issue / possibly target issue
"-lseccomp" detection works but that flag is not used in compilation process later.

Unclear for me if tor on MIPS needs some sandbox.c changes like arm does
( https://gitweb.torproject.org/tor.git/commit/src/common/sandbox.c?id=b0c1c700114aa8d4dfc180d85870c5bbe15fcacb )

champtar, 2015-03-11:
@nmav, maybe libseccomp should depend on KERNEL_SECCOMP?

nmav, 2015-03-12:
Done

closed 2015-03-12 by nmav

dangowrt, 2015-03-15:
This still doesn't account for tor depending on libseccomp if configure finds it.... So tor build still breaks because of that missing dependency now. I reckon this could be fixed in tor/Makefile by depending on +KERNEL_SECCOMP:libseccomp.

commit e3199d1f (2015-03-12): seccomp: depends on CONFIG_KERNEL_SECCOMP

Resolves #935
commit 95aef029 (2015-04-29): tor: disable seccomp support

seccomp is only supported on x86, amd64 and arm in tor.
This deactivated it currently completely which should close #935,
 #1097, #1147 and #1161.