openwrt/packages #936: libseccomp contains not needed Build/Compile (review)

plntyk, 2015-02-20:
For which purpose are the 2 SED ?
```
define Build/Configure
$(SED) 's,-I$$$${includedir},,g' $(PKG_BUILD_DIR)/configure
$(SED) 's,-L$$$${libdir},,g' $(PKG_BUILD_DIR)/configure
$(call Build/Configure/Default)
endef
```
SED does not search&replace anything - at least at my end.
There should be no need to SED around in configure since libseccomp seems to use autotools. (PKG_FIXUP:=autoreconf).

Maybe some hint or documentation regarding Kernel Version / Kernel support needed and arch is necessary. (OpenWrt has some PowerPC support like TPLink WDR4900 mpc85xx )

nmav, 2015-02-27:
I believe the SED values are legacy cruft. I'll remove it.

About documentation, currently my goal is to push for seccomp to be included by default in openwrt kernels. After that, it would make sense to add some documentation in using it. Otherwise, if seccomp is not added in openwrt's kernels the package will not be so useful.

https://lists.openwrt.org/pipermail/openwrt-devel/2015-February/031297.html

champtar, 2015-03-11:
partly done in 095ae50d353a8534c00982ee8c83fe0767ac9206
@nmav, i think you can remove all the Build/Configure part

champtar, 2015-03-19:
done (thanks @nmav)

closed 2015-03-19 by champtar