openwrt/packages #946: Pulseaudio not starting with --system flag

jan233321, 2015-02-22:
Hi All,
I would like to use my router as a remote sound base connected to near HiFi. I'm folowing howto from OpenWrt [wiki](http://wiki.openwrt.org/doc/howto/pulseaudio) , everything went fine except pulseaution start on router.

Using Barrier Breaker 14.07

command:
```
/usr/bin/pulseaudio --system --disallow-exit --disallow-module-loading --disable-shm --exit-idle-time=-1 --realtime=false -vvv
```
gives:
```
D: [pulseaudio] core-rtclock.c: Timer slack is set to 50 us.
D: [pulseaudio] core-util.c: setpriority() worked.
I: [pulseaudio] core-util.c: Successfully gained nice level -11.
I: [pulseaudio] main.c: Found user 'pulse' (UID 51) and group 'pulse' (GID 51).
I: [pulseaudio] main.c: Successfully changed user to "pulse".
I: [pulseaudio] main.c: This is PulseAudio 5.0
D: [pulseaudio] main.c: Compilation host: mips-openwrt-linux-gnu
D: [pulseaudio] main.c: Compilation CFLAGS: -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -I/BB/build/ar71xx/generic/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libiconv-stub/include -I/BB/build/ar71xx/generic/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib/libintl-stub/include -std=gnu99  -Wall -W -Wextra -Wno-long-long -Wno-overlength-strings -Wunsafe-loop-optimizations -Wundef -Wformat=2 -Wlogical-op -Wsign-compare -Wformat-security -Wmissing-include-dirs -Wformat-nonliteral -Wpointer-arith -Winit-self -Wdeclaration-after-statement -Wfloat-equal -Wmissing-prototypes -Wredundant-decls -Wmissing-declarations -Wmissing-noreturn -Wshadow -Wendif-labels -Wcast-align -Wstrict-aliasing -Wwrite-strings -Wno-unused-parameter -ffast-math -fno-common -fdiagnostics-show-option
D: [pulseaudio] main.c: Running on host: Linux mips 3.10.49 #3 Wed Oct 1 14:00:51 CEST 2014
D: [pulseaudio] main.c: Found 1 CPUs.
I: [pulseaudio] main.c: Page size is 4096 bytes
D: [pulseaudio] main.c: Compiled with Valgrind support: no
D: [pulseaudio] main.c: Running in valgrind mode: no
D: [pulseaudio] main.c: Running in VM: no
D: [pulseaudio] main.c: Optimized build: yes
D: [pulseaudio] main.c: FASTPATH defined, only fast path asserts disabled.
I: [pulseaudio] main.c: Machine ID is OpenWrt.
I: [pulseaudio] main.c: Using runtime directory /var/run/pulse.
E: [pulseaudio] core-util.c: Failed to create secure directory (/var/lib/pulse): Permission denied
```

the directory /var is symlinked to /tmp as default in this device

```
ls -l /
drwxr-xr-x    2 root     root           666 Oct  2 07:09 bin
drwxr-xr-x    6 root     root          1200 Feb 22 10:36 dev
drwxr-xr-x    1 root     root             0 Feb 22 10:45 etc
-rwxr-xr-x    1 root     root            78 Sep 20 19:47 init
drwxr-xr-x    1 root     root             0 Feb 22 10:45 lib
drwxr-xr-x    2 root     root             3 Oct  1 11:57 mnt
drwxr-xr-x    9 root     root             0 Jan  1  1970 overlay
dr-xr-xr-x   44 root     root             0 Jan  1  1970 proc
drwxr-xr-x   16 root     root           223 Oct  2 07:09 rom
drwxr-xr-x    1 root     root             0 Feb 22 15:54 root
drwxr-xr-x    1 root     root             0 Feb 22 10:45 sbin
dr-xr-xr-x   11 root     root             0 Jan  1  1970 sys
drwxrwxrwt   15 root     root           380 Feb 22 16:02 tmp
drwxr-xr-x    1 root     root             0 Sep 23 06:56 usr
lrwxrwxrwx    1 root     root             4 Oct  2 07:09 var -> /tmp
drwxr-xr-x    1 root     root             0 Sep 23 06:18 www
```

```
ls -l /var/lib/
drwx------    2 pulse    pulse           40 Feb 22 15:55 pulse
```

Any ideas?

Thank you

Jan.

thess, 2015-02-22:
As far as I can tell - the default init script in the package should create this directory and set its permissions and ownership correctly. If you are using an alternative startup script you will have to take care of these steps yourself.

jan233321, 2015-02-23:
Default script in /etc/init.d/pulseaudio looks like this:

```
#!/bin/sh /etc/rc.common
# Copyright (C) 2011 OpenWrt.org

START=65
STOP=65

USE_PROCD=1
PROG=/usr/bin/pulseaudio

start_service() {
        user_exists pulse 51 || user_add pulse 51
        group_exists pulse 51 || group_add pulse 51
        [ -d /var/run/pulse ] || {
                mkdir -m 0755 -p /var/run/pulse
                chmod 0750 /var/run/pulse
                chown pulse:pulse /var/run/pulse
        }
        [ -d /var/lib/pulse ] || {
                mkdir -m 0755 -p /var/lib/pulse
                chmod 0750 /var/lib/pulse
                chown pulse:pulse /var/lib/pulse
        }

        chown root:pulse /dev/snd/* /dev/mixer /dev/dsp
        chmod 664 /dev/snd/* /dev/mixer /dev/dsp

        procd_open_instance
        procd_set_param command $PROG --system --disallow-exit --disallow-module-loading --disable-shm --exit-idle-time=-1 --realtime=false
        procd_close_instance
}
```

when I run it as-is, then it ends with message mentioned in my original bugreport. When I change run command, so it looks like this:

```
procd_set_param command $PROG  --disallow-module-loading --disable-shm --exit-idle-time=-1 --realtime=false
```

then it run just fine.

If this is a solution (I do not really see the need of --system flag in a router) I will close the issue.

jan233321, 2015-02-23:
/var is symlink to /tmp -> may this lead to the problem with the "secure directory"?

thess, 2015-02-23:
No quick answer yet. It works OK on my test system using CC.

The symlink doesn;t matter and you should be able to run with --system which is appropriate for an embedded system. The only thing I can think of at the moment is be sure that /var/run and /var/lib are not pointing to the same directory.

thess, 2015-03-04:
Without any more information about your problem and my being unable to reproduce your problem, I am closing this issue for now. If you are still having a problem starting pulseaudio, please re-open this and post some more information about your system & configuration.

Check to see if you have any pulse configureation path changes from the default in either /root or /etc/pulse. Try:
> rm -rf /var/lib/pulse
> rm -rf /var/run/pulse

... .before starting service if any of these are persisted over reboot.

closed 2015-03-04 by thess