openwrt/packages #970: Please update wview package to latest source (wview_5.21.7 / may 2014)

martlaak, 2015-03-01:
Weather station package wview port to OpenWrt is working but rather old (wview_5.19.0-jgoerzen / 2012). Please somebody update it to wview_5.21.7 (may 2014).

Attached are patches for the wview_5.21.7, for files:
Makefile
patches/010-fix-cross-compile.patch
patches/020-use-standard-awk.patch
patches/040-WMRUSB-process-absent-dewpoint.patch

NB! patches/050-WMRUSB-fix-segfault-buffer-overflow.patch is not needed anymore as it is incorporated in wview source now.

Regards,
Märt Laak

---
diff -ruN wview-5.19.0-jgoerzen/Makefile wview-5.21.7/Makefile
--- wview-5.19.0-jgoerzen/Makefile	2015-02-28 17:25:53.595765600 +0200
+++ wview-5.21.7/Makefile	2015-02-23 16:59:28.808672200 +0200
@@ -8,12 +8,12 @@
 include $(TOPDIR)/rules.mk
 
 PKG_NAME:=wview
-PKG_VERSION:=5.19.0-jgoerzen
-PKG_RELEASE=$(PKG_SOURCE_VERSION)-r2
-PKG_SOURCE_URL:=git://github.com/jgoerzen/wview.git
-PKG_SOURCE_VERSION:=7bfac6c11e756290c38e7b5862a4c51b6bc6c51e
-PKG_SOURCE_SUBDIR:=$(PKG_NAME)-$(PKG_VERSION)
-PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION)-$(PKG_SOURCE_VERSION).tar.gz
+PKG_VERSION:=5.21.7
+PKG_RELEASE:=1
+
+PKG_SOURCE:=$(PKG_NAME)-$(PKG_VERSION).tar.gz
+PKG_SOURCE_URL:=@SF/$(PKG_NAME)
+PKG_MD5SUM:=a296932574ffb4a3855e505a73daf9c5
 
 PKG_FIXUP:=autoreconf
 
@@ -22,7 +22,7 @@
 define Package/wview
   SECTION:=utils
   CATEGORY:=Utilities
-  DEPENDS:=+libcurl +zlib +libpng +libgd +libreadline +libsqlite3 +libusb-1.0 +radlib +sqlite3-cli +libopenssl
+  DEPENDS:=+libcurl +zlib +libpng +libgd +libreadline +libsqlite3 +libusb-1.0 +radlib +sqlite3-cli +libopenssl +udev
   TITLE:=Wview Weather Station Software
   URL:=http://www.wviewweather.com/
 endef
diff -ruN wview-5.19.0-jgoerzen/patches/010-fix-cross-compile.patch wview-5.21.7/patches/010-fix-cross-compile.patch
--- wview-5.19.0-jgoerzen/patches/010-fix-cross-compile.patch	2015-02-28 17:25:53.579275500 +0200
+++ wview-5.21.7/patches/010-fix-cross-compile.patch	2015-02-21 21:37:40.157653500 +0200
@@ -12,7 +12,7 @@
  wvalarmd_LDADD   =
  
  # define library directories
--wvalarmd_LDFLAGS = -L$(prefix)/lib -L$(prefix)/usr/lib -L/usr/lib
+-wvalarmd_LDFLAGS = -L$(prefix)/lib -L/usr/lib
  
  if CROSSCOMPILE
 -wvalarmd_LDFLAGS += $(prefix)/lib/crt1.o $(prefix)/lib/crti.o $(prefix)/lib/crtn.o
@@ -32,7 +32,7 @@
  wvcwopd_LDADD   =
  
  # define library directories
--wvcwopd_LDFLAGS = -L$(prefix)/lib -L$(prefix)/usr/lib -L/usr/lib
+-wvcwopd_LDFLAGS = -L$(prefix)/lib -L/usr/lib
  
  if CROSSCOMPILE
 -wvcwopd_LDFLAGS += $(prefix)/lib/crt1.o $(prefix)/lib/crti.o $(prefix)/lib/crtn.o
@@ -52,7 +52,7 @@
  wviewftpd_LDADD   = -lssl -lcurl
  
  # define library directories
[... 210 lines cut ...]
              return;
          }
  
diff -ruN wview-5.19.0-jgoerzen/patches/050-WMRUSB-fix-segfault-buffer-overflow.patch wview-5.21.7/patches/050-WMRUSB-fix-segfault-buffer-overflow.patch
--- wview-5.19.0-jgoerzen/patches/050-WMRUSB-fix-segfault-buffer-overflow.patch	2015-02-28 17:25:53.549293500 +0200
+++ wview-5.21.7/patches/050-WMRUSB-fix-segfault-buffer-overflow.patch	1970-01-01 03:00:00.000000000 +0300
@@ -1,18 +0,0 @@
-Index: wview-5.19.0-jgoerzen/stations/WMRUSB/wmrusbprotocol.c
-===================================================================
---- wview-5.19.0-jgoerzen.orig/stations/WMRUSB/wmrusbprotocol.c	2013-03-10 22:24:28.000000000 +0400
-+++ wview-5.19.0-jgoerzen/stations/WMRUSB/wmrusbprotocol.c	2013-03-10 22:25:01.000000000 +0400
-@@ -897,8 +897,11 @@
- // Read raw USB data and buffer it for later processing:
- void wmrReadData (WVIEWD_WORK *work, WMRUSB_MSG_DATA* msg)
- {
--    memcpy(&wmrWork.readData[wmrWork.readIndex], msg->data, msg->length);
--    wmrWork.readIndex += msg->length;
-+    if (wmrWork.readIndex + msg->length <= WMR_BUFFER_LENGTH)
-+    {
-+        memcpy(&wmrWork.readData[wmrWork.readIndex], msg->data, msg->length);
-+        wmrWork.readIndex += msg->length;
-+    }
- 
-     return;
- }

champtar, 2015-03-11:
hi @martlaak 
please send a pull request

thess, 2015-03-12:
ref: #967 
Looking for a volunteer...

closed 2015-07-27 by champtar

hayden-t, 2016-02-28:
hey, im curious about all this, im in a pickle, want to compile this for mipsel, have tried the sdk approach, but as discussed here its not in the feeds anymore, tried with the cross compile toolchain made, but radlib complains of stub.h missing, which afaik is part of glibc which has been replaced by musl , and in the advanced dev option in make menuconfig there is no options at all, such as the use uclibc instead of musl ??