openwrt/packages #977: openconnect: Script '/lib/netifd/vpnc-script' returned error 1

tillwo, 2015-03-02:
I am unable to establish a connection to a Cisco VPN server using openconnect. Getting no error at all using the LuCI web interface, I tried running openconnect on the command line and got the error message `Script '/lib/netifd/vpnc-script' returned error 1`:

    root@OpenWrt:~# openconnect vpn.mydomain.com
    WARNING: This version of openconnect was built without iconv
             support but you appear to be using the legacy character
             set "ASCII". Expect strangeness.
    POST https://vpn.mydomain.com/
    Attempting to connect to server XXX.XXX.XXX.XXX:443
    SSL negotiation with vpn.mydomain.com
    Server certificate verify failed: signer not found
    
    Certificate from VPN server "vpn.mydomain.com" failed verification.
    Reason: signer not found
    Enter 'yes' to accept, 'no' to abort; anything else to view: yes
    Connected to HTTPS on vpn.mydomain.com
    XML POST enabled
    Please enter your username and password.
    GROUP: XXX
    POST https://vpn.mydomain.com/
    XML POST enabled
    Please enter your username and password.
    Username:XXX
    Password:
    POST https://vpn.mydomain.com/
    Got CONNECT response: HTTP/1.1 200 OK
    CSTP connected. DPD 30, Keepalive 20
    Script '/lib/netifd/vpnc-script' returned error 1
    Script '/lib/netifd/vpnc-script' returned error 1
    Connected tun0 as XXX.XXX.XXX.XXX, using SSL
    Established DTLS connection (using GnuTLS). Ciphersuite (DTLS0.9)-(RSA)-(AES-256-CBC)-(SHA1).

The error message seems to be caused by the following code in `/lib/netifd/vpnc-script`:

    […]
    #### Main
    
    if [ -z "$reason" ]; then
    	logger -t openconnect "this script must be called from vpnc" 1>&2
    	exit 1
    fi
    if [ -z "$INTERFACE" ]; then
    	logger -t openconnect "this script must be called for an active interface"
    	exit 1
    fi
    […]

I don't know if this is a bug in openconnect or vpnc-scripts (or neither). Hopefully, this is the right place to report this.

nmav, 2015-03-08:
The vpnc-script included in openwrt doesn't work if you run it from command line. It is made to be run from netifd.

closed 2015-03-08 by nmav

hansbogert, 2015-03-18:
Still I have the same issue that I get no feedback at all in syslog when connecting to luchtbrug.tudelft.nl.

nmav, 2015-03-19:
You can check the syslog with "logread -f|grep openconnect". If you don't see any feedback then most probably openconnect isn't running.

hansbogert, 2015-03-19:
Indeed I don't see anything when trying to (dis)connect using logread. I've filled in all information through luci in the interface section. No feedback in luci as well.

marcosscriven, 2015-06-17:
Sorry to tag onto this - more of a request for help. Like @hansbogert I was able to run this from the command line, and am correctly prompted for my password, and second 'password', which is just 'push' for Duo 2FA.

Here's the config I'm using when trying to set it up by netifd:
```
config interface 'vpn'
        option proto 'openconnect'
        option interface 'wan'
        option server 'vpn.whatever.com'
        option port '80'
        option username 'username'
        option password 'realpassword'
        option serverhash 'sha1:shafromcert'
        # option token_mode 'totp'
        # option token_secret 'push'
        option defaultroute '0'
        option authgroup 'DEFAULT'
```

When I do that in logread I see:
```
Wed Jun 17 21:35:17 2015 daemon.notice netifd: Interface 'vpn' is setting up now
Wed Jun 17 21:35:17 2015 user.notice openconnect: initializing...
Wed Jun 17 21:35:17 2015 user.notice openconnect: executing 'openconnect vpn.whatever.com:80 -i vpn-vpn --non-inter --syslog --script /lib/netifd/vpnc-script  --servercert=sha1:shafromcert --no-system-trust --authgroup DEFAULT -u username --passwd-on-stdin'
Wed Jun 17 21:35:17 2015 daemon.notice netifd: vpn (3274): WARNING: This version of openconnect was built without iconv
Wed Jun 17 21:35:17 2015 daemon.notice netifd: vpn (3274):          support but you appear to be using the legacy character
Wed Jun 17 21:35:17 2015 daemon.notice netifd: vpn (3274):          set "ASCII". Expect strangeness.
Wed Jun 17 21:35:17 2015 daemon.info openconnect[3276]: POST https://vpn.whatever.com:80/
Wed Jun 17 21:35:17 2015 daemon.info openconnect[3276]: Attempting to connect to server 122.122.122.0:80
Wed Jun 17 21:35:17 2015 daemon.info openconnect[3276]: SSL negotiation with vpn.whatever.com
Wed Jun 17 21:35:17 2015 daemon.notice openconnect[3276]: SSL connection failure: The TLS connection was non-properly terminated.
Wed Jun 17 21:35:17 2015 daemon.notice openconnect[3276]: Failed to open HTTPS connection to vpn.whatever.com
Wed Jun 17 21:35:17 2015 daemon.notice netifd: vpn (3274): Failed to obtain WebVPN cookie
Wed Jun 17 21:35:17 2015 user.notice openconnect: bringing down openconnect
Wed Jun 17 21:35:17 2015 daemon.notice netifd: Interface 'vpn' is now down
```

What I don't understand is how I'm meant to set the second password as 'push' with the netifd method?

(NB obviously I replaced the details in this comment with dummy placeholders)

marcosscriven, 2015-06-17:
@nmav - actually, the first issue was simple to fix, just dropped ```option port '80'``` from the config. 

However, there's still no way to pass in a second password I think?

marcosscriven, 2015-06-18:
Further to this, I got it working by dropping the ```--non-inter``` option in openconnect.sh, and appending ```push``` to the pwfile.

nmav, 2015-06-18:
Please open a different issue for the second factor auth. I'm pretty lost here.

marcosscriven, 2015-06-18:
@nmav - yep, have done https://github.com/openwrt/packages/issues/1419

lordarthas1995, 2019-03-16:
> The vpnc-script included in openwrt doesn't work if you run openconnect from command line. It is made to be run from netifd.

How to run it from netifd? I try running it in Luci and it show me not connected

liuxindtc, 2021-01-22:
About "Script '/lib/netifd/vpnc-script' returned error 1" .
According to my way:
openconnect --script **./vpnc-script**  myvpn.com:443

And the vpnc-script is got from [](https://www.infradead.org/openconnect/vpnc-script.html)

iluffk, 2025-03-01:
Thanks! @liuxindtc