openwrt/packages #981: Luci-app-sqm - throttling of wifi not working correctly

bouwew, 2015-03-03:
I have SQM active on eth0.2, which is my WAN. It is set up for 120Mbit/s down, 12Mbit/s up. Working as expected: ~110Mbit/s down, ~11MBit/s up.
I have added a 2nd entry, active on WLAN1-2, my guest wifi network, to throttle the up- and downloadspeed of the guest-devices. It is set to ~5Mbit/s down, ~1 Mbit/s up.
I noticed, while testing using speedtest.net, that entering 5000 for download speed and 1000 for upload speed, the download speed would be 1Mbits/s and the upload speed > 10 Mbit/s.
I played around a bit more and the results shows clearly that the value for upload speed in the 2nd entry determines the download speed on the guest wifi network. Bug? Typo? Copy-paste-error?

I'm on LuCI Branch (git-15.051.48491-4137afe) / OpenWrt Chaos Calmer r44590, luci-app-sqm 3-1.

moeller0, 2015-03-07:
> I'm on LuCI Branch (git-15.051.48491-4137afe) / OpenWrt Chaos Calmer r44590, luci-app-sqm 3-1, sqm-scripts 7-3.
> @tohojo @moeller0

Hi Bouwew,

> I have SQM active on eth0.2, which is my WAN. It is set up for 120Mbps down, 12Mbps up. Working as expected: ~110Mbps down, ~11MBps up.

	What hardware are you running on? Most router�s can only reliably shape up to ~60Mbps total traffic, your 132Mbps certainly requires something beefier than the default oldish MIPSes...

> I have added a 2nd entry, active on WLAN1-2, my guest wifi network, to throttle the up- and downloadspeed of the guest-wifi-devices. It is set to ~5Mbps down, ~1Mbps up.
> I noticed, while testing using speedtest.net, that entering 5000 for download speed and 1000 for upload speed, the download speed would be 1Mbps and the upload speed > 10 Mbps.
> I played around a bit more and the results shows clearly that the value for upload speed in the 2nd entry determines the maximum download speed on the guest wifi network.

	This is to be expected; ingress and egress in the GUI refer to the router�s perspective. In case of shaping on wan your router�s view of ingress and egress is aligned with your wireless client�s view: egress is out to the internet, ingress is from the internet. In case of shaping the connection between the router and a client directly, the router's ingress is your client�s egress and vice versa. I happily agree that this apparent �reversal� is a bit puzzling at first, but once you think it through it is quite logical and reasonable. Maybe we could document that a bit better, but the behavior is as expected and as it should be IMHO.
	Now, why your observed upload was > 10Mbps instead of the 5Mbps you specified I have no good explanation for except wifi buffering often is weird...

Best Regards
	Sebastian

bouwew, 2015-03-07:
Dear Sebastian,

Thank you for your explanation! I now understand what is going on and why.
Btw, I'm running Openwrt on a WDR4900 V1. As far as I understand, a pretty powerful router.
I will close the issue, since there is no issue, just a lack of information :)
That why we experiment...

Best regards, 
Bouwe

closed 2015-03-07 by bouwew

moeller0, 2015-03-07:
Dear Bouwe,

> Dear Sebastian,
> 
> Thank you for your explanation! I now understand what is going on and why.

	Happy to be of help ;)

> Btw, I'm running Openwrt on a WDR4900 V1. As far as I understand, a pretty powerful router.

	Yes, I should have gotten one while they were still available, the ppc sox in those seems to be simply in a different league than most of the cheap MIPS socs. Too bad, that ppc seems to be on the way out of routers...

> I will close the issue, since there is no issue, just a lack of information :)

	Thanks! By the way, I fully understand the confusion around this issue. When we played around with the GUI for sqm-scripts we had a small discussion about how to name the fields and opted for download and upload speed to make it easy to understand for the default case of just shaping on the WAN interface; then we snug in �ingress� and �egress� as additional labels to make it slightly easier to understand the situation you came up with. Showing that the iddue is subtle and probably would warrant better explanations (but we have no real good place to put such infuriation to begin with...)

> That why we experiment...
> 
> Best regards, 
> Bouwe

Best Regards
	Sebastian

wpietri, 2016-07-29:
Thanks for working this out in public. I was struggling with this exact problem.

And thanks for a handy piece of software. I like running an open wireless node so that my neighbors have a backup if they're having problems. But after years without problems, somebody started doing big uploads, killing my interactive shells. This fixed things right up for me!