openwrt/packages #986: Mailsend package not compiled with ssl

makedir, 2015-03-04:
... making it useless. Description says "Mailsend is a simple command line program to send mail via SMTP protocol. SSL is supported via OpenSSL (optional).", but it's not optional, if it's not compiled with in the first place.

plntyk, 2015-03-04:
If you select the SSL option, support will be compiled into the package.
Many packages provide functionality that is not active by default and you have to build by yourself to get this feature.

SSL probably is not included by default because of openssl size.
mailsend is ~40kb alone and with SSL you get libopenssl with additional requiring ~900kB.

makedir, 2015-03-04:
What option...? I am talking about the trunk package. It is not compiled against ssl. Most people (I guess 98%) have libopenssl  and mostly ALSO libpolarssl TOGETHER on their routers.

thess, 2015-03-04:
The default package does not have SSL support. If you build it yourself using the OpenWrt builtroot, menuconfig give you the option of SSL support which automatically selects and includes libopenssl.

That said, I will consider this report as a request for an SSL variant automatically built in addtion to the non-SSL version and have both packages available for download.

thess, 2015-03-04:
OK - create 2 mailsend packages (with & without SSL). See: 0b2e43d07b45fb45fe1b8b0dbd89b7a39a985d59

Hopefully buildbot will return eventually.

closed 2015-03-04 by thess