openwrt/packages #996: Python: File not found (_bsddb.so), build failed

enigmagroup, 2015-03-07:
This happens since today, in all architectures tested so far (x86, x86_64, sunxi).

I also made ```make dirclean``` and restarted the build, same result.

```
rstrip.sh: /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/usr/lib/python2.7/lib-dynload/time.so: shared object
rstrip.sh: /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/usr/lib/python2.7/lib-dynload/operator.so: shared object
rstrip.sh: /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/usr/lib/python2.7/lib-dynload/bz2.so: shared object
rstrip.sh: /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/usr/lib/python2.7/lib-dynload/_io.so: shared object
rstrip.sh: /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/usr/lib/python2.7/lib-dynload/resource.so: shared object
(cd /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 prerm;  )
install -d -m0755 /home/x/openwrt-32.mostrecent/bin/x86/packages/packages
/home/x/openwrt-32.mostrecent/staging_dir/host/bin/ipkg-build -c -o 0 -g 0 /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light /home/x/openwrt-32.mostrecent/bin/x86/packages/packages
Packaged contents of /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-light into /home/x/openwrt-32.mostrecent/bin/x86/packages/packages/python-light_2.7.9-5_x86.ipk
mkdir -p /home/x/openwrt-32.mostrecent/bin/x86/packages /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/python-bsddb/CONTROL /home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo
find /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
copying: '/usr/lib/python2.7/bsddb'
File not found '/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/_bsddb.so'
make[3]: *** [/home/x/openwrt-32.mostrecent/bin/x86/packages/packages/python-bsddb_2.7.9-5_x86.ipk] Error 1
make[3]: Leaving directory `/home/x/openwrt-32.mostrecent/feeds/packages/lang/python'
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory `/home/x/openwrt-32.mostrecent'
make[1]: *** [/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/x/openwrt-32.mostrecent'
make: *** [world] Error 2
```

commodo, 2015-03-07:
So, I do not get any errors on this.
I guess I'd have to spin up a Debian VM and test with that.
Maybe it's some other cross-compilation issue on Debian or Debian-like systems.

I re-tested on my OpenSUSE x86_64 machine and it worked.
I tried with a clean checkout of openwrt + packages feed.
Will come back later with findings.

commodo, 2015-03-07:
Built on Debian 7 x86_64.
I'd need a bit more input on how to reproduce this.

Thanks :)

enigmagroup, 2015-03-07:
Okay hmm. I will rm -rf my git.openwrt.org dir and restart from scratch. Stay tuned.

commodo, 2015-03-08:
I was thinking.
Maybe that `_bsddb.so` module was not built before, but now since it's packetized separately, it's showing that it failed to build.
Python C modules that fail to build, are just ignored and not installed; I wanted to change that, but that caused other issues, with the host Python build (like this https://github.com/openwrt/packages/issues/931)

After the build finishes, and it still fails, could you show me the output from these commands:
# `ls /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/`
# `ls /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86` (or whatever target you're building for
# `ls /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86/bsddb` 

Also, if you could save the output of the build, and check for the part where the Python C extensions get built, that would show what exactly is failing during the build of `_bsddb.so`.
Running `make &> output.txt` will save all the output to file for easier access.

Thanks

enigmagroup, 2015-03-08:
Build failed with a clean checkout. (This time, arch is x86_64).

Ok, here the output:

ls [...]/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/:
```
array.so     cmath.so            _codecs_kr.so    cStringIO.so      _curses.so       future_builtins.so  _hotshot.so   _lsprof.so           operator.so                  readline.so  _sqlite3.so  termios.so
audioop.so   _codecs_cn.so       _codecs_tw.so    _csv.so           datetime.so      gdbm.so             _io.so        math.so              parser.so                    resource.so  _ssl.so      _testcapi.so
binascii.so  _codecs_hk.so       _collections.so  _ctypes.so        _elementtree.so  grp.so              itertools.so  mmap.so              pyexpat.so                   select.so    strop.so     time.so
_bisect.so   _codecs_iso2022.so  cPickle.so       _ctypes_test.so   fcntl.so         _hashlib.so         _json.so      _multibytecodec.so   Python-2.7.9-py2.7.egg-info  _socket.so   _struct.so   unicodedata.so
bz2.so       _codecs_jp.so       crypt.so         _curses_panel.so  _functools.so    _heapq.so           _locale.so    _multiprocessing.so  _random.so                   spwd.so      syslog.so
```

ls [...]/Python-2.7.9/ipkg-x86_64:
```
python-base  python-bsddb  python-light
```
(interesting. /Python-2.7.9/ipkg-x86_64/python-bsddb/ has also .py files in it)

ls [...]/Python-2.7.9/ipkg-x86/bsddb
```
ls: cannot access build_dir/target-x86_64_uClibc-0.9.33.2/Python-2.7.9/ipkg-x86_64/bsddb: No such file or directory
```

commodo, 2015-03-08:
So, the `_bsddb.so` file failed to build in your case.
Btw; my bad; last ls command was supposed to be `ls [...]/Python-2.7.9/ipkg-x86/python-bsddb`
Doesn't matter now, since the `_bsddb.so` module most likely failed to compiled/link.

Could you find in your log output what exactly failed to compile/link/build ?

enigmagroup, 2015-03-08:
Ok sure, let me check. This is built in the regular python package, right?

commodo, 2015-03-08:
Yep.
The main Makefile builds the entire Python build, and the rest of the install rules pick out what to install/package.

enigmagroup, 2015-03-08:
Alright. Here's the output.log: https://enigmabox.net/tmp/python-bsddb-output.log.gz

Collected using the following command: ```make package/python/{clean,compile} V=99 2>&1 | tee output.log```

commodo, 2015-03-08:
So, in your case, you're using ccache apparently, and not the vanilla build.
And there's a flag option that does not seem to be supported.
Error is
`i486-openwrt-linux-uclibc-gcc: error: unrecognized command line option '-R'`

I wonder who adds that option; I don't have it.
I'd say that's why the build fails for `_bsddb.so` and `dbm.so` apparently.

```
ccache_cc -shared -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -fno-strict-aliasing -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable build/temp.linux2-i486-2.7/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_bsddb.o -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/i486-openwrt-linux-uclibc/lib -L. -R/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -ldb-4.7 -lpython2.7 -o build/lib.linux2-i486-2.7/_bsddb.so
```

commodo, 2015-03-08:
Seems the Python setup.py script adds that -R option.
I'll check to see about it.
Maybe you have something installed on your system that the setup.py script detects and includes.

enigmagroup, 2015-03-08:
Uuh. Ok.
Thank you for investigating!

commodo, 2015-03-08:
Could you test these patches ?
http://rank24.net/~sandu/openwrt/

I think the first patch should do it, but I might consider adding the second one 2 too, to prevent further issues like this.
So, if you test only the first patch it would be sufficient.

Thanks

enigmagroup, 2015-03-08:
I've put them into this dir: ```feeds/packages/lang/python/patches/```, correct?

Wonderful! It compiles, with both patches. Output log: https://enigmabox.net/tmp/output-python-afterpatch.log.gz

commodo, 2015-03-08:
Yep, that folder is correct.
I'll update the patches then into the packages tree.

Thanks for testing

enigmagroup, 2015-03-08:
Awesome. Thanks for the fix!

closed 2015-03-08 by enigmagroup

sixelaa, 2015-03-11:
Hello,

I try to compile Python 2.7.9 too and I have the same problem in spite of the patches.

copying: '/usr/lib/python2.7/bsddb'
File not found '/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/_bsddb.so'
make[2]: *** [/home/dir/openwrt/bin/x86/packages/python-bsddb_2.7.9-5_x86.ipk] Error 1

commodo, 2015-03-11:
Could you show me your complete output ?
There should be a reason why this failed to build.

sixelaa, 2015-03-11:
make[2]: Entering directory `/home/dir/openwrt/feeds/sagemcom/lang/python'
mkdir -p /home/dir/openwrt/bin/x86/packages /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-x86/python-bsddb/CONTROL
find /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
copying: '/usr/lib/python2.7/bsddb'
File not found '/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/_bsddb.so'
make[2]: *** [/home/dir/openwrt/bin/x86/packages/python-bsddb_2.7.9-5_x86.ipk] Error 1
make[2]: Leaving directory `/home/dir/openwrt/feeds/packages/lang/python'

commodo, 2015-03-11:
I was hoping something a bit more complete like:
http://olegtown.pw/openwrt/r44651/broken_packages/build_simplejson.txt

Or an output log like enigmaroup added.
The last lines don't really show the cause of the failure.
The failure is somewhere earlier.

sixelaa, 2015-03-11:
```
running install_scripts
copying build/scripts-2.7/2to3 -> /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin
copying build/scripts-2.7/pydoc -> /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin
copying build/scripts-2.7/idle -> /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin
copying build/scripts-2.7/smtpd.py -> /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin
changing mode of /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/2to3 to 755
changing mode of /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/pydoc to 755
changing mode of /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/idle to 755
changing mode of /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/smtpd.py to 755
running install_egg_info
Writing /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/Python-2.7.9-py2.7.egg-info
rm /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/_sysconfigdata.py*
Creating directory /usr/man
Creating directory /usr/man/man1
/usr/bin/install -c -m 644 ./Misc/python.man \
		/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/man/man1/python2.7.1
if test -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python -o -h /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python; \
	then rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python; \
	else true; \
	fi
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin; ln -s python2 python)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python2
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin; ln -s python2.7 python2)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python2-config
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin; ln -s python2.7-config python2-config)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin/python-config
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/bin; ln -s python2-config python-config)
test -d /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig || /usr/bin/install -c -d -m 755 /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig/python2.pc
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig; ln -s python-2.7.pc python2.pc)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig/python.pc
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/pkgconfig; ln -s python2.pc python.pc)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/man/man1/python2.1
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/man/man1; ln -s python2.7.1 python2.1)
rm -f /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/man/man1/python.1
(cd /home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/man/man1; ln -s python2.1 python.1)
if test "xno" != "xno"  ; then \
		case no in \
			upgrade) ensurepip="--upgrade" ;; \
			install|*) ensurepip="" ;; \
		esac; \
		 _PYTHON_PROJECT_BASE=/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9 _PYTHON_HOST_PLATFORM=linux2-i486 PYTHONPATH=/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/build/lib.linux2-i486-2.7:./Lib:./Lib/plat-linux2 python2.7 -m ensurepip \
			$ensurepip --root=/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/ ; \
	fi
make[3]: Leaving directory `/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9'
copying: '/usr/lib/python2.7/bsddb'
File not found '/home/dir/openwrt/build_dir/target-i386_uClibc-0.9.30.1/Python-2.7.9/ipkg-install/usr/lib/python2.7/lib-dynload/_bsddb.so'
make[2]: *** [/home/dir/openwrt/bin/x86/packages/python-bsddb_2.7.9-5_x86.ipk] Error 1
make[1]: *** [package/feeds/packages/lang/python/compile] Error 2
make -r package/python/compile: build failed. Please re-run make with V=99 to see what's going on
make: *** [package/python/compile] Error 1

```
Do you have enough information ?

commodo, 2015-03-11:
This information is not sufficient.
Could you look at this : https://enigmabox.net/tmp/python-bsddb-output.log.gz ?

At some point there should something like this:
```
building '_bsddb' extension
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I. -IInclude -I./Include -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-32.mostrecent/staging_dir/host/bin/Include -I/home/x/openwrt-32.mostrecent/staging_dir/host/bin -c /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_bsddb.c -o build/temp.linux2-i486-2.7/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_bsddb.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
ccache_cc -shared -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -fno-strict-aliasing -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable build/temp.linux2-i486-2.7/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_bsddb.o -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/lib -L/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9 -L/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/i486-openwrt-linux-uclibc/lib -L. -R/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/lib -ldb-4.7 -lpython2.7 -o build/lib.linux2-i486-2.7/_bsddb.so
**i486-openwrt-linux-uclibc-gcc: error: unrecognized command line option '-R'**
building '_sqlite3' extension
creating build/temp.linux2-i486-2.7/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_sqlite
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DMODULE_NAME="sqlite3" -DSQLITE_OMIT_LOAD_EXTENSION=1 -IModules/_sqlite -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I. -IInclude -I./Include -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-32.mostrecent/staging_dir/toolchain-i386_i486_gcc-4.8-linaro_uClibc-0.9.33.2/include -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I/home/x/openwrt-32.mostrecent/staging_dir/host/bin/Include -I/home/x/openwrt-32.mostrecent/staging_dir/host/bin -c /home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_sqlite/cache.c -o build/temp.linux2-i486-2.7/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/Modules/_sqlite/cache.o
cc1: note: someone does not honour COPTS correctly, passed 2 times
ccache_cc -fPIC -fno-strict-aliasing -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=i486 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -DMODULE_NAME="sqlite3" -DSQLITE_OMIT_LOAD_EXTENSION=1 -IModules/_sqlite -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/usr/include -I. -IInclude -I./Include -I/home/x/openwrt-32.mostrecent/staging_dir/target-i386_i486_uClibc-0.9.33.2/include -I/home/x/open [... cut ...]
/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/libpython2.7.so: warning: the use of OBSOLESCENT `tmpnam_r' is discouraged, use `mkstemp'
/home/x/openwrt-32.mostrecent/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-2.7.9/libpython2.7.so: warning: the use of OBSOLESCENT `tempnam' is discouraged, use `mkstemp'
```

In the example above, that is the error that occurs.
Could you run a `make package/python/clean`  first and then re-run the build ?
Then run the build with `make package/python/compile V=99 &> output.txt` upload it somewhere and send a link to that file.
The output is pretty big, so please don't post it here.

sixelaa, 2015-03-11:
Ok, I do it. Thank you.

sixelaa, 2015-03-11:
This is my output : http://expirebox.com/download/b18af08aeb9acc8e1e04ee888560493c.html

commodo, 2015-03-11:
Hmm, seems like the bsddb was not built because of the `setup.py` deciding it cannot build it.
For reference, could you tell me the OS you're trying to build this ?
Ubuntu/Debian ?

sixelaa, 2015-03-11:
It's Ubuntu.

I have another question, but it's not for this issue. I succesfully compiled Python 2.7.3 but when I use it, I don't manage to import .so files like _socket.so for example. That's why I try to compile Python 2.7.9. It seems to be a recurrent problem with Python on embedded systems, was it corrected in this package ?

commodo, 2015-03-11:
So,   `import socket` and `import _socket` both work on 2.7.9.

I don't know if they did not work before; I'm guessing some patching could be done to make it work, and you could try to backport stuff/ideas from 2.7.9 to 2.7.3.

You could also try to look at the git history of the `packages` repo; maybe there is a version of 2.7.3 that works in there.

sixelaa, 2015-03-11:
Ok thank you, I'm going to try with an older package. Otherwise, I will try again to install Python 2.7.9.

commodo, 2015-03-11:
Well, I am currently trying to reproduce this on a Ubuntu 14.04 64 bit VM.
The Python `setup.py` script does some weird stuff on Ubuntu & Debian.

commodo, 2015-03-11:
I forgot to ask; are you using the official packages repository or your own version of the packages repo + other stuff where you'd added the official Python package ?
On my Ubuntu VM Python built fine with everything.

Because if so, then you might be missing `libddb47, and you'd need to put that in one of your packages repo.
If you do not want to do that, you can select all python packages + subpackages except `python-bsddb` and it should be fine.

sixelaa, 2015-03-11:
I use my own version of the packages where I add the official Python package. It may be the source of the problem. I don't need the bsddb module so I think I can remove it.
Thank you for the help.

commodo, 2015-03-11:
Cool.
Glad that's sorted.

To conclude:
The idea is that if you need the entire `python` package, you'd need to also be able to support/include the `python-bsddb` package.
Otherwise, you need to use the `python-light` package, and add whatever you need on top ( `python-sqlite3`, `python-bsddb`, `python-ncurses`).
This really helps to reduce the size of Python on OpenWRT.

sixelaa, 2015-03-17:
Hi again, I try to use python-light but I don't really understand how to add packages in`Makefile`.  For exemple I would like to use `python-package-openssl.mk`. What do I have to write in `Makefile` to add it ?

commodo, 2015-03-17:
Hello,

So in your `Makefile`, in the `DEPENDS` section of your package you need to add/replace `+python` with `+python-openssl`.
And example:
```
define Package/<package_name>
  ................................................
  DEPENDS:=<your-other-dependencies> +python-openssl
endef
```
You can also put `+python-light` instead, but you'll have a python package with no SSL support (because SSL is a bit big for some use-cases).
`python` will install everything, and that will take up some space on your target (which you probably need for something else).

sixelaa, 2015-03-17:
Ok, thank you for your explanations.

albans81, 2017-05-24:
Hello, 
I try to compile Python 2.7.13 and I have problem with _curses.o

there the sction of output with error.
`if [ -f /home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/pkginfo/python.default.install.clean ]; then rm -f /home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/pkginfo/python.default.install /home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/pkginfo/python.default.install.clean; fi; echo "python-unittest" >> /home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/pkginfo/python.default.install
mkdir -p /home/ubuntu/Desktop/lima/bin/ar71xx/packages /home/ubuntu/Desktop/lima/build_dir/target-mips_34kc_musl-1.1.14/Python-2.7.11/ipkg-ar71xx/python-ncurses/CONTROL /home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/pkginfo
find /home/ubuntu/Desktop/lima/build_dir/target-mips_34kc_musl-1.1.14/Python-2.7.11/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
copying: '/usr/lib/python2.7/curses'
File not found '/home/ubuntu/Desktop/lima/build_dir/target-mips_34kc_musl-1.1.14/Python-2.7.11/ipkg-install/usr/lib/python2.7/lib-dynload/_curses.so'
Makefile:243: recipe for target '/home/ubuntu/Desktop/lima/bin/ar71xx/packages/packages/python-ncurses_2.7.11-8_ar71xx.ipk' failed
`

how solve this?

commodo, 2017-05-24:
not sure
maybe first try a `make dirclean` ?

can you send/show me:
* your `.config` file
* the link to the repo you are using ? OpenWrt or LEDE ? [ example: https://github.com/lede-project/source/commit/4c3953ba291abfb1d28b1452b6e519367bf4f242 or https://github.com/openwrt/openwrt/commit/c15bd8d6ee8bac11558c6d82bc8260ee8f20568d  or version of LEDE/OpenWrt [ 15.05, 17.01, etc  ]
* the package feeds you are using 

then i can try to reproduce it locally

commodo, 2017-05-24:
also, did you get a message saying ?
```
WARNING: your configuration is out of sync. Please run make menuconfig, oldconfig or defconfig!
``` 

maybe you need to run `make defconfig` ?

albans81, 2017-05-24:
here my config file:
[config.txt](https://github.com/openwrt/packages/files/1026243/config.txt)

REPO: https://www.python.org/ftp/python/ and i use OpenWRT

Here also the Makefile of package python 
[Makefile.txt](https://github.com/openwrt/packages/files/1026256/Makefile.txt)

can i maybe use 2.7.13? how? 

Python 3 not have this problem.

Thx.

commodo, 2017-05-24:
ok ; 
about using 2.7.13 : I am not sure ; maybe just try copying it over your current python [ if that's how you use it] ?

which version of OpenWrt ? 15.05 ? trunk ?

albans81, 2017-05-25:
Good morning...
after make dirclean and now i have error on compiling alsa-utils...sick...
`checking for ALSA CFLAGS... 
checking for ALSA LDFLAGS...  -lasound -lm -ldl -lpthread
checking for alsa/pcm.h... yes
checking for alsa/mixer.h... yes
checking for alsa/rawmidi.h... yes
checking for alsa/seq.h... yes
checking for alsa/use-case.h... yes
checking for alsa/topology.h... yes
checking for samplerate.h... no
checking for librt... checking for clock_gettime in -lrt... yes
checking for ANSI C header files... yes
checking for NCURSESW... no
checking for ncursesw5-config... no
checking for initscr in -lncursesw... no
configure: error: this packages requires a curses library
Makefile:87: recipe for target '/home/ubuntu/Desktop/lima/build_dir/target-mips_34kc_musl-1.1.14/alsa-utils-1.1.0/.configured_yyyyyyyy' failed
make[3]: *** [/home/ubuntu/Desktop/lima/build_dir/target-mips_34kc_musl-1.1.14/alsa-utils-1.1.0/.configured_yyyyyyyy] Error 1
make[3]: Leaving directory '/home/ubuntu/Desktop/lima/feeds/packages/utils/alsa-utils'
package/Makefile:196: recipe for target 'package/feeds/packages/alsa-utils/compile' failed
make[2]: *** [package/feeds/packages/alsa-utils/compile] Error 2
make[2]: Leaving directory '/home/ubuntu/Desktop/lima'
package/Makefile:193: recipe for target '/home/ubuntu/Desktop/lima/staging_dir/target-mips_34kc_musl-1.1.14/stamp/.package_compile' failed
`
ncurses error... the same of python error??? WHY???

i use this trunk of OpenWRT: https://github.com/8devices/openwrt-8devices

to install new vwersion of python probably just change 

./feeds/packages/lang/python/files/python-package.mk:PYTHON_VERSION_MICRO:=11
to 
./feeds/packages/lang/python/files/python-package.mk:PYTHON_VERSION_MICRO:=14
i will try.

commodo, 2017-05-25:
it does not seem that Python 2.7.14 has been released yet
so, maybe just bump PYTHON_VERSION_MICRO:=11  to PYTHON_VERSION_MICRO:=13

but you also need to bump the hash of the tarball in the main Makefile

i haven't had time to look over this yet ; will try this week

albans81, 2017-05-25:
python 14 sorry my bad.. 13 right!!
Thx for your time...i will wait. Thx again.

commodo, 2017-05-26:
it built for me

steps:
* git clone git@github.com:8devices/openwrt-8devices.git
* cd openwrt-8devices
* ./scripts/feeds update -a && ./scripts/feeds install -a
* add your config.txt as .config
* make defconfig - to ensure config sanity ; file size seems to same to me, some stuff got re-ordered ; but it's ok
* make V=99 - some packages failed to download because they are no longer on the official OpenWrt mirror

Packages that needed to be downloaded manually into the dl folder
* audiofile-0.3.6.tar.gz - got it from here http://pkgs.fedoraproject.org/repo/pkgs/audiofile/audiofile-0.3.6.tar.gz/2731d79bec0acef3d30d2fc86b0b72fd/
* pcre-8.38.tar.bz2 - got it from here https://sourceforge.net/projects/pcre/files/pcre/8.38/

My host system is a Ubuntu 16.04 amd64 .

commodo, 2017-05-26:
to add ; the python version is 2.7.11 ; that is packaged with openwrt-8devices [ in the feeds.conf.default ]

so, maybe, tell me your host system ; or please try a clean checkout 
dirty build folders cause a lot of weird issues

albans81, 2017-05-29:
sorry for delay...
Yea, you are right!! i made a full clean and now is work!! Thx for help. Dam dirty folder!!! Thx again!!