openwrt/packages #1002: openvswitch fails to build for arch with 3.18 kernel (buildbot)

plntyk, 2015-03-08:
I can replicate this issue

if i change DEPENDS with
```
diff --git a/net/openvswitch/Makefile b/net/openvswitch/Makefile
index 00951c3..0742413 100644
--- a/net/openvswitch/Makefile
+++ b/net/openvswitch/Makefile
@@ -52,7 +51,7 @@ define Package/openvswitch
   $(call Package/openvswitch/Default)
   TITLE:=Open vSwitch Userspace Package
   DEPENDS:=+libpcap +libopenssl +librt +libatomic +kmod-openvswitch \
-           @(LINUX_3_8||LINUX_3_10||LINUX_3_13||LINUX_3_14)
+		@!(LINUX_3_18||LINUX_3_19||LINUX_4_0)
 endef
 
 define Package/openvswitch/description
```
it seems to be fixed (kmod-openswitch still wants to be built)

reported by:
ar71xx.nand: http://buildbot.openwrt.org:8010/broken_packages/ar71xx.nand/openvswitch/compile.txt
ramips: http://buildbot.openwrt.org:8010/broken_packages/ramips/openvswitch/compile.txt

commodo, 2015-03-08:
So, 
`@(LINUX_3_8||LINUX_3_10||LINUX_3_13||LINUX_3_14)`
and 
`@!(LINUX_3_18||LINUX_3_19||LINUX_4_0)`
kinda mean the same thing, but I prefer the first since it also addresses future kernels (LINUX_4_1 and so on).

OVS supports kernel up to 3.14.
Will wait for newer versions that support newer kernels.

As for `kmod-openswitch` I think the same needs to be added/adapted.
Will come with a PR.

commodo, 2015-03-08:
Created PR : https://github.com/openwrt/packages/pull/1007

plntyk, 2015-03-11:
fixed with #1007

closed 2015-03-11 by plntyk