openwrt/packages #1009: python-base conflicts with python-light (/usr/lib/python2.7/site-packages/README)

enigmagroup, 2015-03-08:
(I had to remove python-base manually, because there were even more conflicts, sigh).

Fortunately, we have implemented pre-update-hooks for cases like that. You know, we have OpenWrt boxes out there in production at 500 clients - and they all use Python ;)

```
python - 2.7.9-4 - 2.7.9-5
root@box:~# opkg install python
Upgrading python on root from 2.7.9-4 to 2.7.9-5...
Downloading http://91.219.238.219/testing/x86/packages/packages/python_2.7.9-5_x86.ipk.
Installing python-bsddb (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-bsddb_2.7.9-5_x86.ipk.
Installing python-codecs (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-codecs_2.7.9-5_x86.ipk.
Installing python-dbm (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-dbm_2.7.9-5_x86.ipk.
Installing python-gdbm (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-gdbm_2.7.9-5_x86.ipk.
Installing python-ncurses (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-ncurses_2.7.9-5_x86.ipk.
Installing python-openssl (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-openssl_2.7.9-5_x86.ipk.
Installing python-pydoc (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-pydoc_2.7.9-5_x86.ipk.
Installing python-sqlite3 (2.7.9-5) to root...
Downloading http://91.219.238.219/testing/x86/packages/packages/python-sqlite3_2.7.9-5_x86.ipk.
libgdbm was autoinstalled and is now orphaned, removing.
Removing package libgdbm from root...
libexpat was autoinstalled and is now orphaned, removing.
Removing package libexpat from root...
Configuring python-codecs.
Configuring python-dbm.
Configuring python-pydoc.
Configuring python-gdbm.
Configuring python-ncurses.
Configuring python-bsddb.
Configuring python-sqlite3.
Configuring python-openssl.
Collected errors:
 * check_data_file_clashes: Package python wants to install file /usr/lib/python2.7/site-packages/README
        But that file is already provided by package  * python-light
```

commodo, 2015-03-09:
Actually, `python` and `python-light` are conflicting.
A fix is here: https://github.com/commodo/packages/commit/a268861ff1e22f86cd68dc0cc2387022fd157d56
Currently I'm building it to see that it works.

So, it seems that I should change my approach when adding code for Python.
Since you have a setup that quickly shows issues within the python packaging, would it be an idea to approach you first, run it through your system and then if all is ok, push the PR ?

enigmagroup, 2015-03-09:
Just came home back from work, sorry.

> Since you have a setup that quickly shows issues within the python packaging, would it be an idea to approach you first, run it through your system and then if all is ok, push the PR ?

That's very kind of you. So far, I'm quite happy with the workflow; I just wanted you to know to keep our setup in mind.

Or is your repo (https://github.com/commodo/packages) always "newer" than the official repo? I can change my repo source then and build it like this. However, I haven't set up an automated nightly build yet. Is that something you'd love to see in the future?

enigmagroup, 2015-03-09:
Still the same error. Oh, I think I haven't ```make clean``` python... But it's really late now, and not sure if I find time this week for testing :/

commodo, 2015-03-10:
No worries.
I also try to find the time to do stuff after work; sometimes I try to squeeze a few in things at work.

My packages repo is just a temp place where I keep my work-in-progress.
`master` is the same as openwrt/packages `master`
I usually keep changes in branches.
So, let's do a PR for this and then you can test without changing repos.
I mean, I've tested the code and this should solve this issue, but I would have hoped you could have tested it a bit and see what other conflicts exist.

I think I should have mentioned before (my bad) that you need to update your `feeds.conf[.default]` file to point the `packages` repo to 
```
src-git packages https://github.com/commodo/packages.git;python-updates
```
Where `python-updates` is the work that I do.

commodo, 2015-03-10:
Created PR: https://github.com/openwrt/packages/pull/1013

enigmagroup, 2015-03-10:
Oh, just "accidentally" updated my repo and now there's an error in gdbm :( So, another problem. I'll wait few more days and conduct a retest.

So far, great ideas, and thanks once again for your efforts!

closed 2015-03-11 by sbyx

enigmagroup, 2015-03-16:
Wonderful! Everything works like a charm! Thank you very much!

commodo, 2015-03-16:
Glad it works :)
FYI: Python is split now into smaller packages.
You may want to take advantage of that, since you can save quite a few megs.
The only thing is that you need to pick which packages you'll need, but I guess for the most cases, just `python-openssl` would be sufficient.

enigmagroup, 2015-03-16:
Thanks for the tip! I'll take a look when I find time again.

OpenWrt is getting more and more awesome every day :)