openwrt/packages #1010: Trouble with PHP5 opcache

MikePetullo, 2015-03-09:
I am trying to add PHP5's opcache to our PHP5 packages. Everything builds, but running php-cgi causes:

# php-cgi 
PHP Warning:  Module 'ctype' already loaded in Unknown on line 0
Sun Mar  8 20:48:31 2015 (26900): Fatal Error Unable to allocate shared memory segment of 8388608 bytes: unknown: Numerical result out of range (34)

Has anyone seen this?

plntyk, 2015-03-09:
- The warning is possibly related to: http://www.somacon.com/p520.php
- second error was observed with openwrt forum and from a korean user where google translate might help: http://minimonk.net/5148

mhei, 2015-05-17:
This is my simple test setup running on a Asus WL-500gP v2. Please verify with your real world setups and report back. Thanks.

`root@OpenWrt:/tmp# cat /tmp/test.php`
```
<?php
  print_r(opcache_get_status());
```
`root@OpenWrt:/tmp# php-cli  /tmp/test.php`
```
Array
(
    [opcache_enabled] => 1
    [cache_full] =>
    [restart_pending] =>
    [restart_in_progress] =>
    [memory_usage] => Array
        (
            [used_memory] => 5251528
            [free_memory] => 61857336
            [wasted_memory] => 0
            [current_wasted_percentage] => 0
        )

    [interned_strings_usage] => Array
        (
            [buffer_size] => 4194304
            [used_memory] => 92376
            [free_memory] => 4101928
            [number_of_strings] => 1740
        )

    [opcache_statistics] => Array
        (
            [num_cached_scripts] => 1
            [num_cached_keys] => 1
            [max_cached_keys] => 223
            [hits] => 0
            [start_time] => 1431729886
            [last_restart_time] => 0
            [oom_restarts] => 0
            [hash_restarts] => 0
            [manual_restarts] => 0
            [misses] => 1
            [blacklist_misses] => 0
            [blacklist_miss_ratio] => 0
            [opcache_hit_rate] => 0
        )

    [scripts] => Array
        (
            [/tmp/test.php] => Array
                (
                    [full_path] => /tmp/test.php
                    [hits] => 0
                    [memory_consumption] => 520
                    [last_used] => Fri May 15 22:44:46 2015
                    [last_used_timestamp] => 1431729886
                    [timestamp] => 1431728600
                )

        )

)
```

commit cd0bcd23 (2015-05-17): php5: add php5-mod-opcache (fixes #1010)

This patch adds build infrastructure for PHP's OPcache extension.
Compared with the other extension, this is a Zend module and it
need a little workaround during cross-compiling.
commit f0a04488 (2015-12-28): php5: add php5-mod-opcache (fixes #1010)

This patch adds build infrastructure for PHP's OPcache extension.
Compared with the other extension, this is a Zend module and it
need a little workaround during cross-compiling.
closed 2015-05-17 by mhei