openwrt/packages #1014: LVM2.2.02.117 Error 2

Love4Taylor, 2015-03-10:
/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/usr/lib/libblkid.so: warning: This version of __fpending returns bytes remaining in buffer for both narrow and wide streams.  glibc's version returns wide chars in buffer for the wide stream case.
../lib/liblvm-internal.a(report.o): In function `report_object':
report.c:(.text+0x3514): undefined reference to `dm_report_object_is_selected'
collect2: error: ld returned 1 exit status
make[5]: *** [lvm] Error 1
make[5]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.117/tools'
make[4]: *** [tools] Error 2
make[4]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.117'
make[3]: *** [/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/LVM2.2.02.117/.built] Error 2
make[3]: Leaving directory `/home/love4taylor/openwrt/trunk/feeds/packages/utils/lvm2'
make[2]: *** [package/feeds/packages/lvm2/compile] Error 2
make[2]: Leaving directory `/home/love4taylor/openwrt/trunk'
make[1]: *** [/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/love4taylor/openwrt/trunk'

dangowrt, 2015-03-10:
you'll have to
make clean
and not just update repos and build again.
otherwise the old still present version of device mapper breaks the build.

champtar, 2015-03-11:
@Love4Taylor, is this still an issue ?

closed 2015-03-15 by dangowrt