openwrt/packages #1019: aria2 build error

Love4Taylor, 2015-03-11:
LibuvEventPoll.cc: In member function 'virtual void aria2::LibuvEventPoll::poll(const timeval&)':
LibuvEventPoll.cc:144:59: error: invalid conversion from 'void (*)(uv_timer_t*, int) {aka void (*)(uv_timer_s*, int)}' to 'uv_timer_cb {aka void (*)(uv_timer_s*)}' [-fpermissive]
     uv_timer_start(timer, timer_callback, timeout, timeout);
                                                           ^
In file included from LibuvEventPoll.h:43:0,
                 from LibuvEventPoll.cc:44:
/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/usr/include/uv.h:760:15: error:   initializing argument 2 of 'int uv_timer_start(uv_timer_t*, uv_timer_cb, uint64_t, uint64_t)' [-fpermissive]
 UV_EXTERN int uv_timer_start(uv_timer_t* handle,
               ^
make[7]: *** [LibuvEventPoll.lo] Error 1
make[7]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/aria2-1.18.7/src'
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/aria2-1.18.7/src'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/aria2-1.18.7'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/aria2-1.18.7'
make[3]: *** [/home/love4taylor/openwrt/trunk/build_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/aria2-1.18.7/.built] Error 2
make[3]: Leaving directory `/home/love4taylor/openwrt/trunk/feeds/packages/net/aria2'
make[2]: *** [package/feeds/packages/aria2/compile] Error 2
make[2]: Leaving directory `/home/love4taylor/openwrt/trunk'
make[1]: *** [/home/love4taylor/openwrt/trunk/staging_dir/target-arm_mpcore_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/love4taylor/openwrt/trunk'

plntyk, 2015-03-11:
https://github.com/openwrt/packages/pull/1011 should fix this

dupidog, 2015-07-23:
Try following patch which I've sent to maintainer. 

```
From: Zhangben Wu <dupidog@gmail.com>
Date: Thu, 23 Jul 2015 09:20:49 +0800 
Subject: [PATCH] [package] aria2: add dependence libuv, fix compile error in CC and trunk

---

--- old/net/aria2/Makefile 
+++ new/net/aria2/Makefile 
@@ -38,7 +38,7 @@ define Package/aria2 
SUBMENU:=File Transfer 
TITLE:=lightweight download utility 
URL:=http://aria2.sourceforge.net/ 
- DEPENDS:=+zlib +ARIA2_METALINK:libxml2 +libstdcpp +ARIA2_OPENSSL:libopenssl +ARIA2_GNUTLS:libgnutls 
+ DEPENDS:=+zlib +ARIA2_METALINK:libxml2 +libstdcpp +ARIA2_OPENSSL:libopenssl +ARIA2_GNUTLS:libgnutls +libuv 
endef 

define Package/aria2/description 

--- old/src/LibuvEventPoll.cc 2015-07-23 09:20:49.139641711 +0800 
+++ new/src/LibuvEventPoll.cc	2015-07-23 09:13:05.127774373 +0800 
@@ -141,7 +141,7 @@ 
if (timeout >= 0) { 
auto timer = new uv_timer_t; 
uv_timer_init(loop_, timer); 
- uv_timer_start(timer, timer_callback, timeout, timeout); 
+ uv_timer_start(timer, (uv_timer_cb)timer_callback, timeout, timeout); 

uv_run(loop_, UV_RUN_DEFAULT); 
```

yousong, 2016-08-13:
The issue should already be resolved with the version bump.

closed 2016-08-13 by yousong