openwrt/packages #1033: Video streaming doesn't work for Nexx 3020h

iholub, 2015-03-12:
Used image:
https://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/
openwrt-ramips-mt7620-wt3020-8M-squashfs-sysupgrade.bin

Installed packages:
opkg install kmod-video-uvc
opkg install mjpg-streamer
opkg install usbutils
opkg install v4l-utils

Following commands are working:
v4l2-ctl --list-formats-ext
v4l2-ctl -V
lsusb

Mjpg-streamer configuration:

config mjpg-streamer 'core'
option enabled '1'
option input 'uvc'
option output 'http'
option device '/dev/video0'
option resolution '640x480'
option fps '5'
option led 'auto'
option www '/www/webcam'
option port '8080'

Tested in Firefox browser (.../8080/?action=stream), is says that image can not be displayed because it contains errors.

No errors in dmesg command output.

Also tried to install "motion" package, video stream also doesn't work.

Tested with web cameras:
Logitech C270 and Logitech C210

thess, 2015-03-13:
A couple of quick questions (more info needed)
* Does /dev/video0 exist?
* Examine the System Log (via LuCI or logread). It should show messages from mjpg_streamer startup and connections.

iholub, 2015-03-13:
yes, it exists
root@OpenWrt:~# ls -la /dev/video0
crw-r--r--    1 root     root       81,   0 Mar 13 19:53 /dev/video0

and here is the log

Fri Mar 13 19:53:00 2015 kern.info kernel: [   64.420000] usb 2-1: new full-speed USB device number 2 using ohci-platform
Fri Mar 13 19:53:00 2015 kern.info kernel: [   64.660000] usb 2-1: not running at top speed; connect to a high speed hub
Fri Mar 13 19:53:00 2015 kern.debug kernel: [   64.680000] usb 2-1: no of_node; not parsing pinctrl DT
Fri Mar 13 19:53:00 2015 kern.debug kernel: [   64.680000] hub 2-1:1.0: no of_node; not parsing pinctrl DT
Fri Mar 13 19:53:00 2015 kern.info kernel: [   64.680000] hub 2-1:1.0: USB hub found
Fri Mar 13 19:53:00 2015 kern.info kernel: [   64.680000] hub 2-1:1.0: 4 ports detected
Fri Mar 13 19:53:00 2015 kern.info kernel: [   65.000000] usb 2-1.3: new full-speed USB device number 3 using ohci-platform
Fri Mar 13 19:53:01 2015 kern.info kernel: [   65.430000] usb 2-1.3: not running at top speed; connect to a high speed hub
Fri Mar 13 19:53:01 2015 kern.debug kernel: [   65.530000] usb 2-1.3: no of_node; not parsing pinctrl DT
Fri Mar 13 19:53:01 2015 kern.debug kernel: [   65.530000] uvcvideo 2-1.3:1.0: no of_node; not parsing pinctrl DT
Fri Mar 13 19:53:01 2015 kern.info kernel: [   65.540000] uvcvideo: Found UVC 1.00 device <unnamed> (046d:0819)
Fri Mar 13 19:53:01 2015 kern.info kernel: [   65.660000] input: UVC Camera (046d:0819) as /devices/101c1000.ohci/usb2/2-1/2-1.3/2-1.3:1.0/i     nput/input0
Fri Mar 13 19:53:01 2015 user.info MJPG-streamer [1288]: starting application
Fri Mar 13 19:53:01 2015 user.info MJPG-streamer [1288]: MJPG Streamer Version: svn rev: exported
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: Using V4L2 device.: /dev/video0
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: Desired Resolution: 640 x 480
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: Frames Per Second.: 5
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: Format............: JPEG
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: TV-Norm...........: DEFAULT
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: www-folder-path...: /www/webcam/
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: HTTP TCP port.....: 8080
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: username:password.: disabled
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: commands..........: enabled
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: starting input plugin input_uvc.so
Fri Mar 13 19:53:02 2015 user.info MJPG-streamer [1288]: starting output plugin: output_http.so (ID: 00)
Fri Mar 13 19:53:08 2015 kern.notice kernel: [   73.300000] random: nonblocking pool is initialized
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:18 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:19 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:21 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:21 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:21 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:21 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:21 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:22 2015 user.info MJPG-streamer [1288]: serving client: 192.168.2.33
Fri Mar 13 19:54:30 2015 authpriv.info dropbear[1554]: Child connection from 192.168.2.33:52099
Fri Mar 13 19:54:33 2015 authpriv.notice dropbear[1554]: Password auth succeeded for 'root' from 192.168.2.33:52099
Fri Mar 13 19:55:28 2015 authpriv.info dropbear[1719]: Child connection from 192.168.2.33:52101
Fri Mar 13 19:55:34 2015 authpriv.notice dropbear[1719]: Password auth succeeded for 'root' from 192.168.2.33:52101

iholub, 2015-03-13:
The problem is solved!
I installed today's March 13 firmware (I used from March 11 before)
https://downloads.openwrt.org/snapshots/trunk/ramips/mt7620/
And camera works now!

closed 2015-03-13 by iholub