openwrt/packages #1040: python: the package gets "installed" every time "scripts/feeds install -a" is run

hnyman, 2015-03-15:
There seems to be something strange with the python package since the update 3-5 days ago: the package gets "installed" every time "scripts/feeds install -a" is run. Even in case where "scripts/feeds install -a" is run several times consecutively, the python package is found again and reinstalled.

```
perus@ub1410:/Openwrt/trunk$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing package 'python'
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed management.
perus@ub1410:/Openwrt/trunk$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing package 'python'
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed management.
perus@ub1410:/Openwrt/trunk$
```
There is probably some duplication in definitions with the new python/Default and plain python. Make menuconfig shows only the description of "python", while there is no trace of python-default and its description.

You should probably rethink the naming and mutual inclusion of the packages. Should "python" be "python-full" or something like that? Is there need for separate python-default and python-base (and python-light)?

@commodo

champtar, 2015-03-19:
hi @hnyman,
I'm not reproducing it
can you try:
```
./scripts/feeds update -a
./scripts/feeds uninstall -a
./scripts/feeds install -a
./scripts/feeds install -a
```

hnyman, 2015-03-19:
uninstall step did not have any effect. I still see "installing python" every time.

hnyman, 2015-03-19:
I downloaded a fresh copy of the Openwrt buildroot, and the problem does not materialise there. 
So, it is probably something related to my current buildroot. 
I will close the issue.

closed 2015-03-19 by hnyman

commodo, 2015-03-19:
Sorry I didn't see this issue earlier.
So, with the feeds script I saw some weird stuff when splitting python into packages (I don't recall if I saw something like this), but I always did a fresh checkout and that solved it.
I think the indexing of the feeds needs a bit of work, because when I add a new package to the feeds, the feeds script doesn't always pick it up properly in the menuconfig.