openwrt/packages #1044: named fails to start on boot due to no interfaces were up to this time

toad42, 2015-03-16:
named fails to start on boot because no interfaces were up to this time.
It needs a rewrite of the start script to recognize interface changes, maybe I try it.

I workaround this for now with a sleep 10 in the start script and adding "interface-interval 1;" in the "named.conf".

nmeyerhans, 2015-04-27:
I could be mistaken, but I think hotplug scripts are the appropriate fix here. Something like the following in /etc/hotplug.d/iface seems to work:

```
NAME=named

[ "$ACTION" = "ifup" -a "$INTERFACE" = "lan" ] && {
    logger -p user.notice "Restarting $NAME due to $ACTION $INTERFACE"
    /etc/init.d/$NAME restart
} &
```

On the other hand, I don't see why this should be necessary. At the very least `lo` should be up before named tries to start, due to boot script ordering.

diizzyy, 2017-06-07:
@nmeyerhans 
Still relevant or can we close this?

nmeyerhans, 2017-06-07:
Yeah, let's close. A lot of time has passed and there haven't been any additional reports or followups to this issue, and as I stated previously, I believe that the order of the init script execution should ensure that at least `lo` is up before named.

closed 2017-06-07 by nmeyerhans