openwrt/packages #1051: if use github tag for PKG_SOURCE_URL, filename conflict, all project will be "x.y.z.tar.gz"

sinkcup, 2015-03-19:
look at https://github.com/wifidog/wifidog-gateway/releases
github tag download link is https://github.com/wifidog/wifidog-gateway/archive/1.2.0.tar.gz

so openwrt package code `Makefile` should like this:

    PKG_NAME:=wifidog
    PKG_VERSION:=1.2.0
    PKG_RELEASE:=1

    PKG_MD5SUM:=c1c5de15279d0224b6ea1ed72d416a93
    PKG_SOURCE:=$(PKG_VERSION).tar.gz
    PKG_SOURCE_URL:=https://github.com/wifidog/wifidog-gateway/archive/

when make this package, file will be downloaded as "dl/1.2.0.tar.gz", all project use the same filename, will conflict.
I have read the code of `openwrt/scripts/download.pl`, it use PKG_SOURCE as filename, and we can't change it.

    system("mv", "$target/$filename.dl", "$target/$filename");

suggestion: auto rename downloaded file as $(PKG_NAME)-$(PKG_VERSION), or allow us to rename file, e.g. : 

    PKG_SOURCE_OUTPUT=$(PKG_NAME)-$(PKG_VERSION).tar.gz

Now we have no alternative but to download a tag, rename, then upload to github release. like:

https://github.com/shadowsocks/openwrt-shadowsocks/releases

[image]

mhei, 2015-04-29:
Please do not confuse the tag archives (tarballs) provided by Github and the release archive tarballs from the individual projects. The first ones are a simple export/copy of the sources, the later are created by autotools stuff (run "make dist" within the source directory) and include some more "magic" for the well-known "configure", "make", "make install" trilogy.
Well, actually, the difference is only a autoreconf ... call, but this is another topic.

If you want to refer to a dedicated version, then use the (autotools created) release tarballs and the url for these ones. If you want to refer to a not-yet-released tag, then use PKG_SOURCE_PROTO:=git and adopt the Makefiles variables accordingly (see package knxd for good example). OpenWrt's makefile will handle the rest and there will be not conflict with other packages. Then you have to add a "PKG_FIXUP:=autoreconf" line into the makefile.

In other words, I think the current implementation is enough to support various scenarios and I see no need to change it.  However, just my 2cts :smile:

psycho-nico, 2015-05-01:
luaposix is afftected too, but I can't use the same fix. Fetching the source using git aborts with a cryptic error message : "No submodule mapping found in .gitmodules"

sinkcup, 2015-05-01:
Thanks for reviewing so many packages, @psycho-nico

champtar, 2015-07-27:
think we can close this for now

closed 2015-07-27 by champtar