openwrt/packages #1052: can't compile libsigc++

richlv, 2015-03-19:
in current trunk, libsigc++ compilation fails :
```
autoreconf: running: /files/openwrt/staging_dir/host/bin/autoconf --include=/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal --include=/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal --include=m4 --include=. --prepend-include=/files/openwrt/staging_dir/host/share/aclocal --force
autoreconf: running: /files/openwrt/staging_dir/host/bin/autoheader --include=/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/host/share/aclocal --include=/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/share/aclocal --include=m4 --include=. --prepend-include=/files/openwrt/staging_dir/host/share/aclocal --force
autoreconf: running: /files/openwrt/staging_dir/host/bin/automake --add-missing --force-missing
Makefile.am:20: error: ENABLE_DOCUMENTATION does not appear in AM_CONDITIONAL
build/doc-reference.am:70: error: ENABLE_DOCUMENTATION does not appear in AM_CONDITIONAL
docs/Makefile.am:32:   'build/doc-reference.am' included from here
build/doc-reference.am:80: error: DIST_DOCTOOLS does not appear in AM_CONDITIONAL
docs/Makefile.am:32:   'build/doc-reference.am' included from here
autoreconf: /files/openwrt/staging_dir/host/bin/automake failed with exit status: 1
(cd /files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/./; if [ -x ./configure ]; then find /files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/ -name config.guess | xargs -r chmod u+w; find /files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/ -name config.guess | xargs -r -n1 cp --remove-destination /files/openwrt/scripts/config.guess; find /files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/ -name config.sub | xargs -r chmod u+w; find /files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/ -name config.sub | xargs -r -n1 cp --remove-destination /files/openwrt/scripts/config.sub; AR="mips-openwrt-linux-uclibc-gcc-ar" AS="mips-openwrt-linux-uclibc-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -fpic" LD=mips-openwrt-linux-uclibc-ld NM="mips-openwrt-linux-uclibc-gcc-nm" CC="mips-openwrt-linux-uclibc-gcc" GCC="mips-openwrt-linux-uclibc-gcc" CXX="mips-openwrt-linux-uclibc-g++" RANLIB="mips-openwrt-linux-uclibc-gcc-ranlib" STRIP=mips-openwrt-linux-uclibc-strip OBJCOPY=mips-openwrt-linux-uclibc-objcopy OBJDUMP=mips-openwrt-linux-uclibc-objdump SIZE=mips-openwrt-linux-uclibc-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -fpic " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -msoft-float -mips16 -minterlink-mips16 -fpic " CPPFLAGS="-I/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/include -I/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/include -I/files/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/include -I/files/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include -fno-strict-aliasing -fno-inline  " LDFLAGS="-L/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/usr/lib -L/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/lib -L/files/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/usr/lib -L/files/openwrt/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/lib "   ./configure --target=mips-openwrt-linux --host=mips-openwrt-linux --build=x86_64-suse-linux --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   --enable-shared --enable-static  ; fi; )
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /files/openwrt/include/site/mips-openwrt-linux-uclibc
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for mips-openwrt-linux-strip... mips-openwrt-linux-uclibc-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '100' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
./configure: line 3054: syntax error near unexpected token `0.9.7'
./configure: line 3054: `MM_PREREQ(0.9.7)'
Makefile:63: recipe for target '/files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/.configured_' failed
make[3]: *** [/files/openwrt/build_dir/target-mips_34kc_uClibc-0.9.33.2/libsigc++-2.4.1/.configured_] Error 2
make[3]: Leaving directory '/files/openwrt/feeds/packages/libs/libsigc++'
package/Makefile:174: recipe for target 'package/feeds/packages/libsigc++/compile' failed
make[2]: *** [package/feeds/packages/libsigc++/compile] Error 2
make[2]: Leaving directory '/files/openwrt'
package/Makefile:171: recipe for target '/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/files/openwrt/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/files/openwrt'
/files/openwrt/include/toplevel.mk:179: recipe for target 'world' failed
make: *** [world] Error 2
```

moonman, 2015-03-23:
Please close. Has been fixed.https://github.com/openwrt-mirror/openwrt/commit/a70ec69ae8fae0258167693212d5e5fae5e0a67d

closed 2015-03-23 by champtar

richlv, 2015-03-23:
yep, seems to compile ok in latest trunk :)

moonman, 2015-03-23:
Btw, this library is openwrt trunk, not in packages for openwrt

richlv, 2015-03-23:
oh, sorry about that - it got pulled in by installing rtorrent package, so i assumed it's all packages :)