openwrt/packages #1054: collectd-mod-openvpn integration in luci_statistics

Thomas-1985, 2015-03-19:
The script /usr/bin/stat-genconfig which is executing by /etc/init.d/luci_statistics does not recognize the following configuration in /etc/config/luci_statistics

```
config statistics 'collectd_openvpn'
	option enable '1'
	option StatusFile '/var/log/openvpn-status.log'
	option CollectIndividualUsers 'false'
	option CollectUserCount 'false'
	option CollectCompression 'true'
	option ImprovedNamingSchema 'true'
```

If this configuration is manually specified in /etc/collectd.conf:
```
LoadPlugin

<Plugin openvpn>
	StatusFile "/var/log/openvpn-status.log"
	CollectIndividualUsers false
	CollectUserCount false
	CollectCompression true
	ImprovedNamingSchema true
</Plugin>
```

 it is being created the RRD database /tmp/rrd/<hostname>/openvpn-openvpn-status.log/ :
```
-rw-r--r--    1 root     root         10384 Sep 24 23:25 compression-data_in.rrd
-rw-r--r--    1 root     root             0 Sep 24 23:05 compression-data_out.rrd
-rw-r--r--    1 root     root         10384 Sep 24 23:25 if_octets-overhead.rrd
-rw-r--r--    1 root     root         10384 Sep 24 23:25 if_octets-traffic.rrd
```

 But after rebooting the router LuCI script deleted configuration anyway.
There are no access to images as they are not created also.
Thanks.

Thomas-1985, 2015-06-25:
I tried again with version 4.10.8-3 of collectd-mod-openvpn and the issue is still the same.

I worked around LuCI by writing the openvpn-plugin snipped to /etc/collectd/conf.d/openvpn and now the rrd is created.
However, i am not seeing any graph via LuCI.

hnyman, 2016-12-31:
Likely fixed by https://github.com/openwrt/luci/commit/3289e134953887482bdaa28c28567917bb91facb that added openvpn support for Luci statistics.

closed 2016-12-31 by hnyman