openwrt/packages #1064: Add missing python packages

alirz1, 2015-03-23:
Hello,
When trying to install a python app "Flexget" I came to realize that some packages are missing. Please include the following missing packages

python python-sqlite3             OK
pyyaml                                    missing
python-sqlite                           missing
python-expat                          missing
python-openssl                      OK
python-bzip2                          missing
distribute                                missing

commodo, 2015-03-23:
Well, nobody is maintaining some of them at the moment.

- pyaml - needs to be added and a maintainer found
- python-sqlite - I'm a bit puzzled as to why this exists/is needed, since Python has sqlite3 built-in; not sure what difference between them is
- python-expat - should be in there, just not in a separate package
- python-bzip2 - same as python-expat
- distribute - needs to be added and a maintainer found

The others that are marked OK are actually part of the standard python installation, and have been packaged separately to offer the possibility of reducing the installation of Python on a target.

These packages can still be found in the oldpackages repo if they're still needed, but they're not maintained.

alirz1, 2015-03-23:
Thanks for the response. 
Is this the old package repo? http://downloads.openwrt.org/snapshots/trunk/mvebu/generic/packages/oldpackages

I'm not a developer and have no clue on developing/ maintaining packages.

commodo, 2015-03-23:
Ah, I should have said before where the old packages repo is.
If you take a look in the `feeds.conf.default` file in a OpenWRT fresh checkout, you should find it.
There are multiple official feeds in there.

The old packages [and other feeds] is commented out by default.
```
#src-git oldpackages http://git.openwrt.org/packages.git
```
If you un-comment that line it should appear.

alirz1, 2015-03-23:
I think this involves rebuilding openwrt from source? 
Seems a bit out of my technical scope :)

commodo, 2015-03-23:
So, I don't know of a repo where the old packages are kept.
The link you posted initially doesn't seem to have them:
 https://downloads.openwrt.org/snapshots/trunk/mvebu/generic/packages/

I did not fully read your initial message about not having a clue about maintaining packages.
I just browsed through their page here: http://flexget.com/wiki/InstallWizard/OpenWrt
 
Looks like those instructions are a bit old, since before packages were moved to Github.
What about installing `python-pip` first and running `pip install flexget` ?
`pip`is supported in the new packages.

alirz1, 2015-03-23:
Alright now that you've asked about it i'll go in details :)
So yes, i am able to install flexget using PIP. However that install takes about quite a bit of space for some reason. While it does install, it fills about 95% of my router's / parition.

Previously i was using "easy_install" on MCWRT to flexget on my mounted USB flash drive and that worked fine.
So with CC openwrt, i tried to use PIP to install to my USB as: 

pip install flexget -t /mnt/sda/flexget

While that seems to install everything.. The actual binary for flexget that you would launch to start the app, doesnt seem to get "built or created" anywhere on the file system.
However when i install flexget using pip without specifying the destination path, it installs in the router / and the binary shows up properly under /usr/lib/flexget  if i remember correctly

commodo, 2015-03-23:
Well, pip is pretty fat, so you need a router with a fair amount of space on a system to use it (I'd say 32 MB flash at a minimum, but 64+ MB is where it starts to get comfortable).

I'd say that if you have around 16 MB, you'd need to start with `python-light` or `python-openssl` and manually add files as they're needed.

Seems that the amount of packages required by FlexGet is pretty big.
I mean
```
install_requires = ['FeedParser>=5.1.3', 'SQLAlchemy >=0.7.5, !=0.9.0, <0.9.99', 'PyYAML',
                    'beautifulsoup4>=4.1, !=4.2.0, <4.4', 'html5lib>=0.11', 'PyRSS2Gen', 'pynzb', 'progressbar', 'rpyc',
                    'jinja2', 'requests>=1.0, !=2.4.0, <2.99', 'python-dateutil!=2.0, !=2.2', 'jsonschema>=2.0',
                    'python-tvrage', 'tmdb3', 'path.py', 'guessit>=0.9.3', 'apscheduler']
```
Those seem pretty big libs, and I'm not sure if they would fit on a 16 MB, and even on a 32 MB flash it looks difficult to squeeze them.

Regarding the `pip install flexget -t /mnt/sda/flexget` not installing everything, it could most likely be a bug in the FlexGet install-script.
I won't deny the fact that it could also be a bug in the OpenWRT `python-pip` package.

If you want, we could take this in a separate discussion via email (so as not to spam people).
My email is on my Github profile.

alirz1, 2015-03-23:
I have a WRT1900AC router. It has 128MB flash but most of it is allocated for / tmp in openwrt.
/ only gets about 30MB for installable apps. While that is enough for PIP to install in the root. I dont want to run flexget withing the NAND memory of the router. Unnecessary writes can wear out the memory.

commodo, 2015-04-24:
@alirz1 or @sbyx or @thess 
Is this still an issue ?
If not, please close.

alirz1, 2015-04-24:
Was able to use PIP eventually. Closing ticket

closed 2015-04-24 by alirz1