openwrt/packages #1075: PolarSSL support in Transmission (Bittorrent)

diizzyy, 2015-03-26:
Transmission doesn't yet support PolarSSL in a release however it is supported in transmission trunk.

Transmission r14481 does however compile (daemon) with the following changes in the Makefile:

Line 35: Remove ```+libopenssl``` and replace with ```+libpolarssl +zlib```
Line 86: Move everything down one line and add ```--with-crypto=polarssl \ ```

010_libtransmission_fallocate64_eglibc.patch also fails but I don't think it's needed anymore? It compile fine without it at least.

Unfortunately I wasn't able to figure out how to adapt the URL and extraction but this probably helps a few ppl.

Tested in trunk and package checkout about 12h old, using musl and libstdc++.

Perhaps add this as an additional port?

obsy, 2015-03-26:
Just wait for stable edition. If the new release appear, new option will added.

diizzyy, 2016-04-05:
Current version supports this, unfortunately I haven't manage to make a working Makefile even using OpenVPN as a template. Can you please have a look at it?

obsy, 2016-04-06:
See #2578

diizzyy, 2016-04-07:
Cool, thanks!

closed 2016-04-07 by diizzyy