openwrt/packages #1077: python3 compile error

taeasy, 2015-03-27:
make[1]: Entering directory `/home/main/x86'
make[2]: Entering directory `/home/main/x86/package/libs/toolchain'
if [ -f /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install
if [ -f /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean ]; then rm -f /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/pkginfo/toolchain.default.install
make[2]: Leaving directory `/home/main/x86/package/libs/toolchain'
make[2]: Entering directory `/home/main/x86/feeds/packages/utils/bzip2'
make[2]: Leaving directory `/home/main/x86/feeds/packages/utils/bzip2'
make[2]: Entering directory `/home/main/x86/package/libs/zlib'
make[2]: Leaving directory `/home/main/x86/package/libs/zlib'
make[2]: Entering directory `/home/main/x86/feeds/packages/libs/libxml2'
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/libxml2'
make[2]: Entering directory `/home/main/x86/package/libs/uclibc++'
make[2]: Leaving directory `/home/main/x86/package/libs/uclibc++'
make[2]: Entering directory `/home/main/x86/feeds/packages/libs/db47'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47
install -d -m0755 /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47/usr/lib
cp -fpR /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/db-4.7.25.NC/ipkg-install/usr/lib/libdb-*.so /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47/usr/lib/
SHELL= /home/main/x86/staging_dir/host/bin/flock /home/main/x86/tmp/.root-copy.flock -c 'cp -fpR /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47/. /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47
touch /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp/.libdb47_installed
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx
install -d -m0755 /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx/usr/lib
cp -fpR /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/db-4.7.25.NC/ipkg-install/usr/lib/libdb_cxx-*.so /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx/usr/lib/
SHELL= /home/main/x86/staging_dir/host/bin/flock /home/main/x86/tmp/.root-copy.flock -c 'cp -fpR /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx/. /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libdb47xx
touch /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp/.libdb47xx_installed
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/db47'
make[2]: Entering directory `/home/main/x86/feeds/packages/libs/gdbm'
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/gdbm'
make[2]: Entering directory `/home/main/x86/feeds/packages/libs/libffi'
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/libffi'
make[2]: Entering directory `/home/main/x86/feeds/packages/lang/python3'
make[2]: Leaving directory `/home/main/x86/feeds/packages/lang/python3'
make[2]: Entering directory `/home/main/x86/package/libs/libreadline'
make[2]: Leaving directory `/home/main/x86/package/libs/libreadline'
make[2]: Entering directory `/home/main/x86/package/libs/ncurses'
make[2]: Leaving directory `/home/main/x86/package/libs/ncurses'
make[2]: Entering directory `/home/main/x86/package/libs/ncurses'
make[2]: Leaving directory `/home/main/x86/package/libs/ncurses'
make[2]: Entering directory `/home/main/x86/package/libs/ncurses'
make[2]: Leaving directory `/home/main/x86/package/libs/ncurses'
make[2]: Entering directory `/home/main/x86/package/libs/ncurses'
make[2]: Leaving directory `/home/main/x86/package/libs/ncurses'
make[2]: Entering directory `/home/main/x86/feeds/packages/libs/sqlite3'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3
install -d -m0755 /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3/usr/lib
cp -fpR /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/sqlite-autoconf-3080803/ipkg-install/usr/lib/libsqlite3.so.* /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3/usr/lib/
SHELL= /home/main/x86/staging_dir/host/bin/flock /home/main/x86/tmp/.root-copy.flock -c 'cp -fpR /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3/. /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-libsqlite3
touch /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp/.libsqlite3_installed
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli
install -d -m0755 /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli/usr/bin
install -m0755 /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/sqlite-autoconf-3080803/ipkg-install/usr/bin/sqlite3 /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli/usr/bin/
SHELL= /home/main/x86/staging_dir/host/bin/flock /home/main/x86/tmp/.root-copy.flock -c 'cp -fpR /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli/. /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-sqlite3-cli
touch /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp/.sqlite3-cli_installed
make[2]: Leaving directory `/home/main/x86/feeds/packages/libs/sqlite3'
make[2]: Entering directory `/home/main/x86/package/libs/ocf-crypto-headers'
make[2]: Leaving directory `/home/main/x86/package/libs/ocf-crypto-headers'
make[2]: Entering directory `/home/main/x86/package/libs/openssl'
make[2]: Leaving directory `/home/main/x86/package/libs/openssl'
make[2]: Entering directory `/home/main/x86/feeds/packages/lang/python3'
rm -rf /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-python3-asyncio
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-python3-asyncio
find /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
copying: '/usr/lib/python3.4/asyncio'
[... 17 lines cut ...]
mkdir -p /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/stamp /home/main/x86/staging_dir/target-i386_i486_uClibc-0.9.33.2/root-x86/tmp-python3-ctypes
find /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
copying: '/usr/lib/python3.4/ctypes'
copying: '/usr/lib/python3.4/lib-dynload/_ctypes.cpython-34.so'
copying: '/usr/lib/python3.4/lib-dynload/_ctypes_test.cpython-34.so'
[... cut ...]
copying: '/usr/lib/python3.4/decimal.py'
File not found '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/ipkg-install/usr/lib/python3.4/lib-dynload/_decimal.cpython-34.so'
make[2]: *** [/home/main/x86/bin/x86/packages/packages/python3-decimal_3.4.3-2_x86.ipk] Error 1
make[2]: Leaving directory `/home/main/x86/feeds/packages/lang/python3'
make[1]: *** [package/feeds/packages/python3/compile] Error 2
make[1]: Leaving directory `/home/main/x86'
make: *** [package/python3/compile] Error 2

commodo, 2015-03-27:
Well, not sure if it's a compilation error yet.
Seems like this file was not built:
```
File not found '/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/ipkg-install/usr/lib/python3.4/lib-dynload/_decimal.cpython-34.so'
```

Could you tell me details about your build/host system ?
Debian/Ubuntu/etc, x86/x86_64 ? 
Which version ?

The general problem in this cases seems to be with the `setup.py` script which is too "smart" and that can sometimes fail because it picks up stuff from the host and not the build environment.

Also, could you delete this folder `/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/` and rebuild ?

And in the build log, there should be 
```
Building 'decimal' extension
```
and the root cause of the build failure should be somewhere close to that.

And you can capture the build log to file (for simplicity) with
```
make &> output.txt
```

taeasy, 2015-03-27:
host system is Debian 7.8 x86_64 , i have deleted  folder /home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/ and rebuild ,but the errer still.

taeasy, 2015-03-27:
output.txt
http://104.224.133.76/output.txt

commodo, 2015-03-27:
Seems the compilation error is 
```
/home/main/x86/build_dir/target-i386_i486_uClibc-0.9.33.2/Python-3.4.3/Modules/_decimal/libmpdec/mpdecimal.c:49:22: fatal error: fenv.h: No such file or directory
     #include <fenv.h>
                      ^
```

Will spin up a VM and try that and come up with a fix.
Most likely this build failure is not new, but has emerged recently because I've split Python3 into  packages.

Thanks
Alex

commodo, 2015-03-27:
Seems this is a bug only on x86 targets.
After PR https://github.com/openwrt/packages/pull/1078 it should work.
Tested and worked on Debian 7/64 and OpenSUSE 13.2/64.

taeasy, 2015-03-27:
it's working now,thank you very much.

closed 2015-03-27 by taeasy