openwrt/packages #1088: Transmission build fail

wongsyrone, 2015-03-29:
fallocate64 changed after trunk commit cfb8415a75e9
```
make[5]: Entering directory '/home/wong/github/openwrt-1/build_dir/target-mips_34kc_uClibc-0.9.33.2/transmission-2.84/libtransmission'
  CC       announcer.o
  CC       announcer-http.o
  CC       announcer-udp.o
  CC       bandwidth.o
  CC       bitfield.o
  CC       blocklist.o
  CC       cache.o
  CC       clients.o
  CC       completion.o
  CC       ConvertUTF.o
  CC       crypto.o
  CC       fdlimit.o
In file included from fdlimit.c:39:0:
/home/wong/github/openwrt-1/staging_dir/toolchain-mips_34kc_gcc-4.8-linaro_uClibc-0.9.33.2/include/fcntl.h:256:12: error: conflicting types for 'fallocate64'
 extern int fallocate64 (int __fd, int __mode, __off64_t __offset, __off64_t __len);
            ^
fdlimit.c:28:14: note: previous declaration of 'fallocate64' was here
   extern int fallocate64 (int fd, int mode, uint64_t offset, uint64_t len);
              ^
Makefile:1121: recipe for target 'fdlimit.o' failed
make[5]: *** [fdlimit.o] Error 1
make[5]: Leaving directory '/home/wong/github/openwrt-1/build_dir/target-mips_34kc_uClibc-0.9.33.2/transmission-2.84/libtransmission'
```

guentir, 2015-03-29:
same here on Archlinux X86_64 (mpc85xx)

wongsyrone, 2015-03-29:
https://dev.openwrt.org/ticket/19344

wongsyrone, 2015-03-29:
I deleted patch `010_libtransmission_fallocate64_eglibc.patch` which content is:
```diff
--- a/libtransmission/fdlimit.c	2014-07-01 20:08:59.000000000 +0300
+++ b/libtransmission/fdlimit.c	2014-07-13 13:30:17.000000000 +0300
@@ -22,7 +22,8 @@
  #include <fcntl.h>
 #endif
 
-#ifdef HAVE_FALLOCATE64
+/* no need to define fallocate64 in glibc/eglibc case */
+#if defined (__UCLIBC__) && (HAVE_FALLOCATE64)
   /* FIXME can't find the right #include voodoo to pick up the declaration.. */
   extern int fallocate64 (int fd, int mode, uint64_t offset, uint64_t len);
 #endif
```
and replace it with file `010-delete-extern-fallocate64.patch` which content is
```diff
--- ./libtransmission/fdlimit.c	2015-03-29 17:02:37.736463097 +0800
+++ ./libtransmission.new/fdlimit.c	2015-03-29 17:01:38.156461786 +0800
@@ -22,11 +22,6 @@
  #include <fcntl.h>
 #endif
 
-#ifdef HAVE_FALLOCATE64
-  /* FIXME can't find the right #include voodoo to pick up the declaration.. */
-  extern int fallocate64 (int fd, int mode, uint64_t offset, uint64_t len);
-#endif
-
 #ifdef HAVE_XFS_XFS_H
  #include <xfs/xfs.h>
 #endif
```
Because the function definition exported by the libtransmission/fdlimit.c is lightly different than the function definition in fcntl.h that is installed by uclibc from r45100. The previous [patch](https://lists.openwrt.org/pipermail/openwrt-devel/2012-July/016199.html) tends to export function to work both with uclibc and eglibc, but it is useless now cause uclibc and eglibc don't need export.

compile successfully after changes above.

obsy, 2015-03-29:
@guentir: can you confirm that it work on mpc85xx?

guentir, 2015-03-29:
@obsy, i will try it, please be patience.

guentir, 2015-03-29:
works now, errors are gone.

closed 2015-03-31 by wongsyrone